arendst/Tasmota #9540: BUG in rules: var<x>#state & assigning string to variable

askold2, 2020-10-14:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Using the var<x>#state or mem<x>#state trigger in the rules causes an "Event too long" error if any variable (even not used in the rule) is assigned a string value.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff D1 dimmer
- [x] Tasmota binary firmware version number used: 8.5.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
{"NAME":"Sonoff D1","GPIO":[255,148,255,149,255,255,0,0,0,56,255,0,0],"FLAG":0,"BASE":74}
{"Module":{"0":"Sonoff D1"}}
{"GPIO0":{"9":"Switch1"},"GPIO1":{"148":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"149":"Serial Rx"},"GPIO4":{"0":"None"},"GPIO5":{"18":"Button2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
{"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":34,"Free":477,"Rules":"on var1#state==1 do power on endon"}
{"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
{"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

15:42:35 MQT: stat/tasmota_CF1118/STATUS = {"Status":{"Module":0,"DeviceName":"DimmerD1","FriendlyName":["DimmerD1"],"Topic":"tasmota_CF1118","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[12,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:42:35 MQT: stat/tasmota_CF1118/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T01:43:50","StartupUTC":"2020-10-14T12:58:45","Sleep":50,"CfgHolder":4617,"BootCount":79,"BCResetTime":"2020-10-09T10:34:45","SaveCount":1764,"SaveAddress":"F8000"}}
15:42:35 MQT: stat/tasmota_CF1118/STATUS2 = {"StatusFWR":{"Version":"8.5.1(tasmota)","BuildDateTime":"2020-10-02T10:09:47","Boot":7,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"402/699"}}
15:42:35 MQT: stat/tasmota_CF1118/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BM4644","UR2323"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000A00B","0A05C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
15:42:35 MQT: stat/tasmota_CF1118/STATUS4 = {"StatusMEM":{"ProgramSize":596,"Free":404,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
15:42:35 MQT: stat/tasmota_CF1118/STATUS5 = {"StatusNET":{"Hostname":"tasmota_CF1118-4376","IPAddress":"192.168.100.4","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer":"94.140.14.140","Mac":"C8:2B:96:CF:11:18","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:42:35 MQT: stat/tasmota_CF1118/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.100.200","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CF1118","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
15:42:35 MQT: stat/tasmota_CF1118/STATUS7 = {"StatusTIM":{"UTC":"2020-10-14T14:42:35","Local":"2020-10-14T15:42:35","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:09","Sunset":"18:01"}}
15:42:35 MQT: stat/tasmota_CF1118/STATUS10 = {"StatusSNS":{"Time":"2020-10-14T15:42:35","Switch1":"OFF"}}
15:42:35 MQT: stat/tasmota_CF1118/STATUS11 = {"StatusSTS":{"Time":"2020-10-14T15:42:35","Uptime":"0T01:43:50","UptimeSec":6230,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":2,"SSId":"UR2323","BSSId":"E0:24:7F:28:87:56","Channel":12,"RSSI":26,"Signal":-87,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

15:46:47 CMD: rule1 on var1#state==1 do power on endon
15:46:47 SRC: WebConsole from 192.168.100.2
15:46:47 CMD: Group 0, Index 1, Command "RULE", Data "on var1#state==1 do power on endon"
15:46:47 RUL: Stored uncompressed, would compress from 34 to 24 (-29%)
15:46:47 MQT: stat/tasmota_CF1118/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":34,"Free":477,"Rules":"on var1#state==1 do power on endon"}
15:46:57 WIF: Checking connection...
15:47:01 MQT: Received Topic "domoticz/out", Data Size 431, Data "{
	"Battery" : 255,
	"LevelNames" : "Off|MiMix|Enthusiastic|GuitarClassic|IceWorldPiano|LeisureTime|Childhood|MorningStreamlet|MusicBox|Orange|Thinker",
	"RSSI" : 12,
	"SelectorStyle" : "1",
	"description" : "",
	"dtype" : "Light/Switch",
	"hwid" : "2",
	"id" : "DCB3A5EE",
	"idx" : 6,
	"name" : "Xiaomi Gateway Alarm Clock",
	"nvalue" : 0,
	"stype" : "Selector Switch",
	"svalue1" : "0",
	"switchType" : "Selector",
	"unit" : 4
}
"
15:47:01 DOM: idx 6, nvalue 0
15:47:01 CMD: var1 1
15:47:01 SRC: WebConsole from 192.168.100.2
15:47:01 CMD: Group 0, Index 1, Command "VAR", Data "1"
15:47:01 MQT: stat/tasmota_CF1118/RESULT = {"Var1":"1"}
15:47:01 RUL: VAR1#STATE==1 performs "power on"
15:47:01 SRC: Rule
15:47:01 CMD: Group 0, Index 1, Command "POWER", Data "on"
15:47:01 MQT: domoticz/in = {"idx":27,"nvalue":1,"svalue":"1","Battery":100,"RSSI":2}
15:47:01 MQT: stat/tasmota_CF1118/RESULT = {"POWER":"ON"}
15:47:01 MQT: stat/tasmota_CF1118/POWER = ON
15:47:01 DMP: AA 55 01 04 00 00 05
15:47:01 CFG: Saved to flash at F4, Count 1768, Bytes 4096
15:47:11 CMD: var2 1
15:47:11 SRC: WebConsole from 192.168.100.2
15:47:11 CMD: Group 0, Index 2, Command "VAR", Data "1"
15:47:11 MQT: stat/tasmota_CF1118/RESULT = {"Var2":"1"}
15:47:17 WIF: Checking connection...
15:47:18 CMD: var1 qwerty
15:47:18 SRC: WebConsole from 192.168.100.2
15:47:18 CMD: Group 0, Index 1, Command "VAR", Data "qwerty"
15:47:18 MQT: stat/tasmota_CF1118/RESULT = {"Var1":"qwerty"}
15:47:18 RUL: Event too long (25)
15:47:25 CMD: var2 qwerty
15:47:25 SRC: WebConsole from 192.168.100.2
15:47:25 CMD: Group 0, Index 2, Command "VAR", Data "qwerty"
15:47:25 MQT: stat/tasmota_CF1118/RESULT = {"Var2":"qwerty"}
15:47:25 RUL: Event too long (25)

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

weblog 3
rule1 on var1#state==1 do power on endon
var1 1
var1 qwerty
var2 1
var2 qwerty

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

It is expected that a variable can be assigned any value, not just a numeric value. In any case, a variable not used in a rule should not affect this rule.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

I used this rule in tasmota 8.2.0 and it worked there, but stopped working in 8.5.1:

Rule1 on system#boot var1 + ENDON ON var1#state=+ do var2 - endon ON var1#state=- do var2 + endon ON switch1#state=2 DO power toggle ENDON ON switch1#state=4 DO dimmer %var1% ENDON ON switch1#state=5 DO var1 %var2% ENDON ON switch1#state=6 DO var1 + ENDON

I started to figure out why and found that now assigning a string value to a variable causes an error. Moreover, the error occurs even when the string is assigned to any variable, not just the one used in the rule.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-10-14:
Looks like a general JSON parsing problem after JSMN, for non-numeric values. Here's a small example working in 8.3.1 but failing on 8.5.1:
`on event#abc do var1 %value% endon`
`event abc=def`
Error message is the same *Event too long*

arendst, 2020-10-15:
Pls report if this solves your issue.

sfromis, 2020-10-15:
My small test case works now :-)

askold2, 2020-10-15:
Yes, it works!

closed 2020-10-15 by askold2