arendst/Tasmota #9541: IRSEND with variable Mem not working

cool-lox, 2020-10-14:
### PROBLEM DESCRIPTION
Can`t address IRSEND with Mem Variable, or using wrong syntax.
I´m using YTF IR Bridge Module, with Tasmota-ir.bin 9.0.0.2
Example with direct codes works:
IRSEND 0,+8950-4570+540-580C-600+515-1690CdC-575CdCdFe+545-1715F-1695C-605+520gCkFjCkCgIeC-550CgCjFkCgCgCdCkCjFdCdFlCdChF-1720F-40320A-2310C
Result: {"IRSend":"Done"}

But I like to use with variable Mem1 = +8950-4570+540-580C-600+515-1690CdC-575CdCdFe+545-1715F-1695C-605+520gCkFjCkCgIeC-550CgCjFkCgCgCdCkCjFdCdFlCdChF-1720F-40320A-2310C
But when I try: IRSEND 0,%Mem1%  I get {"IRSend":"Invalid RawData"}

also other try with {"Mem2":"0,+8950-4570+540-580C-600+515-1690CdC-575CdCdFe+545-1715F-1695C-605+520gCkFjCkCgIeC-550CgCjFkCgCgCdCkCjFdCdFlCdChF-1720F-40320A-2310C"}
IRSEND Mem2 gives {"IRSend":"Invalid RawData"}
also IRSEND %Mem2% not working

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Module 62
- [x] Tasmota binary firmware version number used: Tasmota-ir.bin 9.0.0.2
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-10-14:
Mem strings over ten characters are not supported as there is just no room available to store them.

So it's as designed and cannot be solved. Use external storage like mqtt to send the raw data.

cool-lox, 2020-10-14:
Aha thanks, but why I can see in Mem1 the characters stored ?
RESULT = {"Mem1":"+8895-4450+565-545CdC-1660CdCdCdCd+560-550FeCe+540-570FeCeCeCeCeCeFeHiCeCeCgFeCdFgCgFeCdCdCeCdCeC-39530+8900-2245H"}
Thats why I thought I can use the variable.

arendst, 2020-10-14:
Ok. Then in that case it can only be used in rules. Fimd how to define an event using irsend and %mem1%. That could work.

cool-lox, 2020-10-14:
I tried again with new rule and %Mem2%  and it WORKS thanks

RESULT = {"Mem2":"0,+8950-4570+540-580C-600+515-1690CdC-575CdCdFe+545-1715F-1695C-605+520gCkFjCkCgIeC-550CgCjFkCgCgCdCkCjFdCdFlCdChF-1720F-40320A-2310C"}

RESULT = {"Rule3":"ON","Once":"OFF","StopOnError":"OFF","Length":40,"Free":471,"Rules":"ON Power2#State=1 DO IRSEND %Mem2% ENDON"}

RUL: POWER2#STATE=1 performs "IRSEND 0,+8950-4570+540-580C-600+515-1690CdC-575CdCdFe+545-1715F-1695C-605+520gCkFjCkCgIeC-550CgCjFkCgCgCdCkCjFdCdFlCdChF-1720F-40320A-2310C"

RESULT = {"IRSend":"Done"}

Thanks Theo

sfromis, 2020-10-14:
There are 3 rule buffers, but you can have more than one rule (on...endon) in each buffer. Not a memory problem, and with compression each buffer can have over 1KB of rules.

cool-lox, 2020-10-14:
mhh, when I set rule3:
rule3 ON Power2#State=1 DO IRSEND %Mem1% ENDON
This message appears:  "Stored uncompressed, would compress from 40 to 38 (-5%)"
But when I try a similar rule4 it gives ERROR
rule4 ON Power2#State=0 DO IRSEND %Mem2% ENDON

cool-lox, 2020-10-14:
Thanks for 3 rule buffer info, I made more on..endon commands in rule1. And all with direct IRSEND codes, without %mem<x>". It works so far. (Rule4 does not exsist, is clear now.)

closed 2020-10-14 by cool-lox