arendst/Tasmota #9545: RGB controller with Nuvoton chip changes color on its own (Sonoff L1)

frettchen98, 2020-10-14:
### PROBLEM DESCRIPTION
My RGB controller with esp 8285 and novoton chip is changing color on its own to red, blue, green

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff L1
- [x] Tasmota binary firmware version number used: 8.5.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: tasmotizer 1.2
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
21:36:50 CMD: Backlog Template; Module; GPIO 255
21:36:51 RSL: RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
21:36:51 RSL: RESULT = {"Module":{"70":"Sonoff L1"}}
21:36:51 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"148":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"149":"Serial Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
no rules

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
21:38:07 CMD: status 0
21:38:07 RSL: STATUS = {"Status":{"Module":70,"DeviceName":"TasmotaLed1","FriendlyName":["LED-Strip"],"Topic":"tasmota_C6FD66","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:38:07 RSL: STATUS1 = {"StatusPRM":{"Baudrate":19200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/","RestartReason":"Software/System restart","Uptime":"0T00:10:51","StartupUTC":"2020-10-14T20:27:16","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2020-10-13T03:34:49","SaveCount":296,"SaveAddress":"F4000"}}
21:38:07 RSL: STATUS2 = {"StatusFWR":{"Version":"8.5.1(tasmota)","BuildDateTime":"2020-10-02T10:09:47","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"379/699"}}
21:38:07 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Muecke-IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
21:38:07 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":596,"Free":404,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
21:38:07 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_C6FD66-7526","IPAddress":"10.0.30.39","Gateway":"10.0.30.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.30.1","Mac":"DC:4F:22:C6:FD:66","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:38:07 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-10-14T20:38:07","Local":"2020-10-14T21:38:07","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:09","Sunset":"18:01"}}
21:38:07 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-10-14T21:38:07"}}
21:38:07 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-10-14T21:38:07","Uptime":"0T00:10:51","UptimeSec":651,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":85,"Color":"D90202","HSBColor":"0,99,85","Channel":[85,1,1],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Muecke-IoT","BSSId":"C6:FB:E4:DA:08:49","Channel":8,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
21:38:48 RSL: RESULT = {"POWER":"ON","Dimmer":85,"Color":"D9C702","HSBColor":"55,99,85","Channel":[85,78,1]}
21:38:49 RSL: RESULT = {"POWER":"ON","Dimmer":85,"Color":"D99502","HSBColor":"41,99,85","Channel":[85,58,1]}
21:38:49 RSL: RESULT = {"POWER":"ON","Dimmer":85,"Color":"D9AB02","HSBColor":"47,99,85","Channel":[85,67,1]}
21:38:52 RSL: RESULT = {"POWER":"ON","Dimmer":85,"Color":"D94E02","HSBColor":"21,99,85","Channel":[85,31,1]}
21:38:53 RSL: RESULT = {"POWER":"ON","Dimmer":85,"Color":"D90E02","HSBColor":"3,99,85","Channel":[85,5,1]}
21:38:54 RSL: RESULT = {"POWER":"ON","Dimmer":85,"Color":"D90202","HSBColor":"0,99,85","Channel":[85,1,1]}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
I use the slider in main menu to switch color to e.g. orange, which is working and can be seen on the LED strip, but after seconds it goes step by step to red

If I switch to teal it goes to blue ...

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

color stays at the choosen color

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

I already reset everything, because I thought this issues was related to mqtt, which it wasnt

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-10-15:
Hi, please update to latest Tasmota version and try again. Thanks

frettchen98, 2020-10-15:
I'm now on version 9.0.0.2, but the issue stays exactly the same. 
I did some more tests. This issue does not effect toggle function and brightness, only color. Even when I use the IR remote control the color doesn't stay and changes back to red, green or blue, but only if I choose L1 (70) functionality.

arendst, 2020-10-16:
Pls report the output of command ``scheme``

arendst, 2020-10-16:
Currently I cannot reproduce. I've tried both 8.5.1 and 9.0.0.2 and both perform as before. I also disabled MQTT as you did.

My console output with weblog 4 reports a lot more information when a color is changed. Are you sure your reported console output is all you receive?

Mine look at least like this:
```
10:52:30 SRC: Switch
10:52:30 CMD: Group 0, Index 2, Command "COLOR", Data "7f00ff"
10:52:30 RSL: RESULT = {"POWER":"ON","Dimmer":85,"Color":"6C00D9","HSBColor":"270,100,85","Channel":[42,0,85]}
10:52:31 CFG: Saved to flash at F8, Count 938, Bytes 4096
10:52:32 WIF: Checking connection...
10:52:47 SRC: Switch
10:52:47 CMD: Group 0, Index 2, Command "COLOR", Data "00ffbf"
10:52:47 RSL: RESULT = {"POWER":"ON","Dimmer":85,"Color":"00D9A3","HSBColor":"165,100,85","Channel":[0,85,64]}
10:52:47 CFG: Saved to flash at F7, Count 939, Bytes 4096
```

puffel8611, 2020-10-16:
I’m sorry, didn’t pay enough attention there. Was the normal Output, Not Weblog 4

arendst, 2020-10-17:
From your logging I havr to conclude the color changes are comming from your Nuvotron chip. 

The way the L1 works is that IR remote detection and light control is all done by the Nuvotron chip. The ESP8266 chip running Tasmota is just a kind of slave following the nuvotron chip. Tasmota uses serial comms between the ESP8266 chip and the nuvotron chip. 

What you experience is color control initiated by the nuvotron chip communicated to Tasmota over serial hence the line ``SRC: Switch``. What could happen is that the IR remote is sending commands to the nuvotron to change colors in a sequential manner.

Another, dark reason could be that the current nuvotron firmware (not under control of Tasmota) performs this task for some reason.

arendst, 2020-10-17:
It looks like it's only doing this on Green (color RRGGBB) and just counting down to zero. Wierd.

puffel8611, 2020-10-19:
You are right it is counting down to zero, but its not only green. That is depending on the color I choose. Can be red and blue as well. The funny thing is: It also happens, when I choose the color by IR. And I can definitely rule out the IR remote, because I haven`t had one anywhere near sometimes. 

I'm almost sure it has to do with the Nuvoton firmware (unfortunately). For me it seems like the firmware wants to have some bi-directional communication with esp, otherwise it doesn`t accept the color.

arendst, 2020-11-08:
I'm revisiting the L1 code. I see anomalies releated to scaling up the color values needed for the L1. On the other hand I don't do scaling down so there is something wrong in my code.

In addition I think your Nuvotron microcode is newer than mine and it responds to every color change with it's current color slightly different from what I calculated by scaling. This triggers a loop where the colors decrease to zero.

Nice issue.

arendst, 2020-11-08:
Pls try the latest dev version and report back.

closed 2020-11-09 by ascillato2