arendst/Tasmota #9547: Power Monitoring Not Populating

jonootto, 2020-10-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I'm using a tuya lightswitch which includes power monitoring, this feature has suddenly stopped working on all of my devices, and I can't identify any changes which have broken it.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [  x ] Device used (e.g., Sonoff Basic): BrilliantSmart Light Switch (TuyaMCU)
- [ x ] Tasmota binary firmware version number used: 9.0.0.2 - tasmota.bin.gz
  - [ x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:

```
  Configuration output here:
17:02:49 CMD: Backlog Template; Module; GPIO 255
17:02:49 MQT: stat/LivingRoom/RESULT = {"NAME":"Generic","GPIO":[0,0,0,0,0,224,0,0,0,0,64,0,0,0],"FLAG":0,"BASE":54}
17:02:49 MQT: stat/LivingRoom/RESULT = {"Module":{"54":"Tuya MCU"}}
17:02:49 MQT: stat/LivingRoom/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
N/A

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
``
`17:03:41 CMD: Status 0
17:03:41 MQT: stat/LivingRoom/STATUS = {"Status":{"Module":54,"DeviceName":"LivingRoom","FriendlyName":["Living Room","Dining Room"],"Topic":"LivingRoom","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:03:41 MQT: stat/LivingRoom/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:04:31","StartupUTC":"2020-10-15T03:59:10","Sleep":1,"CfgHolder":4617,"BootCount":40,"BCResetTime":"2020-10-02T02:43:50","SaveCount":228,"SaveAddress":"F8000"}}
17:03:41 MQT: stat/LivingRoom/STATUS2 = {"StatusFWR":{"Version":"9.0.0.2(tasmota)","BuildDateTime":"2020-10-14T07:07:20","Boot":31,"Core":"2_7_4_3","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"440/699"}}
17:03:41 MQT: stat/LivingRoom/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["RG-IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00CA8009","2805C8000100068000005A00000000000000","00000200","00006000","00000000"]}}
17:03:41 MQT: stat/LivingRoom/STATUS4 = {"StatusMEM":{"ProgramSize":600,"Free":400,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
17:03:41 MQT: stat/LivingRoom/STATUS5 = {"StatusNET":{"Hostname":"LivingRoom-5425","IPAddress":"192.168.30.8","Gateway":"192.168.30.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.30.1","Mac":"BC:DD:C2:8D:35:31","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
17:03:41 MQT: stat/LivingRoom/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.254","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8D3531","MqttUser":"mqttadmin","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:03:41 MQT: stat/LivingRoom/STATUS7 = {"StatusTIM":{"UTC":"2020-10-15T04:03:41","Local":"2020-10-15T17:03:41","StartDST":"2020-09-27T02:00:00","EndDST":"2020-04-05T02:00:00","Timezone":99,"Sunrise":"19:11","Sunset":"05:59"}}
17:03:41 MQT: stat/LivingRoom/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
17:03:41 MQT: stat/LivingRoom/STATUS10 = {"StatusSNS":{"Time":"2020-10-15T17:03:41","ENERGY":{"TotalStartTime":"2020-10-02T20:22:59","Total":0.188,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
17:03:41 MQT: stat/LivingRoom/STATUS11 = {"StatusSTS":{"Time":"2020-10-15T17:03:41","Uptime":"0T00:04:31","UptimeSec":271,"Heap":22,"SleepMode":"Dynamic","Sleep":1,"LoadAvg":999,"MqttCount":1,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"RG-IoT","BSSId":"0E:EC:DA:E9:1D:97","Channel":6,"RSSI":100,"Signal":-48,"LinkCount":1,"Downtime":"0T00:00:03"}}}`
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

`15:50:46 {"TuyaReceived":{"Data":"55AA0107000865020004000000007A","Cmnd":7,"CmndData":"6502000400000000","DpType2Id101":0,"101":{"DpId":101,"DpIdType":2,"DpIdData":"00000000"}}}
15:50:46 TYA: fnId=0 is set for dpId=101
15:50:46 TYA: RX value 0 from dpId 101 
15:50:46 {"TuyaReceived":{"Data":"55AA010700086602000400000377F5","Cmnd":7,"CmndData":"6602000400000377","DpType2Id102":887,"102":{"DpId":102,"DpIdType":2,"DpIdData":"00000377"}}}
15:50:46 TYA: fnId=32 is set for dpId=102
15:50:46 TYA: RX value 887 from dpId 102 
15:50:46 {"TuyaReceived":{"Data":"55AA01070008670200040000034CCB","Cmnd":7,"CmndData":"670200040000034C","DpType2Id103":844,"103":{"DpId":103,"DpIdType":2,"DpIdData":"0000034C"}}}
15:50:46 TYA: fnId=31 is set for dpId=103
15:50:46 TYA: RX value 844 from dpId 103 
15:50:46 {"TuyaReceived":{"Data":"55AA010700086802000400000946CC","Cmnd":7,"CmndData":"6802000400000946","DpType2Id104":2374,"104":{"DpId":104,"DpIdType":2,"DpIdData":"00000946"}}}
15:50:46 TYA: fnId=33 is set for dpId=104
15:50:46 TYA: RX value 2374 from dpId 104 `

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

I have configured the voltage, current and power inputs, these appear to being read correctly in the logs:
`Backlog TuyaMCU 11,1; TuyaMCU 12,2; TuyaMCU 32,102; TuyaMCU 31,103; TuyaMCU 33,104; `

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expect the power values to populate on the landing page, instead they all read 0, except for the total power which did increment when this used to work.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-10-15:
Tbh, will be nearly impossible to solve your issue. If it worked before and now it doesnt if you have nothing changed.
Looking at your logs, you use latest development version. Have you upgraded Tasmota? If so go back to your previous
version. There is always a chance that in development version things break.

jonootto, 2020-10-16:
Thanks, yes. I’ve tried both release and development versions, resetting the configuration and different builds to no avail.

jonootto, 2020-10-16:
Ok, I've reverted to 8.5.0, it now works again, i suggest this may be a bug in the newer versions

jonootto, 2020-10-16:
Is there any chance it's related to #9330 ?

ozadr1an, 2020-10-17:
I've got the same issue, confirmed that works fine in 8.5.0 but is broken from 8.5.1 onwards.  It still is an issue as well in 9.0.0.2 . 

If I do a weblog 4 , I can see the values are being reported from TuyaMCU the same,  its not updating into tasmota though. 

Under 8.5.1:
`06:45:56 TYA: fnId=0 is set for dpId=101
06:45:56 TYA: RX value 0 from dpId 101 
06:45:57 {"TuyaReceived":{"Data":"55AA0107000866020004000000007B","Cmnd":7,"CmndData":"6602000400000000","DpType2Id102":0,"102":{"DpId":102,"DpIdType":2,"DpIdData":"00000000"}}}
06:45:57 TYA: fnId=32 is set for dpId=102
06:45:57 TYA: RX value 0 from dpId 102 
06:45:57 {"TuyaReceived":{"Data":"55AA0107000867020004000000007C","Cmnd":7,"CmndData":"6702000400000000","DpType2Id103":0,"103":{"DpId":103,"DpIdType":2,"DpIdData":"00000000"}}}
06:45:57 TYA: fnId=31 is set for dpId=103
06:45:57 TYA: RX value 0 from dpId 103 
06:45:57 {"TuyaReceived":{"Data":"55AA0107000868020004000009941A","Cmnd":7,"CmndData":"6802000400000994","DpType2Id104":2452,"104":{"DpId":104,"DpIdType":2,"DpIdData":"00000994"}}}
06:45:57 TYA: fnId=33 is set for dpId=104
06:45:57 TYA: RX value 2452 from dpId 104 
.....
06:47:19 MQT: tele/Breakfast_Nook_Switch/SENSOR = {"Time":"2020-10-17T06:47:19","ENERGY":{"TotalStartTime":"2020-10-12T16:49:13","Total":0.21183,"Yesterday":0.03185,"Today":0.00000,"Period":0.000,"Power":0.000,"ApparentPower":0.000,"ReactivePower":0.000,"Factor":0.00,"Voltage":0.000,"Current":0.000}}
06:47:28 CMD: Status 8
06:47:28 MQT: stat/Breakfast_Nook_Switch/STATUS8 = {"StatusSNS":{"Time":"2020-10-17T06:47:28","ENERGY":{"TotalStartTime":"2020-10-12T16:49:13","Total":0.21183,"Yesterday":0.03185,"Today":0.00000,"Power":0.000,"ApparentPower":0.000,"ReactivePower":0.000,"Factor":0.00,"Voltage":0.000,"Current":0.000}}}` 

Under 8.5.0:
`
06:50:52 TYA: fnId=0 is set for dpId=101
06:50:52 {"TuyaReceived":{"Data":"55AA0107000866020004000000007B","Cmnd":7,"CmndData":"6602000400000000","DpType2Id102":0,"102":{"DpId":102,"DpIdType":2,"DpIdData":"00000000"}}}
06:50:52 TYA: fnId=32 is set for dpId=102
06:50:52 TYA: Rx ID=102 Current=0
06:50:52 {"TuyaReceived":{"Data":"55AA0107000867020004000000007C","Cmnd":7,"CmndData":"6702000400000000","DpType2Id103":0,"103":{"DpId":103,"DpIdType":2,"DpIdData":"00000000"}}}
06:50:52 TYA: fnId=31 is set for dpId=103
06:50:52 TYA: Rx ID=103 Active_Power=0
06:50:52 {"TuyaReceived":{"Data":"55AA0107000868020004000009941A","Cmnd":7,"CmndData":"6802000400000994","DpType2Id104":2452,"104":{"DpId":104,"DpIdType":2,"DpIdData":"00000994"}}}
06:50:52 TYA: fnId=33 is set for dpId=104
06:50:52 TYA: Rx ID=104 Voltage=2452
....
06:49:57 MQT: stat/Breakfast_Nook_Switch/STATUS8 = {"StatusSNS":{"Time":"2020-10-17T06:49:57","ENERGY":{"TotalStartTime":"2020-10-12T16:49:13","Total":0.21183,"Yesterday":0.03185,"Today":0.00000,"Power":0.000,"ApparentPower":0.000,"ReactivePower":0.000,"Factor":0.00,"Voltage":247.100,"Current":0.000}}}`

Note the Voltage is correctly reflected in the STATUS8 line under 8.5.0 but not 8.5.1

arendst, 2020-10-17:
Pls try the latest dev version and report back if it solves your energy issue.

jonootto, 2020-10-17:
> 
> 
> Pls try the latest dev version and report back if it solves your energy issue.

Thanks so much! That's fixed it.
[image]

Jason2866, 2020-10-17:
Great it works again. Please close the issue. Thx!

closed 2020-10-17 by jonootto

webfrank, 2020-11-16:
Hi,
   I had the same problem and upgraded to latest dev 9.1.0.1, the web interface is correctly populated but MQTT data is not.
I'm using a ZMAi-90 and it seems Tasmota is sending mqtt messages once in a while. Am I missing a setting?