arendst/Tasmota #9548: DS18B20 not available after Deepsleep wakeup

helmar74, 2020-10-15:
Hi,

I build a battery powered sensor with a 18B20 sensor. I've the following issue:
Connect a DS18B20 to GPIO2 (Pin D4). 
Values will be shown in WebUI
Send DEvice to Deepleep (deepsleeptime 60)
After the Devices wakes up, the DS18B20 isn'T recognized anymore. 
The only way bringing back the values ist to restart the device.
The DS18B20 Values are shown again, after next deepsleep the sensor isn'T recognized anymore

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): Wemos Nini D1 Pro
- [X ] Tasmota binary firmware version number used: 8.5.1
  - [ ] Pre-compiled
  - [X ] Self-compiled
    - [X ] IDE / Compiler used: PlatformIO
- [X] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Connect a DS18B20 to GPIO2 (Pin D4). 
Values will be shown in WebUI
Send DEvice to Deepleep (deepsleeptime 60)
After the Devices wakes up, the DS18B20 isn'T recognized anymore. 
The only way bringing back the values ist to restart the device.
The DS18B20 Values are shown again, after next deepsleep the sensor isn'T recognized anymore

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2020-10-15:
I will take this over. Do you have also a relay defined to make the DS18B20 powerless during deepsleep? Otherwise, this can suck down your battery. Anyhow I will try to reproduce without a relay.

arendst, 2020-10-15:
Try a different gpio too considering the current is used for serial communication.

helmar74, 2020-10-15:
I'm using this configuration (with the same GPIO) since 2 years. There was never an issue

helmar74, 2020-10-15:
> I will take this over. Do you have also a relay defined to make the DS18B20 powerless during deepsleep? Otherwise, this can suck down your battery. Anyhow I will try to reproduce without a relay. Can you comment how you wire the DS18B20. Do you use the 2-wire hack with a resistor or have the 3-wire version?

[image]

For Connecting Ground I use the "Magic GPIO15 (D8)

In standby the power consumption is ~ 100µA

helmar74, 2020-10-15:
It looks, like this is an issue of version 8.5.1
I downgraded to Tasmota 8.5.0 there everything is working as expected.
Going back to 8.5.1 brings back the error. After deepsleep wake up the DS18B20 isn't initialized anymore.
I was able to reproduce this with 2 different Wemos Mini D1

stefanbode, 2020-10-15:
I tried to reproduce it with the current dev branch and was NOT able to reproduce. Also the very simple case D4 to middle pin and GND connect without VCC connect works fine with deepsleep

helmar74, 2020-10-15:
I was using 8.5.1 stable and there the error was reproducable.

stefanbode, 2020-10-15:
Ok, fine, then we can close the case as it is working in 9.0.1

helmar74, 2020-10-15:
> Ok, fine, then we can close the case as it is working in 9.0.1

Could you reproduce it in stable 8.5.1?

stefanbode, 2020-10-15:
Your problem is connected with the wakeup time on the pin D8. I assume the pin goes high too late and therefore the init process of the DS18B20 had no chance to find it. I try to reproduce with 9.0 to use D8 and go high a bit late and had the same problem. I will take a look if and how you can postpone or reinit the start of the DS18B20

helmar74, 2020-10-15:
> Your problem is connected with the wakeup time on the pin D8. I assume the pin goes high too late and therefore the init process of the DS18B20 had no chance to find it. I try to reproduce with 9.0 [...]

I don't think that's related. With version 8.5.0 it's working without an issue. Every time the devices wakes up, the DS18B20 is initiated and the temperature sent via MQTT to my broker. So I don't think it's an timing issue
With 8.5.1 it is broken

stefanbode, 2020-10-15:
It must be a timing issue because after a deepsleep the device starts from scratch. 
In `tasmota.ino` the start looks following:
```
  WifiConnect();

  SetPowerOnState();

  AddLog_P2(LOG_LEVEL_INFO, PSTR(D_PROJECT " %s %s " D_VERSION " %s%s-" ARDUINO_CORE_RELEASE), PROJECT, SettingsText(SET_DEVICENAME), my_version, my_image);
#ifdef FIRMWARE_MINIMAL
  AddLog_P2(LOG_LEVEL_INFO, PSTR(D_WARNING_MINIMAL_VERSION));
#endif  // FIRMWARE_MINIMAL

  memcpy_P(log_data, VERSION_MARKER, 1);  // Dummy for compiler saving VERSION_MARKER

  RtcInit();

#ifdef USE_ARDUINO_OTA
  ArduinoOTAInit();
#endif  // USE_ARDUINO_OTA

  XdrvCall(FUNC_INIT);
  XsnsCall(FUNC_INIT)
```;
The first SetPowerOnState enables your relays and power up the sensor. The `XsnsCall(FUNC_INIT)` call the init of the DS18. If at this time the DS is not ready, you're lost. If you have a self compiled file you can prove my assumption by changing the order of the first two lines: `SetPowerOnState(); WifiConnect();`

Connecting the wifi takes always some time. This must be far enough to get the sensor up and running. Just try it.

helmar74, 2020-10-15:
> It must be a timing issue because after a deepsleep the device starts from scratch.
> In `tasmota.ino` the start looks following:
> 
> ```
>   WifiConnect();
> 
>   SetPowerOnState();
> 
>   AddLog_P2(LOG_LEVEL_INFO, PSTR(D_PROJECT " %s %s " D_VERSION " %s%s-" ARDUINO_CORE_RELEASE), PROJECT, SettingsText(SET_DEVICENAME), my_version, my_image);
> #ifdef FIRMWARE_MINIMAL
>   AddLog_P2(LOG_LEVEL_INFO, PSTR(D_WARNING_MINIMAL_VERSION));
> #endif  // FIRMWARE_MINIMAL
> 
>   memcpy_P(log_data, VERSION_MARKER, 1);  // Dummy for compiler saving VERSION_MARKER
> 
>   RtcInit();
> 
> #ifdef USE_ARDUINO_OTA
>   ArduinoOTAInit();
> #endif  // USE_ARDUINO_OTA
> 
>   XdrvCall(FUNC_INIT);
>   XsnsCall(FUNC_INIT);
> ```
> 
> The first SetPowerOnState enables your relays and power up the sensor. The `XsnsCall(FUNC_INIT)` call the init of the DS18. If at this time the DS is not ready, you're lost. If you have a self compiled file you can prove my assumption by changing the order of the first two lines: `SetPowerOnState(); WifiConnect();`
> 
> Connecting the wifi takes always some time. This must be far enough to get the sensor up and running. Just try it.
> 
> @arendst : Does it make sense to swap the lines, in general, to give more time to power up devices at start?

OK, thanks. Hopefully you find a way to speed up the sensor initialization. The magic iof D8 is the only way I found bring down the power consumption during deepsleep...

stefanbode, 2020-10-15:
@helmar74 : The Wemos is not a battery chip. The onboard USB chip uses quite a lot of energy. All my battery-powered devices use a bare ESP8266 with a voltage regulator. These devices in the garden with a 18650 battery survive a year with an hourly wakeup. The drain of the wemos mini normally will suck down the battery in less than a month (max).

stefanbode, 2020-10-15:
Regarding: Why D8 is the only one that works: https://tasmota.github.io/docs/DeepSleep/#executing-commands-before-entering-deepsleep scroll down to "Findings" and you see D8 is the only one that go DOWN automatically in deepsleep.

helmar74, 2020-10-15:
> Regarding: Why D8 is the only one that works: https://tasmota.github.io/docs/DeepSleep/#executing-commands-before-entering-deepsleep scroll down to "Findings" and you see D8 is the only one that go DOWN automatically in deepsleep.

Yes I know that, thank you ;-)

helmar74, 2020-10-15:
> @helmar74 : The Wemos is not a battery chip. The onboard USB chip uses quite a lot of energy. All my battery-powered devices use a bare ESP8266 with a voltage regulator. These devices in the garden [...]

No, if you power it via 3,3V Pin not. May battery last's 10-12 month.  If you power it with 3,3V the complete "USB STack" isn't used and the WEmos Mini is acting as a bare ESP8266

helmar74, 2020-10-15:
> OK, thanks. Hopefully you find a way to speed up the sensor initialization. The magic iof D8 is the only way I found bring down the power consumption during deepsleep...

I tried switching the lines. This had no effect, the DS18B20 is still not initialized

tasmota.ini:
`// Die beiden Einträge getauscht wegen Wakeup Timer DS18B20 nach Deepsleep. https://github.com/arendst/Tasmota/issues/9548

  SetPowerOnState();
  
  WifiConnect();`

arendst, 2020-10-15:
Switching the lines makes no sense because the WifiConnect is asynchronous; it just setsup the wifi start sequence which will trigger the next second.

The user might want to use ``SO56 0`` to speed up wifi connection using known BSSID.

KDMcMullan, 2020-10-15:
I can't follow the thread, so don't know if it's still pertinent, but I have a problem where DS18B20 is not seen after a wake from deep sleep. v8.5.1. If I power my wemos D1 from USB it works fine. It used to work OKay with my previous version, which (I THINK) was v6.5. Hope this is helpful.

stefanbode, 2020-10-16:
We made a small video chat and the 8.5.1 really looks strange. 9.0 seem to fix it. The DS does not get recognized. But a normal "restart 1" fixes the problem and the DS is back again. The DS also does not report in 8.5.1 the temperature if it is permanently powered. Therefore a problem with the timing is excluded. @arendst : We can leave the code as it is. Normally the assumption is that after a deepsleep the device boots normally, but there is also a difference in the INFO1.. INFO3. These values are not reported after a deepsleep and I have no idea why and how the device recognize the difference between a normal reboot and deepsleep

helmar74, 2020-10-16:
> Switching the lines makes no sense because the WifiConnect is asynchronous; it just setsup the wifi start sequence which will trigger the next second.
> 
> The user might want to use `SO56 0` to speed up wifi connection using known BSSID.

setoption56 is set to "OFF"

arendst, 2020-10-16:
On users request the info messages are not being send after deepsleep wakeup. Search for REASON_DEEP_SLEEP_AWAKE in the code. There is an issue about this but I don't know by heart. It saves a lot of wakeup mqtt traffic. In addition SO90 was felt useful too.

helmar74, 2020-10-16:
Hi,

I compiled the development version 9.0.0.2 and it seems to work with this version.
Summary:
8.5.0: Deepsleep and DS18B20 is working
8.5.1:Deepsleep and DS18B20 is __not__ working
9.0.0.2: (Current development version): Deepsleep and DS18B20 is working

@arendst : Maybe it's worth that the is mentioned in the 8.5.1 changelog as a known issue?

@stefanbode : Thank you very much for your assistance and the nice video call yesterday evening.

I close the issue, as it's obviously solved with upcoming version 9.0.x

Regards

helmar

closed 2020-10-16 by helmar74