arendst/Tasmota #9553: Sonoff Zigbee Bridge Zigbee ACK timeout

Oblivionevil, 2020-10-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I use the Bridge for zigbee devices to control other devices with websend
After some time Zigbee stop working with this log 
{"ZbState":{"Status":99,"Message":"Failed state","Error":"ACK timeout","Code":81}}
before this happens zigbee is very slow after restart of the Bridge everything is fine.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Sonoff Zigbee Bridge
- [ x] Tasmota binary firmware version number used: 9.0.0.2
  - [ x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: ESPEasyFlasher
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
22:40:52 CMD: Backlog Template; Module; GPIO 255
22:40:53 RSL: RESULT = {"NAME":"Flash ZBBridge","GPIO":[224,5088,0,5120,224,225,0,0,0,576,0,0,32,0],"FLAG":0,"BASE":18}
22:40:53 RSL: RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
22:40:53 RSL: RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}
  Configuration output here:

```
- [x ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
22:42:51 CMD: Backlog Rule1; Rule2; Rule3`:
22:42:51 RUL: Rule1 on clock#Timer=1 do backlog Rule2 1 ; Rule3 0 ENDON       on clock#Timer=2 do backlog Rule2 0 ; Rule3 1 ENDON       on clock#Timer=3 do backlog WebSend [192.168.2.242] /win&A=0 ; WebSend [192.168.2.178] /win&A=0 ENDON       on clock#Timer=4 do backlog WebSend [192.168.2.242] /win&A=255 ; WebSend [192.168.2.178] /win&A=255 ENDON       on zbreceived#Klolüfterschalter#power=2 do backlog WebSend [192.168.2.175] /cm?cmnd=POWER1 Toggle ; WebSend [192.168.2.175] /cm?cmnd=POWER2 Toggle ENDON       on
22:42:51 RUL: Rule1 +zbreceived#Treppe#power=1 do WebSend [192.168.2.223] /cm?cmnd=dimmer 100 ENDON       on zbreceived#Treppe#power=0 do WebSend [192.168.2.223] /cm?cmnd=dimmer 0 ENDON       on zbreceived#Eingang#Occupancy=1 do WebSend [192.168.2.205] /cm?cmnd=dimmer 100 ENDON       on zbreceived#Eingang#Occupancy=0 do WebSend [192.168.2.205] /cm?cmnd=dimmer 0 ENDON       on zbreceived#Sebastianwürfel#AqaraCube=tap do WebSend [192.168.2.134] /cm?cmnd=power1 Toggle ENDON       on zbreceived#Sebastianwürfel#AqaraRotate<0 do
22:42:51 RUL: Rule1 +WebSend [192.168.2.134] /cm?cmnd=Channel1 - ENDON       on zbreceived#Sebastianwürfel#AqaraRotate>0 do WebSend [192.168.2.134] /cm?cmnd=Channel1 + ENDON
22:42:51 RSL: RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":1162,"Free":160,"Rules":"on clock#Timer=1 do backlog Rule2 1 ; Rule3 0 ENDON       on clock#Timer=2 do backlog Rule2 0 ; Rule3 1 ENDON       on clock#Timer=3 do backlog WebSend [192.168.2.242] /win&A=0 ; WebSend [192.168.2.178] /win&A=0 ENDON       on clock#Timer=4 do backlog WebSend [192.168.2.242] /win&A=255 ; WebSend [192.168.2.178] /win&A=255 ENDON       on zbreceived#Klolüfterschalter#power=2 do backlog WebSend [192.168.2.175] /cm?cmnd=POWER1 Toggle ; WebSend [192.168.2.175] /cm?cmnd=POWER2 Toggle ENDON       on zbreceived#Tr..."}
22:42:51 RUL: Rule2 on zbreceived#Kloflur#Occupancy=1 do WebSend [192.168.2.202] /win&A=255 ENDON       on zbreceived#Kloflur#Occupancy=0 do WebSend [192.168.2.202] /win&A=0 ENDON       on zbreceived#Flur_oben#Occupancy=1 do WebSend [192.168.2.9] /cm?cmnd=dimmer 100 ENDON       on zbreceived#Flur_oben#Occupancy=0 do WebSend [192.168.2.9] /cm?cmnd=dimmer 0 ENDON       on zbreceived#Sebastianlicht#power=2 do WebSend [192.168.2.134] /cm?cmnd=power1 Toggle ENDON       on zbreceived#Sebastianlicht#power=0 do WebSend
22:42:51 RUL: Rule2 +[192.168.2.134] /cm?cmnd=Channel1 + ENDON       on zbreceived#Sebastianlicht#power=1 do WebSend [192.168.2.134] /cm?cmnd=Channel1 - ENDON       on zbreceived#Klolüfterschalter#power=1 do  WebSend [192.168.2.224] /cm?cmnd=POWER1 Toggle ENDON
22:42:51 RSL: RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":738,"Free":288,"Rules":"on zbreceived#Kloflur#Occupancy=1 do WebSend [192.168.2.202] /win&A=255 ENDON       on zbreceived#Kloflur#Occupancy=0 do WebSend [192.168.2.202] /win&A=0 ENDON       on zbreceived#Flur_oben#Occupancy=1 do WebSend [192.168.2.9] /cm?cmnd=dimmer 100 ENDON       on zbreceived#Flur_oben#Occupancy=0 do WebSend [192.168.2.9] /cm?cmnd=dimmer 0 ENDON       on zbreceived#Sebastianlicht#power=2 do WebSend [192.168.2.134] /cm?cmnd=power1 Toggle ENDON       on zbreceived#Sebastianlicht#power=0 do WebSend [192.168.2.134]..."}
22:42:52 RUL: Stored uncompressed, would compress from 2 to 3 (-0%)
22:42:52 RSL: RESULT = {"Rule3":"ON","Once":"OFF","StopOnError":"OFF","Length":2,"Free":509,"Rules":"`:"}

```
- [x ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
22:43:40 CMD: Status 0
22:43:40 RSL: STATUS = {"Status":{"Module":75,"DeviceName":"ZbBridge","FriendlyName":["ZbBridge"],"Topic":"tasmota_DC0B16","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:43:40 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-minimal.bin","RestartReason":"Power On","Uptime":"0T02:55:03","StartupUTC":"2020-10-15T17:48:37","Sleep":0,"CfgHolder":4617,"BootCount":66,"BCResetTime":"2020-09-12T19:17:31","SaveCount":411,"SaveAddress":"FB000"}}
22:43:40 RSL: STATUS2 = {"StatusFWR":{"Version":"9.0.0.2(zbbridge)","BuildDateTime":"2020-10-13T13:57:50","Boot":31,"Core":"2_7_4_3","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"385/699"}}
22:43:40 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Smart",""],"TelePeriod":300,"Resolution":"95C180C0","SetOption":["00008001","2805C8000100060000005A00000000000000","00008001","0000620A","00000000"]}}
22:43:40 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":532,"Free":468,"Heap":22,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F100786","04500001","00000002","00000000","00000000","00020000","00000000"],"Drivers":"1,2,4,7,9,10,20,21,23,41","Sensors":""}}
22:43:40 RSL: STATUS5 = {"StatusNET":{"Hostname":"ZbBridge","IPAddress":"192.168.2.140","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"A4:CF:12:DC:0B:16","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
22:43:40 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-10-15T20:43:40","Local":"2020-10-15T22:43:40","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+02:00","Sunrise":"08:11","Sunset":"18:59"}}
22:43:40 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-10-15T22:43:40"}}
22:43:40 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-10-15T22:43:40","Uptime":"0T02:55:03","UptimeSec":10503,"Vcc":3.469,"Heap":22,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Smart","BSSId":"EC:08:6B:E5:83:A8","Channel":1,"RSSI":100,"Signal":-31,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
{"ZbState":{"Status":99,"Message":"Failed state","Error":"ACK timeout","Code":81}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
After 2-3 hours it happens

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-10-16:
Which zigbee EFR firmware do you have flashed?

Oblivionevil, 2020-10-16:
ncp-uart-sw_6.7.6_115200.ota

Jason2866, 2020-10-16:
Mhh, so it seems the rule/websend stuff is conflicting with the communication ESP82XX to the EFR MCU. My zbbridge (pure mqtt use) did not have ever this error.
Can you try with fewer rules/websends ?

Oblivionevil, 2020-10-16:
i have realized the whole smarthome via websend and zigbee i had hoped to get along without raspberry

Jason2866, 2020-10-16:
Nearly all Open Source Home Automation solutions use mqtt to communicate.
It only needs a mqtt broker. It is very leightweight and can be run for example on a OpenWRT router.

Oblivionevil, 2020-10-16:
Interesting all router running openwrt =)

Oblivionevil, 2020-10-16:
Is there a good manual (howto) for mqtt control other devices with zigbee

closed 2020-12-02 by ascillato2