arendst/Tasmota #9554: Reported baud rate not used until re-sent

KDMcMullan, 2020-10-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When I reboot TASMOTA, I query the baud rate. I'm told it is 57600. When I send a byte, oscilloscope shows that the The byte is about 75us long, suggesting a rate of about 115k.
I reset the baud rate to 57600, and resend the byte.
The byte is now about 150us long, suggesting a rate of about 57k6.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): WEMOS D1
- [X] Tasmota binary firmware version number used: 8.5.1
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: esptool 2.6
- [X Provide the output of command: `Backlog Template; Module; GPIO 255`:

  Configuration output here:
23:22:04 MQT: roomba/stat/RESULT = {"NAME":"Generic","GPIO":[255,148,255,149,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
23:22:04 MQT: gen/stat/RESULT = {"Module":{"18":"Generic"}}
23:22:04 MQT: gen/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"148":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"149":"Serial Rx"},"GPIO4":{"21":"Relay1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
23:25:04 MQT: gen/stat/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["gen"],"Topic":"gen","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:25:04 MQT: gen/stat/STATUS1 = {"StatusPRM":{"Baudrate":57600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T00:08:39","StartupUTC":"2020-10-15T22:16:25","Sleep":50,"CfgHolder":4617,"BootCount":21,"BCResetTime":"2020-10-13T22:57:32","SaveCount":74,"SaveAddress":"FA000"}}
23:25:04 MQT: gen/stat/STATUS2 = {"StatusFWR":{"Version":"8.5.1(tasmota)","BuildDateTime":"2020-10-02T10:09:47","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"336/699"}}
23:25:04 MQT: gen/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Scrabo",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00C08009","2805C8000100060000005A00000000000000","00800008","00006000","00000000"]}}
23:25:04 MQT: gen/stat/STATUS4 = {"StatusMEM":{"ProgramSize":596,"Free":404,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
23:25:04 MQT: gen/stat/STATUS5 = {"StatusNET":{"Hostname":"iotgen","IPAddress":"192.168.1.159","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"CC:50:E3:08:53:10","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
23:25:04 MQT: gen/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.111","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_085310","MqttUser":"device","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
23:25:04 MQT: gen/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-10-15T22:25:04","Local":"2020-10-15T23:25:04","StartDST":"2020-03-29T01:00:00","EndDST":"2020-10-25T02:00:00","Timezone":99,"Sunrise":"07:33","Sunset":"18:12"}}
23:25:04 MQT: gen/stat/STATUS10 = {"StatusSNS":{"Time":"2020-10-15T23:25:04"}}
23:25:04 MQT: gen/stat/STATUS11 = {"StatusSTS":{"Time":"2020-10-15T23:25:04","Uptime":"0T00:08:39","UptimeSec":519,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Scrabo","BSSId":"EC:BE:DD:37:EB:17","Channel":6,"RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
- set baudrate 57600.
- transmit byte, note that byte is 57k6.
- repower TASMOTA device.
- transmit byte.
- note that byte is sent at 115k.
- query baudrate, note that TASMOTA reports 57k6.
- set baudrate 57600.
- transmit byte, note that byte is now 57k6.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I would expect that when I set the baudrate to 57k6, then repower teh device TASMOTA continues to use 57k6. In fact, it seems it "remembers" that is should be using 57k6, but actually uses 115k.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

KDMcMullan, 2020-10-15:
Having set baudrate 57600:
[image]
Same byte, but after a power cycle:
[image]
Note that after a power cycle TASMOTA still reports that it is using 57600 baud.

arendst, 2020-10-16:
And the issue is?

KDMcMullan, 2020-10-16:
Just for clarity:
1) After a power cycle, TASMOTA will incorrectly report via the console that it is using the previously set baud rate (in my case, it reports 57k6 but it might actually be using 115k).
2) I must resend TASMOTA the baudrate command (57k6) after every power cycle.
Many thanks.

arendst, 2020-10-16:
In that case use a rule to set baudrate on reboot.

Just for my understanding, if you say 

> - transmit byte, note that byte IS sent at 57k6.

how do you transmit it? Using the serial console, web console, serialsend...

arendst, 2020-10-16:
The latest change solves at least the miscommunication of the current baudrate.

To change the baudrate on restart use a rule like ``on system#boot do baudrate 57600 endon``

closed 2020-10-16 by ascillato2

KDMcMullan, 2020-10-16:
> how do you transmit it? Using the serial console, web console, serialsend...

Web console serialsend, yes.

KDMcMullan, 2020-10-16:
> The latest change solves at least the miscommunication of the current baudrate.

I'm terribly glad it was not my imagination! Thank you.

> To change the baudrate on restart use a rule like `on system#boot do baudrate 57600 endon`

Naturally. I shall add such a note to the documentation, too. Thanks again, folks!