arendst/Tasmota #9555: Add support for the PJON protocol

gioblu, 2020-10-16:
**Have you looked for this feature in other issues and in the docs?**  

Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

No

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

Add the support for the [PJON library](https://github.com/gioblu/PJON), that would enable Tasmota to run over [PJON](https://github.com/gioblu/PJON/blob/master/specification/PJON-protocol-specification-v4.0.md) but in particular over [PJDL](https://github.com/gioblu/PJON/blob/master/src/strategies/SoftwareBitBang/specification/PJDL-specification-v5.0.md) using the [SoftwareBitBang](https://github.com/gioblu/PJON/blob/master/src/strategies/SoftwareBitBang/) physical layer. This single wire physical layer, developed in more than 10 years of experimentation, could be considered an alternative to 1-Wire, it has been reviewed by many makers and academic researchers and proved to be reliable and have a range of more than 2000m and provide connectivity for up to 255 devices using the same conductor. The [AnalogSampling](https://github.com/gioblu/PJON/tree/master/src/strategies/AnalogSampling) physical layer instead enable optical wireless communication even with just one visible light LED (used for both reception and transmission phase).

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

I do not see any other protocol with the mentioned capabilities, and it would be cool to leverage of those in Tasmota setups.

**Additional context**  
_Add any other context or screenshots about the feature request here._

See video demonstrations:
- SoftwareBitBang: https://www.youtube.com/watch?v=GWlhKD5lz5w
- AnalogSampling: https://www.youtube.com/watch?v=1BeGYMjg-DI

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-10-16:
Any ready available sensors using your PJON?

gioblu, 2020-10-16:
Ciao @arendst thank you very much for your quick reply and compliments for the huge work done on Tasmota.
With PJON you can create whatever sensor using really constrained MCUs, like ATtiny85, ATmega328 ecc. 
Many users have a "master" that is often an ESP connected with a SoftwareBitBang bus to other nodes (compatibility table [here](https://github.com/gioblu/PJON/tree/master/src/strategies/SoftwareBitBang#compatibility)) to quickly and cheaply implement their needs. For this reason I thought adding PJON to tasmota could enable quicker and simpler integration with other mcus and custom nodes. I would say it is mostly used by makers although we are working on some modules that will be soon in the public domain. See the examples directory to have an idea of how the library is used.

Jason2866, 2020-10-16:
Since it is used by makers there is  makers way :-)
Connected it via Tasmota Client. https://tasmota.github.io/docs/TasmotaClient

sfromis, 2020-10-16:
Anyone can create a Tasmota device driver for PJON, and submit a PR for consideration.

gioblu, 2020-10-16:
>Since it is used by makers there is makers way :-)
Connected it via Tasmota Client. https://tasmota.github.io/docs/TasmotaClient

Ciao @Jason2866 the link you provide is about connecting arduino compatible devices to tasmota via serial (point-to-point few meters)? PJON connects up to 255 arduino compatible devices over a single wire of up to 2 kilometers, or using a LEDs as a wireless transceiver, and fits in attiny85 along with the application. I do not see how you could achieve this using the actual TasmotaClient. 

I do see you use 1-Wire as an underlying communication protocol. Couldn't [PJDL](https://github.com/gioblu/PJON/blob/master/src/strategies/SoftwareBitBang/specification/PJDL-specification-v5.0.md) be an interesting alternative, considering it has a range of 2000m against 200m of 1-Wire? Up to 255 devices against 20 of 1-Wire? It has a unified implementation with a wider support? It even works using the human body or water ducts as the conductor?

> Anyone can create a Tasmota device driver for PJON, and submit a PR for consideration.

Considering that flashing 2 arduino-compatible devices such as the esps you do use and connect them with a wire is pretty quick, I suspect it may have more sense for you to evaluate its performance running the provided examples. I just opened a feature request to make you aware of the protocol and its ratings. If you find it interesting for your use case I can develop the required device driver to make it work with Tasmota.

Jason2866, 2020-10-16:
We will not implement PJON. You are welcome to provide a PR. Keep in mind the low resources of the ESP82XX. As PJON would be additional to the other features and functions of Tasmota i think the needed flash and iRAM need is too high. If you manage to do in a few kbytes flash and max 2k iRAM usage there is a good chance it will be included as a option.

gioblu, 2020-10-16:
PJON uses around 4k of flash and 46 bytes of ram (yes, bytes) in its more constrained configuration. As mentioned above it is able to operate in an ATtiny85 (8KB ISP flash memory, 512-Byte SRAM) **along with** the application.

Jason2866, 2020-10-16:
Yeah, we count iRAM usage in bytes too 😀. 4kbyte flash usage sounds good too!

gioblu, 2020-10-16:
Ok thank you for your time @Jason2866 will work something for you as soon as possible.

ascillato2, 2020-12-02:
This issue will be auto-closed because there hasn't been any activity for quite a while. Feel free to open a new one if you still experience this problem.

closed 2020-12-02 by ascillato2