arendst/Tasmota #9556: Program Flash Size - changed after OTA

ruzickav, 2020-10-16:
PROBLEM DESCRIPTION
I am using Wemos D1 mini - like this one: https://www.aliexpress.com/item/4001291931302.html?spm=a2g0o.productlist.0.0.1e9b6e51MgqXS3&algo_pvid=a5a71a06-e570-4d88-acf8-66b1ef6dbe80&algo_expid=a5a71a06-e570-4d88-acf8-66b1ef6dbe80-0&btsid=2100bb4716028474184551390ea152&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

The behavior is the same for this version: https://www.aliexpress.com/item/32631693796.html?spm=a2g0o.productlist.0.0.1e9b6e51MgqXS3&algo_pvid=a5a71a06-e570-4d88-acf8-66b1ef6dbe80&algo_expid=a5a71a06-e570-4d88-acf8-66b1ef6dbe80-2&btsid=2100bb4716028474184551390ea152&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

I have flashed this Wemos over USB and than the Program Flash Size in the GUI of Tasmota is 4096kB. Than after OTA (successful)  update the Program Flash Size in the GUI of Tasmota is changed to 1024kB.
Since this time I am not able to OTA update to sensors or displays editions, because those are bigger than free Flash SIze.

REQUESTED INFORMATION
- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Wemos D1 mini
- [ x] Tasmota binary firmware version number used: 8.5.1
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ x] Flashing tools used: Firstly Tasmotizer, than OTA
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`: irelevant
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
13:20:32 MQT: stat/WemosD1MiniAquarium/STATUS = {"Status":{"Module":18,"DeviceName":"WemosD1MiniAquarium","FriendlyName":["WemosD1MiniAquarium"],"Topic":"WemosD1MiniAquarium","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:20:32 MQT: stat/WemosD1MiniAquarium/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://192.168.2.30:9541/data/firmwares/tasmota.bin","RestartReason":"Software/System restart","Uptime":"27T17:10:11","StartupUTC":"2020-09-18T18:10:21","Sleep":50,"CfgHolder":4617,"BootCount":75,"BCResetTime":"2020-03-26T09:39:41","SaveCount":38589,"SaveAddress":"F6000"}}
13:20:32 MQT: stat/WemosD1MiniAquarium/STATUS2 = {"StatusFWR":{"Version":"8.5.0(tasmota)","BuildDateTime":"2020-09-09T11:41:02","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"446/699"}}
13:20:32 MQT: stat/WemosD1MiniAquarium/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WifiH",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00000200","00006000","00000000"]}}
13:20:32 MQT: stat/WemosD1MiniAquarium/STATUS4 = {"StatusMEM":{"ProgramSize":595,"Free":408,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
13:20:32 MQT: stat/WemosD1MiniAquarium/STATUS5 = {"StatusNET":{"Hostname":"WemosD1MiniAquarium","IPAddress":"192.168.2.129","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"DC:4F:22:7E:68:8F","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:20:32 MQT: stat/WemosD1MiniAquarium/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.30","MqttPort":1883,"MqttClientMask":"WemosD1MiniAquarium","MqttClient":"WemosD1MiniAquarium","MqttUser":"vladimir","MqttCount":12,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
13:20:32 MQT: stat/WemosD1MiniAquarium/STATUS7 = {"StatusTIM":{"UTC":"2020-10-16T11:20:32","Local":"2020-10-16T13:20:32","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"08:13","Sunset":"18:57"}}
13:20:32 MQT: stat/WemosD1MiniAquarium/STATUS10 = {"StatusSNS":{"Time":"2020-10-16T13:20:32","COUNTER":{"C1":2051},"DS18B20":{"Id":"030297946216","Temperature":26.3},"TempUnit":"C"}}
13:20:32 MQT: stat/WemosD1MiniAquarium/STATUS11 = {"StatusSTS":{"Time":"2020-10-16T13:20:32","Uptime":"27T17:10:11","UptimeSec":2394611,"Heap":24,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":12,"POWER":"ON","Dimmer":85,"Fade":"ON","Speed":20,"LedTable":"ON","Wifi":{"AP":1,"SSId":"WifiH","BSSId":"E4:8D:8C:4B:2D:8B","Channel":6,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

TO REPRODUCE
- take Wemos D1 mini and flash Tasmota over USB
- check Program Flash Size reported
- update over wifi / OTA
- check Program Flash Size reported

EXPECTED BEHAVIOUR
Expecting that the Program Flash Size stay at 4095kB and can be used for update of FW version

SCREENSHOTS
After USB flash
[image]

After OTA flash from www
[image]

Jason2866, 2020-10-16:
As designed. Tasmota is designed and build for 1M no SPIFFS. 
Since it uses the same firmware for wired and OTA flash there can be no difference in the used flash,
You have used different firmwares or your USB flashing method is not the recommended one.

ruzickav, 2020-10-16:
OK - thanks. Didn't find that info.

But is there any way how to OTA update. OTA from minimal to display is not working - "not enough free space" is loged to console. The sentence is maybe incorrect, not see it right now. But you probably know what I am talking about.... :-)

Jason2866, 2020-10-16:
You have flashed the device most proabably (someday) with a not compatible memory model.
That do often work until the day you want to do a OTA upgrade. The error you see will happen.
Only way is erase flash and flash it with Platformio upload function or use Tasmotizer.

ruzickav, 2020-10-16:
I am flashing only by Tasmotizer with "Erase before flashing" checked....

sfromis, 2020-10-16:
Use the compressed .gz file

ruzickav, 2020-10-16:
"Use the compressed .gz file for OTA upgrade" - the simplest advice is the best one :-) Thank you - this is the right way to get it working. I have never thought that can use zipped file.

closed 2020-10-16 by ruzickav