arendst/Tasmota #9560: APDS-9960 sensor reporting incorrect value for CCT

demian85, 2020-10-16:
### PROBLEM DESCRIPTION
APDS9960 CCT shows incorrect information, I think...

### REQUESTED INFORMATION

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Wemos D1
- [ ] Tasmota binary firmware version number used: 8.5.1
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: Tasmocompiler with `USE_APDS9960` flag enabled.
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
20:23:57 RSL: stat/tasmota_660C89/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
20:23:57 RSL: stat/tasmota_660C89/RESULT = {"Module":{"0":"Generic"}}
20:23:57 RSL: stat/tasmota_660C89/RESULT = {"GPIO0":{"2":"AM2301"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"5":"I2C SCL"},"GPIO13":{"6":"I2C SDA"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```

```
- [ ] Provide the output of this command: `Status 0`:
```
 20:25:50 RSL: stat/tasmota_660C89/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_660C89","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:25:50 RSL: stat/tasmota_660C89/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T20:20:03","StartupUTC":"2020-10-15T23:05:47","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2020-10-12T16:29:10","SaveCount":18,"SaveAddress":"3F9000"}}
20:25:50 RSL: stat/tasmota_660C89/STATUS2 = {"StatusFWR":{"Version":"8.5.1(tasmota)","BuildDateTime":"2020-10-12T15:23:43","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"350/699"}}
20:25:50 RSL: stat/tasmota_660C89/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Garchatel WiFi",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
20:25:50 RSL: stat/tasmota_660C89/STATUS4 = {"StatusMEM":{"ProgramSize":553,"Free":2516,"Heap":26,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F8AC58F","04260001","54B400CD","410013C0","C804FDC1","00004004","00000000"],"Drivers":"1,2,3,4,5,6,8,9,10,18,19,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6,15,17,18,20,27,56"}}
20:25:50 RSL: stat/tasmota_660C89/STATUS5 = {"StatusNET":{"Hostname":"tasmota_660C89-3209","IPAddress":"192.168.0.28","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"DC:4F:22:66:0C:89","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
20:25:50 RSL: stat/tasmota_660C89/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_660C89","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:25:50 RSL: stat/tasmota_660C89/STATUS7 = {"StatusTIM":{"UTC":"2020-10-16T19:25:50","Local":"2020-10-16T20:25:50","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:13","Sunset":"17:57"}}
20:25:50 RSL: stat/tasmota_660C89/STATUS10 = {"StatusSNS":{"Time":"2020-10-16T20:25:50","ANALOG":{"A0":329},"AM2301":{"Temperature":24.6,"Humidity":68.0,"DewPoint":18.3},"APDS9960":{"Red":58,"Green":51,"Blue":57,"Illuminance":33,"CCT":14354,"Proximity":17},"TempUnit":"C"}}
20:25:50 RSL: stat/tasmota_660C89/STATUS11 = {"StatusSTS":{"Time":"2020-10-16T20:25:50","Uptime":"0T20:20:03","UptimeSec":73203,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Garchatel WiFi","BSSId":"CC:2D:E0:A6:11:71","Channel":8,"RSSI":50,"Signal":-75,"LinkCount":2,"Downtime":"0T00:00:05"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```

```

### TO REPRODUCE
See Web UI

### EXPECTED BEHAVIOUR
CCT value is out of range, isn't it? I thought CCT stands for color temperature and should be in the range of ~2000K to ~6500K. Can someone please confirm this is the correct behavior?
Also, can someone please also confirm which are the possible values for RGB? 0 to 255?

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-10-16:
Tasmot shows what it receives. Why do you think it is false?
Did you make tests with the same sensor and a other sketch?

demian85, 2020-10-16:
Sorry but it is hard to understand that value. As I said, isn't color
temperature supposed to be much lower? I don't even know how to interpret
that value. 2000k is warm, 6000k is cool. What the hell is 23000k? I even
got 53000k. Maybe I should find another formula to calculate real
temperature from RGB values.

sfromis, 2020-10-16:
Tasmota does have a somewhat complex calculation in the sensor driver.
https://github.com/arendst/Tasmota/blob/development/tasmota/xsns_27_apds9960.ino#L350

I've actually added a APDS-9960 recently, and was also starting to wonder about the CCT values seen. Both because of extremely large values (Wikipedia lists up to 9500K for lighting/screens and up to 27000K for "clear blue poleward sky"), but also jumps directly between values like around 3-7000K and over 50000K, without me moving or changing lighting. No clear pattern, but I'm unconvinced that numbers are anywhere in the neighborhood of being reasonable.

As the formula found exclusively relies on the RGB sensor values, which look pretty reasonable, I see the formula as the only place things can go bad, and the many constants and signs make the calculation very sensitive to changing signs.

sfromis, 2020-10-16:
Found a similar issue around the these RGB calculations based on another sensor.
https://github.com/adafruit/Adafruit_TCS34725/issues/21
"The detected colours are way off, and I think I know why"

sfromis, 2020-10-16:
Here's a stackexchange thread about these calculations.
https://dsp.stackexchange.com/questions/8949/how-do-i-calculate-the-color-temperature-of-the-light-source-illuminating-an-ima
Looks like the source of the formula. Someone there also found values outside of plausibility.

sfromis, 2020-10-17:
According to the [datasheet](https://cdn.sparkfun.com/assets/learn_tutorials/3/2/1/Avago-APDS-9960-datasheet.pdf), the RGB values are 16-bit numbers, and getting the sensor close to a red bulb gives me red channel readings over 30000. I'm wondering if the RGB numbers are compatible with the expectations of the CCT formula.

sfromis, 2020-10-17:
The formula can produce extreme values for the slope calculation (variable n) when yc becomes close to 0.1858 leading to a division with a near zero value.

Seeing the formula made me wonder why the FastPrecisePowf function is used with very small integer exponents, when it would seem much simpler (faster and less bytes I guess) to square or cube the n value via simple multiplications.

sfromis, 2020-10-17:
An example of wacky results:
```................... Red Green Blue Illuminance   CCT Proximity
2020-10-17 19:32:57  14    11   14           9 43286        79
2020-10-17 19:37:57  14    11   15           9     0        87```
The last one gave a "n" value of -688.92 when replicating the calculation. While the RGB values look pretty "boring", they are the source of bad output. I suppose that having red and blue higher than green (purple-ish light according to the sensor)  is outside of the range where CCT is useful, as such lights should only produce white-ish light sliding on the Kelvin scale.

demian85, 2020-10-17:
Isn't there a standard formula for CCT calculation from RGB? Shouldn't we
scale RGB values to 0-255 first?
My use case is to change a smart bulb color temperature based on the
reported CCT from the sensor in my garden. So I'm looking something in the
range of 3000-6500K.

demian85, 2020-10-17:
It would be nice if someone could also confirm which is the value range for red, green, and blue. The datasheet says it is 16 bits in total, so I'm assuming 65535/3 is the maximum per channel? Thanks.

sfromis, 2020-10-17:
As previously mentioned, I saw the RGB channels max out at 37889. The question may be what values the "borrowed" formula expects, but in principle, the CCT result should only be affected by the relationship between the channels, irrespective of their scaling.

demian85, 2020-10-18:
Sorry but I don't understand how each channel max out at 37889 when 16 bit should be 65535. It should be 21845 per channel :/

sfromis, 2020-10-19:
Well, 16 bit are needed to represent the number 37889, and there are 16 bits for each of the 3 color channels. If 16 bit would have to contain all 3 channels, that would not allow values up to 21845 per channel, as 5 bit only allows for values from 0 to 31 (one of the 3 channels could get 6 bit, for values from 0 to 63). The number 37889 is not a function of the binary representation, but just what the sensor can measure. A google search reveals that other users of the sensor also see values up to 37889. In short, nothing weird here.

The weirdness comes from the formula which very often generates unusable values, which is not surprising when seeing the details of how it works. It may be a fine approximation under certain conditions, but I'm not aware of what those conditions are, or how the APDS-9960 sensor matches those conditions.

sfromis, 2020-10-19:
Worth noting that the Wikipedia article about [color temperature](https://en.wikipedia.org/wiki/Color_temperature#Approximation) has this note in the section about approximation, just before calculations looking a lot like the formula in the driver:

*If a narrow range of color temperatures is considered—those encapsulating daylight being the most practical case—one can approximate the Planckian locus in order to calculate the CCT in terms of chromaticity coordinates.*

There may be a high risk of 1) that the light measured is outside of the "narrow range", like not being similar to daylight, and 2) that the sensor may not be precise enough to deliver RGB numbers matching that narrow range. As I understand color temperature, it makes most sense when looking at blackbody radiation, or something similar to that. From the same Wikipedia article:

*The color temperature of a light source is the temperature of an ideal black-body radiator that radiates light of a color comparable to that of the light source.*

Such a blackbody does not exist for most of the chromaticity space. Maybe the CCT calculation is of questionable utility, compared to the logical expectation of getting a plausible number in the presence of "some light"....

demian85, 2020-12-05:
So. Is this formula going to be fixed or removed? I don't want to rely on
useless values. Are they?
Can someone please share the right way to calculate color temperature using
approximation? Thanks.

sfromis, 2020-12-05:
The concept of "color temperature" does not make sense for arbitrary values of RGB. Therefore, unusable results are likely to appear very often. There is simply no "right way" most of the time, when it is not a light source similar to black body radiation.

sfromis, 2020-12-05:
Most of the time, just ignore the color temperature approximation.

closed 2020-12-12 by github-actions[bot]