arendst/Tasmota #9561: Can't connect to WiFi 802.11b

Wiednitz, 2020-10-17:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-10-17:
Sorry, not under the control of Tasmota. Tasmota uses for connecting to wifi Arduino ESP8266 core (which uses function from the Espressif closed source NonOS SDK). Tasmota uses latest actual Arduino ESP8266 version (which fixes bugs and security issues). Older build of Tasmota uses older Arduion ESP8266 so there is the possibilty the behave different.
In general there is no problem to connect to 802.11g networks. There are incompabilties between Espressif and some brands of APs. You can try to change settings in your APs. 
The issue you have can **not** be fixed in Tasmota.

sfromis, 2020-10-17:
With the minimal bandwidth needs of Tasmota, "only" using 802.11b should not cause any real-world trouble. In my network, the ESP8266 devices running Tasmota connects to 802.11n which is the "more current" 2.4GHz variant, but do not get near the possible speeds of that.

Wiednitz, 2020-10-17:
yes of course WiFi b is completely sufficient, I was just amazed about this fact

Jason2866, 2020-10-17:
Please close since your question has been answered.
Thx.

closed 2020-10-17 by Wiednitz