arendst/Tasmota #9565: Tasmota Scripting: Can't get RESULT into >E variable

tthk, 2020-10-17:
### PROBLEM DESCRIPTION
>E section of Tasmota Scripting can't assign RESULT data to variable.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Gosund SW1
- [ ] Tasmota binary firmware version number used: 9.0.0.2 (development branch)
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: VS Code PlatformIO
- [X] Flashing tools used: espytool
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
16:05:57 CMD: Backlog Template; Module; GPIO 255
16:05:57 MQT: stat/tasmota_C4F0F3/RESULT = {"NAME":"Gosund Dimmer","GPIO":[1,3200,1,3232,32,0,1,1,320,576,416,1,1,0],"FLAG":0,"BASE":18}
16:05:57 >E
16:05:57 pIn:na
16:05:57 MQT: stat/tasmota_C4F0F3/RESULT = {"Module":{"0":"Gosund Dimmer"}}
16:05:57 >E
16:05:57 pIn:na
16:05:57 MQT: stat/tasmota_C4F0F3/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3232":"Serial Rx"},"GPIO4":{"32":"Button1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"320":"Led_i1"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"416":"PWM1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
16:05:57 >E
16:05:57 pIn:na
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```
16:07:07 CMD: Status 0
16:07:07 MQT: stat/tasmota_C4F0F3/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_C4F0F3","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:07:07 MQT: stat/tasmota_C4F0F3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin","RestartReason":"Power On","Uptime":"0T00:03:14","StartupUTC":"2020-10-17T23:03:53","Sleep":50,"CfgHolder":4619,"BootCount":68,"BCResetTime":"2020-10-15T04:04:42","SaveCount":475,"SaveAddress":"F6000"}}
16:07:07 MQT: stat/tasmota_C4F0F3/STATUS2 = {"StatusFWR":{"Version":"9.0.0.2(tasmota)","BuildDateTime":"2020-10-17T03:55:17","Boot":31,"Core":"2_7_4_3","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"421/699"}}
16:07:07 MQT: stat/tasmota_C4F0F3/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["PureSynergy",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00C08009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
16:07:07 MQT: stat/tasmota_C4F0F3/STATUS4 = {"StatusMEM":{"ProgramSize":614,"Free":388,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","87DAE797","043E8001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
16:07:07 MQT: stat/tasmota_C4F0F3/STATUS5 = {"StatusNET":{"Hostname":"tasmota_C4F0F3-4339","IPAddress":"10.0.253.36","Gateway":"10.0.0.1","Subnetmask":"255.255.0.0","DNSServer":"10.0.0.4","Mac":"C8:2B:96:C4:F0:F3","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
16:07:07 MQT: stat/tasmota_C4F0F3/STATUS6 = {"StatusMQT":{"MqttHost":"nile.puresynergy.com","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C4F0F3","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
16:07:07 MQT: stat/tasmota_C4F0F3/STATUS7 = {"StatusTIM":{"UTC":"2020-10-17T23:07:07","Local":"2020-10-17T16:07:07","StartDST":"2020-03-08T02:00:00","EndDST":"2020-11-01T03:00:00","Timezone":99,"Sunrise":"07:17","Sunset":"18:26"}}
16:07:07 MQT: stat/tasmota_C4F0F3/STATUS10 = {"StatusSNS":{"Time":"2020-10-17T16:07:07"}}
16:07:07 MQT: stat/tasmota_C4F0F3/STATUS11 = {"StatusSTS":{"Time":"2020-10-17T16:07:07","Uptime":"0T00:03:14","UptimeSec":194,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":85,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"PureSynergy","BSSId":"42:37:86:25:18:9E","Channel":4,"RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
16:07:41 CMD: weblog 4
16:07:42 MQT: stat/tasmota_C4F0F3/RESULT = {"WebLog":4}
16:07:42 >E
16:07:42 pIn:na
16:07:42 CFG: Saved to flash at F5, Count 476, Bytes 4096
16:07:46 APP: Button1 multi-press 1
16:07:46 >E
16:07:46 pIn:na
16:07:46 SRC: Button
16:07:46 >E
16:07:46 pIn:na
16:07:46 MQT: stat/tasmota_C4F0F3/RESULT = {"POWER":"ON"}
16:07:46 MQT: stat/tasmota_C4F0F3/POWER = ON
16:07:47 CFG: Saved to flash at F4, Count 477, Bytes 4096
16:07:59 WIF: Checking connection...

```

### TO REPRODUCE

Do anything like pressing button to generate >E message.  Script code should pull in RESULT data.

### EXPECTED BEHAVIOUR
In given Script code, pIn variable should be set to value in RESULT data, like "OFF" or "ON"

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2020-10-18:
scripting reads JSON string objects like rules does. the only difference to rules is that upper, lower case matters.

you must read in numbers into number variables and strings into string variables.

for convenience the strings "ON" and "OFF" are converted into 0 and 1 and can be stored into an number variable

e.g. JSON objects

16:07:46 MQT: stat/tasmota_C4F0F3/RESULT = {"POWER":"ON"}

07:27:25 MQT: tele/sonoff/SENSOR = {"Time":"2020-10-18T07:27:25","ENERGY":{"TotalStartTime":"2020-10-13T13:36:11","Total":0.347,"Yesterday":0.033,"Today":0.008,"Period":1,"Power":11,"ApparentPower":18,"ReactivePower":15,"Factor":0.59,"Voltage":228,"Current":0.079}}

\>D
p=0
et=0
tim=""
\>E
; actually a string "ON" but may be put into number
p=POWER
print %p%
\>T
; a number
et=ENERGY#Total
print %et%
; a string
tim= ENERGY#TotalStartTime
print %tim%

tthk, 2020-10-18:
Thanks for the response, @gemu2015.

I'm still not getting RESULT data for POWER, however, I can get RESULT Command field (generated by entering bad command):

Script:
```
>D 100
IP=10.0.253.36
p=0
ps=""
cmd=""

>E
dp0
print >E
p=0
ps="na"
cmd="na"
p=POWER
ps=POWER
cmd=Command
print p:%p% ps:"%ps%" cmd:"%cmd%"

```

Log:
```
01:45:17 >E
01:45:17 p:0 ps:"na" cmd:"na"
01:45:17 >E
01:45:17 var not found: POWER
01:45:17 p:0 ps:"na" cmd:"na"
01:45:17 MQT: stat/tasmota_C4F0F3/RESULT = {"POWER":"OFF"}
01:45:17 MQT: stat/tasmota_C4F0F3/POWER = OFF
01:45:22 CMD: sdfsd
01:45:22 MQT: stat/tasmota_C4F0F3/RESULT = {"Command":"Unknown"}
01:45:22 >E
01:45:22 p:0 ps:"na" cmd:"Unknown"
01:45:59 >E
01:45:59 p:0 ps:"na" cmd:"na"
01:45:59 >E
01:45:59 var not found: POWER
01:45:59 p:0 ps:"na" cmd:"na"
01:45:59 MQT: stat/tasmota_C4F0F3/RESULT = {"POWER":"ON"}
01:45:59 MQT: stat/tasmota_C4F0F3/POWER = ON
```

POWER field extraction and assigment to p(int) and ps(string) does nothing, even when RESULT had POWER field.
Command field extraction and assigment to cmd(string) works fine, however.

Do I need to compile with some specific flag to extract POWER (and possibly other fields) from RESULT?

gemu2015, 2020-10-18:
you are right. it used to work in older versions. maybe a regression vom JSMN?
i'll investigate.

gemu2015, 2020-10-18:
OK, has been a long time i coded this.

the >E section is called by the POWER function but with no JSON data.
the same is true for the >P section  (#define SCRIPT_POWER_SECTION) which is only called on power events

other EVENTs provide a valid JSON however in >E

the solution is to read power state from the variable pwr[x]  x = 1..N

tthk, 2020-10-19:
Thanks for the response, and especially for a super scripting engine @gemu2015.

POWER not providing JSON make sense since we can get it with pow[x].  It was just unfortunate that I used this as basic test case to test >E RESULT data gathering functionality.

I'll close this issue.

closed 2020-10-19 by tthk