arendst/Tasmota #9566: Treatlife DS01 White LED always off after 8.1.0

SylvainGa, 2020-10-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After an upgrade from 8.1.0 to at least 8.1.0.5 and above, the white LED that is supposed to be ON when the dimmer output is OFF isn't working. The LED always stay OFF.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Treatlife DS01
- [ ] Tasmota binary firmware version number used: 8.5.1
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: FlashESP8266.exe
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
23:18:20 MQT: lumieresalleamanger/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
23:18:21 MQT: lumieresalleamanger/stat/RESULT = {"Module":{"54":"Tuya MCU"}}
23:18:21 MQT: lumieresalleamanger/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"107":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"108":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
No rules

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
23:19:41 MQT: lumieresalleamanger/stat/STATUS = {"Status":{"Module":54,"DeviceName":"LumiereSalleAManger","FriendlyName":["LumiereSalleAManger"],"Topic":"lumieresalleamanger","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:19:41 MQT: lumieresalleamanger/stat/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:30:27","StartupUTC":"2020-10-18T02:49:14","Sleep":50,"CfgHolder":4617,"BootCount":27,"BCResetTime":"2020-10-16T15:29:49","SaveCount":333,"SaveAddress":"FB000"}}
23:19:41 MQT: lumieresalleamanger/stat/STATUS2 = {"StatusFWR":{"Version":"8.5.1(tasmota)","BuildDateTime":"2020-10-02T10:09:47","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"430/699"}}
23:19:41 MQT: lumieresalleamanger/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":1,"LogHost":"192.168.2.84","LogPort":514,"SSId":["BELL776","Sylvain_HS"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00CA8009","2805C8000100060000005A00000000000000","00000208","00006000","00000000"]}}
23:19:41 MQT: lumieresalleamanger/stat/STATUS4 = {"StatusMEM":{"ProgramSize":596,"Free":404,"Heap":21,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
23:19:41 MQT: lumieresalleamanger/stat/STATUS5 = {"StatusNET":{"Hostname":"lumieresalleamanger-2034","IPAddress":"192.168.2.192","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"48:3F:DA:42:67:F2","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
23:19:41 MQT: lumieresalleamanger/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.84","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4267F2","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
23:19:41 MQT: lumieresalleamanger/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-10-18T03:19:41","Local":"2020-10-17T23:19:41","StartDST":"2020-03-08T02:00:00","EndDST":"2020-11-01T02:00:00","Timezone":99,"Sunrise":"02:14","Sunset":"12:55"}}
23:19:41 MQT: lumieresalleamanger/stat/STATUS10 = {"StatusSNS":{"Time":"2020-10-17T23:19:41"}}
23:19:41 MQT: lumieresalleamanger/stat/STATUS11 = {"StatusSTS":{"Time":"2020-10-17T23:19:41","Uptime":"0T00:30:27","UptimeSec":1827,"Heap":21,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":24,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"BELL776","BSSId":"A8:9A:93:FD:CC:2E","Channel":11,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
Not relevant

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flash 8.1.0 onto a DS01, reset 1, unplug/replug DS01, configure Wifi, set to TuyaMCU (54). See the white LED light is ON when the dimmer output is OFF and OFF when the dimmer output is ON
Flash 8.1.0.5 and above onto a DS01, reset 1, unplug/replug DS01. configure Wifi, set to TuyaMCU (54). See the white LED light remains OFF, whether the dimmer output is OFF or ON.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The white LED should follow the reverse of the output power. When there is power, the LED is OFF, when there is no power, the LED is ON.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
Not applicable

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I had a copy of 8.1.0.5 so that's why I could try with that. So I know it was broken at least at 8.1.0.5 but it may have started in a previous release after 8.1.0 which is the latest firmware I have that I know it's working fine. Changes between 8.1.0 and 8.1.0.5 might not be that intensive so it could be a good lead to figure out what broke it.

See issues 8577 and 8117 where some parts of them relate to this issue but they were closed without this issue being resolved.

**(Please, remember to close the issue when the problem has been addressed)**

mcarson75, 2020-10-29:
I had a similar problem with a DS01 and 8.5.0.  The white LED was off and no matter what I did I could not get it to come on.  I reset config and started from scratch with 8.5.0, and the LED started working.  I was playing with settings and was worried I had inadvertently turned it off, but the LED had never come on since I had set up Tasmota until I reset it so I don't think it was settings related.  (I would actually prefer if it was so I could turn the light off on command)

SylvainGa, 2020-10-29:
#response_container_BBPPID{font-family: initial; font-size:initial; color: initial;} Not sure what I did (did a bunch of 8.1 to 8.5.1 to 8.1 to back to 8.5.1) but all my 4 DS01 white LED are now working. Not sure if you had to open them up to flash wired but if you did, you therefore know how to solder so it's a small modification (cut a trace and put a small wire between an WYES3S output and the led) to have the WYES3S control that LED instead of the MCU. I'm thinking of doing that when I have a chance.                                                                                                                                              Envoyé via Boîte de réception BlackBerry Hub+ pour Android                                                                                                                                                                           De : notifications@github.comEnvoyé : 28 octobre 2020 20 h 45À : Tasmota@noreply.github.comRépondre à : reply@reply.github.comCc : gagnon.sylvain@gmail.com; author@noreply.github.comObjet : Re: [arendst/Tasmota] Treatlife DS01 White LED always off after 8.1.0 (#9566)  
I had a similar problem with a DS01 and 8.5.0.  The white LED was off and no matter what I did I could not get it to come on.  I reset config and started from scratch with 8.5.0, and the LED started working.  I was playing with settings and was worried I had inadvertently turned it off, but the LED had never come on since I had set up Tasmota until I reset it so I don't think it was settings related.  (I would actually prefer if it was so I could turn the light off on command)

ascillato2, 2020-11-16:
> Not sure what I did (did a bunch of 8.1 to 8.5.1 to 8.1 to back to 8.5.1) but all my 4 DS01 white LED are now working.

Great to know that now they are working.

closed 2020-11-16 by ascillato2