arendst/Tasmota #9577: First value of RGB color gets set to 0

tatsumara, 2020-10-19:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Sending an RGB value to Color1 results in the first value getting displayed as 0 in the "Color" property of the Result, (for example [66, 255, 62] -> [0, 255, 62, 0]). The light itself displays the right color, but the value is wrong which screws with my MQTT program.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Teckin SB50
- [x] Tasmota binary firmware version number used: 8.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
{"NAME":"Teckin SB50","GPIO":[0,0,0,0,40,0,0,0,38,39,37,0,0],"FLAG":0,"BASE":18}
{"Module":{"0":"Teckin SB50"}}
{"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"40":"PWM4"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"38":"PWM2"},"GPIO13":{"39":"PWM3"},"GPIO14":{"37":"PWM1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Sending an RGB value to Color1

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Color property displays correct value.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-10-19:
Did you follow the syntax specified for the [Color](https://tasmota.github.io/docs/Commands/#color) command?
Like `Color 240,160,80` or with hex values `Color f0a050`
Extraneous characters like [] or blank is likely to cause trouble.
And you did not tell what exact string your MQTT program can't handle.

tatsumara, 2020-10-19:
This is not the issue. As far as I can tell the square brackets do not cause any trouble, since the color gets set correctly, but the Color property in the RESULT MQTT is displaying a false value.

sfromis, 2020-10-19:
Why do you not give info about how to actually replicate the issue. How **exactly** is the Color1 command you use.
And ... If you use square brackets in the command, it is a syntax error to be corrected.

closed 2020-10-19 by tatsumara