arendst/Tasmota #9582: Can't compile with USE_MQTT_TLS

poofyteddy, 2020-10-19:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Error while compiling
```
Repo/tasmota/xdrv_02_mqtt.ino:804:168: error: operands to ?: have different types 'int' and 'char*'
       strlcpy(fingerprint, (SC_CLEAR == Shortcut()) ? "" : (SC_DEFAULT == Shortcut()) ? (1 == XdrvMailbox.index) ? MQTT_FINGERPRINT1 : MQTT_FINGERPRINT2 : XdrvMailbox.data, sizeof(fingerprint));
 ```
with `#define USE_MQTT_TLS` in user_config_override.h

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: latest dev
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: Vscode platformio
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
#define USE_MQTT_TLS
compile env:tasmota

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-10-20:
As the compile error message tries to explain the fingerprint types have changed. That happend the day before yesterday. See changelog and my_user_config.h what you need to do to make compile possible again.

s-hadinger, 2020-10-20:
I realize we don't have a channel to announce breaking changes in the code.

#9570 changed the format for compile time fingerprints, which has the side effect of saving 232 bytes for all Tasmota versions, even the ones that will never use TLS nor fingerprints.

poofyteddy, 2020-10-20:
i am sorry but i am not quite sure i understand :(
I don't override MQTT_FINGERPRINT(1-2), and my `my_user_config.h` is the one from the repo
```
#define MQTT_FINGERPRINT1      0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00  // [MqttFingerprint1] (auto-learn)
#define MQTT_FINGERPRINT2      0xDA,0x39,0xA3,0xEE,0x5E,0x6B,0x4B,0x0D,0x32,0x55,0xBF,0xEF,0x95,0x60,0x18,0x90,0xAF,0xD8,0x07,0x09  // [MqttFingerprint2] (invalid)
```
The change log say `#define MQTT_FINGERPRINT from string to hexnumbers (#9570)` and as far as i know, what is above isn't a string (no quote)

I am obviously missing something very simple, but can't get what ....

s-hadinger, 2020-10-21:
Oh sorry, you are right. Let me revisit this part.

closed 2020-10-21 by s-hadinger

poofyteddy, 2020-10-21:
Thank you for the fix :)

PR #9595 (2020-10-21): Fix TLS fingerprint compilation #9582
## Description:

Fix fingerprint compilation after changes in #9570

**Related issue (if applicable):** fixes #9582

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.3
  - [ ] The code change is tested and works on core ESP32 V.1.12.2
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

_NO [... cut]