arendst/Tasmota #9583: Tuya dimmer doesn't send MQT message on manual brightness change

MagicNoSpam, 2020-10-19:
### PROBLEM DESCRIPTION
Using the latest Tasmota release on a Treatlife Tuya Smart Dimmer. Changing the brightness slider in the web GUI results in the expected MQT messages on the console. However when changing the brightness on the actual switch it does not send the MQT message.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Treatlife Tuya Smart Dimmer
- [ ] Tasmota binary firmware version number used: 8.5.1
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: FTDI
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
22:03:24 SRC: Backlog
22:03:24 CMD: Group 0, Index 1, Command "GPIO", Data "255"
22:03:25 MQT: stat/tasmota_A0FBF9/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"107":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"108":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here: No rules

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
22:05:04 CMD: status 0
22:05:04 SRC: WebConsole from 192.168.2.20
22:05:04 CMD: Group 0, Index 1, Command "STATUS", Data "0"
22:05:04 MQT: stat/tasmota_A0FBF9/STATUS = {"Status":{"Module":54,"DeviceName":"Tuya-Dimmer1","FriendlyName":["Tuya-Dimmer1"],"Topic":"tasmota_A0FBF9","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
22:05:04 MQT: stat/tasmota_A0FBF9/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:25:17","StartupUTC":"2020-10-19T20:39:47","Sleep":50,"CfgHolder":4617,"BootCount":22,"BCResetTime":"2020-10-18T19:01:52","SaveCount":162,"SaveAddress":"F4000"}}
22:05:04 MQT: stat/tasmota_A0FBF9/STATUS2 = {"StatusFWR":{"Version":"8.5.1(tasmota)","BuildDateTime":"2020-10-02T10:09:47","Boot":7,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"387/699"}}
22:05:04 MQT: stat/tasmota_A0FBF9/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["OSIRIS1",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008029","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
22:05:04 MQT: stat/tasmota_A0FBF9/STATUS4 = {"StatusMEM":{"ProgramSize":596,"Free":404,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
22:05:04 MQT: stat/tasmota_A0FBF9/STATUS5 = {"StatusNET":{"Hostname":"tasmota_A0FBF9-7161","IPAddress":"192.168.2.162","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.12","Mac":"E0:98:06:A0:FB:F9","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
22:05:04 MQT: stat/tasmota_A0FBF9/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.12","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A0FBF9","MqttUser":"olaf","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
22:05:04 MQT: stat/tasmota_A0FBF9/STATUS7 = {"StatusTIM":{"UTC":"2020-10-19T21:05:04","Local":"2020-10-19T22:05:04","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:17","Sunset":"17:52"}}
22:05:04 MQT: stat/tasmota_A0FBF9/STATUS10 = {"StatusSNS":{"Time":"2020-10-19T22:05:04"}}
22:05:04 MQT: stat/tasmota_A0FBF9/STATUS11 = {"StatusSTS":{"Time":"2020-10-19T22:05:04","Uptime":"0T00:25:17","UptimeSec":1517,"Heap":23,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":50,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"OSIRIS1","BSSId":"F8:08:4F:FD:28:8E","Channel":1,"RSSI":100,"Signal":-45,"LinkCount":1,"Downtime":"0T00:00:09"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
22:09:27 {"TuyaReceived":{"Data":"55AA030700080202000400000096AF","Cmnd":7,"CmndData":"0202000400000096","DpType2Id2":150,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000096"}}}
22:09:27 TYA: fnId=21 is set for dpId=2
22:09:27 TYA: RX value 150 from dpId 2 
22:09:27 {"TuyaReceived":{"Data":"55AA03070008020200040000011F39","Cmnd":7,"CmndData":"020200040000011F","DpType2Id2":287,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000011F"}}}
22:09:27 TYA: fnId=21 is set for dpId=2
22:09:27 TYA: RX value 287 from dpId 2 
22:09:27 {"TuyaReceived":{"Data":"55AA0307000802020004000001A9C3","Cmnd":7,"CmndData":"02020004000001A9","DpType2Id2":425,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000001A9"}}}
22:09:27 TYA: fnId=21 is set for dpId=2
22:09:27 TYA: RX value 425 from dpId 2 

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1.) flashed latest release of Tasmota onto Treatlife Smart Dimmer
2.) Configuration -> Configure Module -> TuyaMCU (54)
3.) in console entered: 
TuyaMCU 21,2
dimmerrange 1,1000
weblog 4
4.) Slide finger on smart dimmer to change brightness level. Light gets adjusted but no brightness level gets reported via MQT message

### EXPECTED BEHAVIOUR
When using the manual slider on the dimmer I expect an MQT update message similar to this:
MQT: stat/tasmota_A0FBF9/RESULT = {"POWER":"ON","Dimmer":50}

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
This issue is very similar to the solved issue #8498. Looks like it's not solved for the Treatlife dimmer
Btw: A big thank you for creating Tasmota - It's awesome and I used it on several devices, integrated to HA

**(Please, remember to close the issue when the problem has been addressed)**

ihavenonick, 2020-10-20:
the same here with Tasmota 9.0.0.2

effelle, 2020-10-20:
When you press the button/use the slider on the device are you sure the device uses the same dpid? 
Also please enable the optional topic fo Tuya with `TuyaSend9` and paste here. Thanks.

MagicNoSpam, 2020-10-20:
Thanks effelle for looking into this.
Yes, it's also dpid 2. Here is the result of operating the slider on the dimmer after TuyaSend9:
18:31:30 CMD: tuyasend9
18:31:30 TYA: TuyaMCU Stat Topic enabled
18:31:30 MQT: stat/tasmota_A0FBF9/RESULT = {"TuyaSend":"Done"}
18:32:04 MQT: stat/tasmota_A0FBF9/TUYASEND2 = 2,150
18:32:04 MQT: stat/tasmota_A0FBF9/TUYASEND1 = 1,1
18:32:04 MQT: stat/tasmota_A0FBF9/RESULT = {"POWER":"ON"}
18:32:04 MQT: stat/tasmota_A0FBF9/POWER = ON (retained)
18:32:05 MQT: stat/tasmota_A0FBF9/TUYASEND2 = 2,150
18:32:05 MQT: stat/tasmota_A0FBF9/TUYASEND2 = 2,257
18:32:05 MQT: stat/tasmota_A0FBF9/TUYASEND2 = 2,425
18:32:05 MQT: stat/tasmota_A0FBF9/TUYASEND2 = 2,703

effelle, 2020-10-20:
I hate this type of bugs. On one configuration is working fine, on another test device is not. 
On my todo list for the next update, thanks for reporting.

effelle, 2020-10-21:
@MagicNoSpam please repeat the very same test with weblog 4 and paste the result here, thanks!

cprussin, 2020-10-22:
@effelle same issue here.  I believe this is what you're looking for:

```
09:04:12 {"TuyaReceived":{"Data":"55AA0307000802020004000001D1EB","Cmnd":7,"CmndData":"02020004000001D1","DpType2Id2":465,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000001D1"}}}
```

effelle, 2020-10-22:
No I need to see a full log with time stamps when tuyasend9 is enabled.

cprussin, 2020-10-22:
Apologies on the misunderstanding, here you go:

```
12:45:16 CMD: TuyaSend9
12:45:16 TYA: TuyaMCU Stat Topic enabled
12:45:16 MQT: stat/entertainment-console-lights/RESULT = {"TuyaSend":"Done"}
12:45:31 MQT: stat/entertainment-console-lights/TUYASEND2 = 2,158
12:45:33 MQT: stat/entertainment-console-lights/TUYASEND2 = 2,237
12:45:36 MQT: stat/entertainment-console-lights/TUYASEND1 = 1,1
12:45:36 MQT: tele/entertainment-console-lights/STATE = {"Time":"2020-10-22T12:45:36","Uptime":"0T00:02:17","UptimeSec":137,"Heap":19,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Dimmer":10,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Centar","BSSId":"1C:F2:9A:8E:AD:F2","Channel":1,"RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:05"}}
12:45:36 MQT: stat/entertainment-console-lights/RESULT = {"POWER":"ON"}
12:45:36 MQT: stat/entertainment-console-lights/POWER = ON
12:45:36 MQT: stat/entertainment-console-lights/TUYASEND2 = 2,237
12:45:37 MQT: stat/entertainment-console-lights/TUYASEND2 = 2,158
12:45:39 MQT: stat/entertainment-console-lights/TUYASEND2 = 2,79
```

I pushed a few buttons after enabling tuyasend9, I hope that gives you enough info.

cprussin, 2020-10-22:
d'oh sorry, I forgot I had just dropped my weblog to 2 before I ran that.  Here's the log after I moved weblog back to 4:

```
12:45:16 CMD: TuyaSend9
12:45:16 TYA: TuyaMCU Stat Topic enabled
12:45:16 MQT: stat/entertainment-console-lights/RESULT = {"TuyaSend":"Done"}
12:45:31 MQT: stat/entertainment-console-lights/TUYASEND2 = 2,158
12:45:33 MQT: stat/entertainment-console-lights/TUYASEND2 = 2,237
12:45:36 MQT: stat/entertainment-console-lights/TUYASEND1 = 1,1
12:45:36 MQT: tele/entertainment-console-lights/STATE = {"Time":"2020-10-22T12:45:36","Uptime":"0T00:02:17","UptimeSec":137,"Heap":19,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Dimmer":10,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Centar","BSSId":"1C:F2:9A:8E:AD:F2","Channel":1,"RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:05"}}
12:45:36 MQT: stat/entertainment-console-lights/RESULT = {"POWER":"ON"}
12:45:36 MQT: stat/entertainment-console-lights/POWER = ON
12:45:36 MQT: stat/entertainment-console-lights/TUYASEND2 = 2,237
12:45:37 MQT: stat/entertainment-console-lights/TUYASEND2 = 2,158
12:45:39 MQT: stat/entertainment-console-lights/TUYASEND2 = 2,79
12:47:44 CMD: WebLog 4
12:47:44 MQT: stat/entertainment-console-lights/RESULT = {"WebLog":4}
12:47:44 CFG: Saved to flash at F4, Count 48, Bytes 4096
12:47:46 WIF: Checking connection...
12:47:46 {"TuyaReceived":{"Data":"55AA031C00001E","Cmnd":28}}
12:47:46 TYA: RX unknown command
12:47:53 {"TuyaReceived":{"Data":"55AA03070005010100010011","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
12:47:53 MQT: stat/entertainment-console-lights/TUYASEND1 = 1,0
12:47:53 TYA: fnId=11 is set for dpId=1
12:47:53 TYA: RX Relay-1 --> MCU State: Off Current State:On
12:47:53 SRC: Switch
12:47:53 MQT: tele/entertainment-console-lights/STATE = {"Time":"2020-10-22T12:47:53","Uptime":"0T00:04:34","UptimeSec":274,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":99,"MqttCount":1,"POWER":"OFF","Dimmer":10,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Centar","BSSId":"1C:F2:9A:8E:AD:F2","Channel":1,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:05"}}
12:47:53 MQT: stat/entertainment-console-lights/RESULT = {"POWER":"OFF"}
12:47:53 MQT: stat/entertainment-console-lights/POWER = OFF
12:47:53 {"TuyaReceived":{"Data":"55AA03070008020200040000004F68","Cmnd":7,"CmndData":"020200040000004F","DpType2Id2":79,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000004F"}}}
12:47:53 MQT: stat/entertainment-console-lights/TUYASEND2 = 2,79
12:47:53 TYA: fnId=21 is set for dpId=2
12:47:53 TYA: RX value 79 from dpId 2 
12:47:53 CFG: Saved to flash at FB, Count 49, Bytes 4096
12:47:54 TYA: Send "55aa00000000ff"
12:47:54 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
12:47:54 TYA: Heartbeat
12:47:55 {"TuyaReceived":{"Data":"55AA03070008020200040000009EB7","Cmnd":7,"CmndData":"020200040000009E","DpType2Id2":158,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000009E"}}}
12:47:55 MQT: stat/entertainment-console-lights/TUYASEND2 = 2,158
12:47:55 TYA: fnId=21 is set for dpId=2
12:47:55 TYA: RX value 158 from dpId 2 
12:47:56 {"TuyaReceived":{"Data":"55AA0307000802020004000000ED06","Cmnd":7,"CmndData":"02020004000000ED","DpType2Id2":237,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000000ED"}}}
12:47:56 MQT: stat/entertainment-console-lights/TUYASEND2 = 2,237
12:47:56 TYA: fnId=21 is set for dpId=2
12:47:56 TYA: RX value 237 from dpId 2 
```

effelle, 2020-10-22:
Thanks for reporting.
Got it and fixed, will be available tomorrow on dev channel.

closed 2020-10-23 by arendst

MagicNoSpam, 2020-10-23:
I confirm that the issue is fixed. 
Thanks for this very fast turnaround!

PR #9606 (2020-10-23): Fix mqtt topic for dimmers
## Description:

Fix topic regression for dimmers levels when value is set using phisical buttons/sliders.

**Related issue (if applicable):** fixes #9583

## Checklist:
  - [X] The pull request is done against the latest dev branch
  - [X] Only relevant files were touched
  - [X] Only one feature/fix was added per PR.
  - [X] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.5
  - [ ] The code change is tested and works on core ESP32 V.1.12.2
  - [X] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor- [... cut]