arendst/Tasmota #9587: Web Admin Password Requirements

shravan2x, 2020-10-20:
### PROBLEM DESCRIPTION
The docs are not very clear about web admin password requirements:
1. What is the maximum password length? 32 seems to work but 64 doesn't. Other users have had trouble with this in the past #8720.
2. The UI doesn't throw an error when the password contains special characters. If I accidentally entered a special character how do I log in?

(I managed to reset my password via MQTT but I think this info should be present in the docs)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Tuya Light
- [x] Tasmota binary firmware version number used: 8.5.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Esptool.py
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Set a password of 64 characters, or with special characters.

### EXPECTED BEHAVIOUR
UI displays an error or saves password correctly.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-10-20:
Diving in currently as I indeed was under the impression that any password would do as long as it fits the 600 character pool.

Just found out it doesn't....

Seems to be an issue in the Arduino core library. Investigating...

arendst, 2020-10-21:
The max webpassword length seems to be related to the Arduino Core Base64 character encoding/decoding default chars per line size. It is also related to the username size. The total size of username + password is 52 characters.

As the current default (and fixed) administrator password is ``admin`` being 5 chars, the maximum password length is 47 characters.

arendst, 2020-10-21:
See Arduino Core follow-up here https://github.com/esp8266/Arduino/issues/7664

Jason2866, 2020-10-21:
Fixed with PR #9596  Give it a try

closed 2020-10-22 by ascillato2