arendst/Tasmota #9592: Including Smart Meter Interface breaks hardware serial bridge (at least receiving).

TheChatty, 2020-10-20:
### PROBLEM DESCRIPTION
[Compiling SMI](https://tasmota.github.io/docs/Smart-Meter-Interface/) into the build breaks hardware serial bridge (cannot receive data).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP01 1M
- [x] Tasmota binary firmware version number used: 9a21dc8 (9.0.0.2 from 20.10.2020)
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: gitpod
- [x] Flashing tools used: esptool.py 2.8
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

{"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
{"Module":{"1":"Sonoff Basic"}}
{"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},...

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

There is no output even though the device connected to GPIO3 sends diligently.

```

### TO REPRODUCE
Compile build with no changes to source to find SerialReceived working.
Compile build with [SMI included](https://tasmota.github.io/docs/Smart-Meter-Interface/) to find SerialReceived NOT working.

Prior configuration just includes connecting to WiFi, set baudrate and initial `serialsend5 00`.

### EXPECTED BEHAVIOUR
Hardware SerialReceived to be working, e.g. receiving the data which is sent from the other device.

gemu2015, 2020-10-21:
did you read the sml documentation ?

https://tasmota.github.io/docs/Smart-Meter-Interface/

this driver defines all used GPIO in the driver or script.

when not using a script the default descriptor is used which occupies pin3 for serial input thus conflicting with serial bridge. (if enabled on pin 3)

when using an sml script you can define the GPIO you want to use for serial meters in the script and choosing another pin would free GPIO3 for serial bridge.

TheChatty, 2020-10-21:
I read the docs. Now again. Nowhere does it talk about default GPIOs. Instead it says if a configured smart meter inteferes with an otherwise configured PIN it will produce a "duplicate GPIO defined" error in the log and the meter descriptor will be ignored.

But I said to reproduce the above mentioned problem you just need to compile SMI into Tasmota and don't configure **any** meter. And the problem is not sending messages, it only block receiving messages.

gemu2015, 2020-10-21:
ok, it has to be clarified in the docs that the checking of duplicate GPIOs is only true for the script driven driver. the "old" hardcoded interface should normally no longer be used. it is left over for special purposes.
(e.g. if someone for some reason must use rules)

gemu2015, 2020-10-21:
docs updated!

TheChatty, 2020-10-21:
I find the whole SML doc to be quite unstructured but at least my case is now documented which explains why this issue is not a bug.

meingraham, 2020-10-21:
@TheChatty 

The firmware and the documentation both open source and are maintained by volunteers, i.e., unpaid. We encourage users like you to enhance the documentation.

TheChatty, 2020-10-21:
@meingraham: Once I fully understood what is what I might contribute.

@gemu2015: I need to reopen because when I configure my meter to use GPIO2 (means don't touch 1+3) I still can't see a response (SerialReceived log).

gemu2015, 2020-10-22:
@TheChatty 
i assume you used an ESP8266
i tested with a Wemos ESP8266 and it works like expected (sml input at gpio4, serial bridge works)
i forgot to mention and it is still missing in the docs that on an ESP32 the serial port behaves different.
i will update the docs.

by the way you should not use GPIO2 for input. dig the net about pins better not to use for input on an ESP

if you still have problems you need to describe more in detail what you are doing (post descriptor, logs etc)

TheChatty, 2020-10-22:
I'm using [this](https://www.aliexpress.com/item/4000505567851.html) board.

I just retested it to be sure. I have Tasmota 9.0.0.2 with SML installed, no GPIO configured.

- boot without any script
  - SerialSend works, SerialReceived does not work  --> correct, [explained by gemu](https://github.com/arendst/Tasmota/issues/9592#issuecomment-713514187)
- reconfigure script to use GPIO4 (no reboot)
  - SerialSend works, **SerialReceived does not work**
- boot with script configured to GPIO1+3
  - SerialSend works, SerialReceived does not work --> correct, script occupies GPIOs
- reconfigure script to use GPIO4 (no reboot)
  - SerialSend works, **SerialReceived does not work**
- boot with script configured to GPIO4
  - SerialSend works, SerialReceived works --> correct, GPIOs are usable

After the no-script situation has been clarified by gemu, I think the reconfiguration of script GPIOs needs to free unused GPIOs.

Also as a new user I'd rather prefer to have "the old behavior" removed.

My real intention is to be able to use **TCP bridge** (see #9586) and **SMI** simultaneously on the same modbus device.

gemu2015, 2020-10-23:
exactly that is the normal behavior

i made a note in docs about a required restart

TheChatty, 2020-10-23:
Not handled in my desired direction but nevertheless handled thus closed.

@gemu2015: Is there any documentation (besides the code itself) about the
> "old" hardcoded interface [...] left over for special purposes

?

closed 2020-10-23 by TheChatty

gemu2015, 2020-10-23:
as already said the old interface makes only sense in case you want to use rules and not scripts together with sml

from the source code see below
select a meter from the predefined ones.
or create a new descriptor and edit the entries.
the descriptor is the same as in a script with slightly different separators 
and some new options are not supported like MODBUS.

// this descriptor method is no longer supported
// but still functional for simple meters
// use scripting method instead
// meter list , enter new meters here
//=====================================================
#define EHZ161_0 1
#define EHZ161_1 2
#define EHZ363 3
#define EHZH 4
#define EDL300 5
#define Q3B 6
#define COMBO3 7
#define COMBO2 8
#define COMBO3a 9
#define Q3B_V1 10
#define EHZ363_2 11
#define COMBO3b 12
#define WGS_COMBO 13
#define EBZD_G 14

// select this meter
#define METER EHZ161_1

TheChatty, 2020-10-24:
@gemu2015: Undefining your last quoted line leads to my expected behavior: no undocumented feature is enabled by default and claims the rxd GPIO. In my opinion this should be condition in git.

gemu2015, 2020-10-26:
now there is a hardcoded Meter NO_OP in the meter list
that leads to an empty meter that does nothing at all
#define METER SML_NO_OP

ill do a pull request soon