arendst/Tasmota #9598: UI reports '200%' battery, although value 100 is sent by the device

jantielens, 2020-10-22:
### PROBLEM DESCRIPTION
Battery status is visualized incorrectly in the web UI.

E.g. devices are reporting a battery percentage of 100%, but the UI displays '200%' in the tooltip.

[image]

Received message:
`08:53:29 MQT: tele/sonoffbridge3/SENSOR = {"ZbReceived":{"0x11CE":{"Device":"0x11CE","ModelId":"lumi.sensor_wleak.aq1","BatteryVoltage":3.05,"BatteryPercentage":100,"Xiaomi_64":0,"Endpoint":1,"LinkQuality":157}}}
`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff Zigbee Bridge, Aqara Water Sensor & Eurotronic 
- [X] Tasmota binary firmware version number used: 9.0.0.2
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
09:16:25 CMD: Backlog Template; Module; GPIO 255
09:16:25 MQT: stat/sonoffbridge3/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
09:16:25 MQT: stat/sonoffbridge3/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
09:16:25 MQT: stat/sonoffbridge3/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
09:17:47 CMD: Status 0
09:17:47 MQT: stat/sonoffbridge3/STATUS = {"Status":{"Module":75,"DeviceName":"SonoffZbridge3","FriendlyName":["SonoffZbridge3"],"Topic":"sonoffbridge3","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:17:47 MQT: stat/sonoffbridge3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-zbbridge.bin","RestartReason":"Software/System restart","Uptime":"0T01:05:19","StartupUTC":"2020-10-22T07:12:28","Sleep":50,"CfgHolder":4617,"BootCount":25,"BCResetTime":"2020-09-29T16:44:34","SaveCount":56,"SaveAddress":"F8000"}}
09:17:47 MQT: stat/sonoffbridge3/STATUS2 = {"StatusFWR":{"Version":"9.0.0.2(zbbridge)","BuildDateTime":"2020-10-21T05:16:31","Boot":31,"Core":"2_7_4_3","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"431/699"}}
09:17:47 MQT: stat/sonoffbridge3/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["jnfl","jnflbuiten24"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
09:17:47 MQT: stat/sonoffbridge3/STATUS4 = {"StatusMEM":{"ProgramSize":532,"Free":468,"Heap":25,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F100786","04500001","00000002","00000000","00000000","00020000","00000000"],"Drivers":"1,2,4,7,9,10,20,21,23,41","Sensors":""}}
09:17:47 MQT: stat/sonoffbridge3/STATUS5 = {"StatusNET":{"Hostname":"sonoffzbridge3","IPAddress":"192.168.1.103","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"A4:CF:12:DC:06:B5","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
09:17:47 MQT: stat/sonoffbridge3/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.30","MqttPort":1883,"MqttClientMask":"sonoffbridge3","MqttClient":"sonoffbridge3","MqttUser":"mqttuser","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
09:17:47 MQT: stat/sonoffbridge3/STATUS7 = {"StatusTIM":{"UTC":"2020-10-22T08:17:47","Local":"2020-10-22T09:17:47","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:22","Sunset":"17:46"}}
09:17:47 MQT: stat/sonoffbridge3/STATUS10 = {"StatusSNS":{"Time":"2020-10-22T09:17:47"}}
09:17:47 MQT: stat/sonoffbridge3/STATUS11 = {"StatusSTS":{"Time":"2020-10-22T09:17:47","Uptime":"0T01:05:19","UptimeSec":3919,"Vcc":3.480,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"jnfl","BSSId":"42:37:86:0F:AF:72","Channel":5,"RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
(see above)

### TO REPRODUCE
Pair device, wait for battery info to be sent.

### EXPECTED BEHAVIOUR
I'd expect to see 100% in the tooltip.

### SCREENSHOTS
(see above)

### ADDITIONAL CONTEXT
In earlier (non dev) versions, the UI/tooltips are working correctly.

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-10-22:
Thanks for reporting. I will fix this.

s-hadinger, 2020-10-22:
Fixed. I don't have access to ZBBridge in the next days, so I'm fixing in the blind. Can you please confirm?

closed 2020-10-22 by s-hadinger

jantielens, 2020-10-23:
> Fixed. I don't have access to ZBBridge in the next days, so I'm fixing in the blind. Can you please confirm?

Thanks, I just deployed and now battery percentage is reported correctly!

PR #9607 (2020-10-22): Zigbee fix battery percentage
## Description:

**Related issue (if applicable):** fixes #9598 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.5
  - [x] The code change is tested and works on core ESP32 V.1.12.2
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

_NOTE: The code change must pass CI tests. **Your PR cannot be [... cut]