arendst/Tasmota #9602: Alexa can't find only ESP32 Tasmota device

brunorichter, 2020-10-22:
### PROBLEM DESCRIPTION
I have a problem with esp32 and I can't find how to solve it. Alexa does not find the emulated devices, both Wemo and Hue. I tested it with an ESP8266 here and everything worked. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used: ESP32
- [X] Tasmota binary firmware version number used:  Tasmota 9.0.0.2 
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used:   esptool.py
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
15:39:55 SRC: WebConsole from 192.168.100.70
15:39:56 CMD: Backlog Template; Module; GPIO 255
15:39:56 SRC: WebConsole from 192.168.100.70
15:39:56 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
15:39:56 SRC: Backlog
15:39:56 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
15:39:56 RSL: stat/tasmota_478550/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
15:39:56 SRC: Backlog
15:39:56 CMD: Group 0, Index 1, Command "MODULE", Data ""
15:39:56 RSL: stat/tasmota_478550/RESULT = {"Module":{"0":"ESP32-DevKit"}}
15:39:56 SRC: Backlog
15:39:56 CMD: Group 0, Index 1, Command "GPIO", Data "255"
15:39:56 RSL: stat/tasmota_478550/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"416":"PWM1"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO ...
```
- [] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
15:41:00 CMD: Group 0, Index 1, Command "BACKLOG", Data "Rule1; Rule2; Rule3"
15:41:00 SRC: Backlog
15:41:00 CMD: Group 0, Index 1, Command "RULE", Data ""
15:41:00 RSL: stat/tasmota_478550/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
15:41:00 SRC: Backlog
15:41:00 CMD: Group 0, Index 2, Command "RULE", Data ""
15:41:00 RSL: stat/tasmota_478550/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
15:41:01 SRC: Backlog
15:41:01 CMD: Group 0, Index 3, Command "RULE", Data ""
15:41:01 RSL: stat/tasmota_478550/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
15:41:09 WIF: Checking connection...
```
- [X] Provide the output of this command: `Status 0`:
```
15:41:44 SRC: WebConsole from 192.168.100.70
15:41:48 CMD: Status 0
15:41:48 SRC: WebConsole from 192.168.100.70
15:41:48 CMD: Group 0, Index 1, Command "STATUS", Data "0"
15:41:48 RSL: stat/tasmota_478550/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota1"],"Topic":"tasmota_478550","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:41:48 RSL: stat/tasmota_478550/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software reset CPU","Uptime":"0T00:02:45","StartupUTC":"2020-10-22T14:39:03","Sleep":50,"CfgHolder":4617,"BootCount":19,"BCResetTime":"2020-10-21T22:56:20","SaveCount":56}}
15:41:48 RSL: stat/tasmota_478550/STATUS2 = {"StatusFWR":{"Version":"9.0.0.2(tasmota)","BuildDateTime":"2020-10-21T17:12:02","Core":"1_0_4","SDK":"v3.2.3-14-gd3e562907","CpuFrequency":80,"Hardware":"ESP32","CR":"388/699"}}
15:41:48 RSL: stat/tasmota_478550/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"MqttLog":0,"SysLog":4,"LogHost":"","LogPort":514,"SSId":["AZEVEDO 2G",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006001","00000000"]}}
15:41:48 RSL: stat/tasmota_478550/STATUS4 = {"StatusMEM":{"ProgramSize":1050,"Free":1984,"Heap":111,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","00140001","000000CF","010013C0","40000981","00004000","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,12,20,21,24,26,27,29","Sensors":"1,2,3,5,6"}}
15:41:48 RSL: stat/tasmota_478550/STATUS5 = {"StatusNET":{"Hostname":"tasmota_Bruno","IPAddress":"192.168.100.80","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.1","Mac":"30:AE:A4:47:85:50","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:41:48 RSL: stat/tasmota_478550/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_478550","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
15:41:48 RSL: stat/tasmota_478550/STATUS7 = {"StatusTIM":{"UTC":"2020-10-22T14:41:48","Local":"2020-10-22T15:41:48","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:22","Sunset":"17:46"}}
15:41:48 RSL: stat/tasmota_478550/STATUS10 = {"StatusSNS":{"Time":"2020-10-22T15:41:48"}}
15:41:48 RSL: stat/tasmota_478550/STATUS11 = {"StatusSTS":{"Time":"2020-10-22T15:41:48","Uptime":"0T00:02:45","UptimeSec":165,"Heap":111,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":17,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"AZEVEDO 2G","BSSId":"8C:E5:EF:3F:48:78","Channel":11,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Everything works well, but the Alexa can't detect the emulated device, both in Wemo and Hue

### EXPECTED BEHAVIOUR
The expected behaviour is the connection between Alexa and Tasmota device.
In the esp8266 that works, the message appears on the console
```
21:30:49 UPP: Hue 3 response packets sent to 192.168.100.66:50000
21:30:49 HTP: Hue setup
21:35:29 RSL: tele/tasmota_633C12/STATE = {"Time":"2020-10-21T21:35:29","Uptime":"0T00:05:11","UptimeSec":311,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"AZEVEDO 2G","BSSId":"8C:E5:EF:3F:48:78","Channel":11,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:05"}}
```

### SCREENSHOTS

### ADDITIONAL CONTEXT
I have a ESP32 and a ESP8266. I did several tests with both but I couldn't find out what's wrong

**sorry about my English. It is not my native language

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-10-22:
I'm not even sure anyone tested Alexa support on esp32.

brunorichter, 2020-10-27:
A while ago It worked

dmarucci75, 2020-10-28:
I have the same problem.
Someone did it

brunorichter, 2020-11-03:
What is the next step? Wait for a response from the developers?

dmarucci75, 2020-11-04:
I think so

brunorichter, 2020-11-07:
this new release 9.1 don't solve this issue

closed 2020-12-11 by github-actions[bot]