arendst/Tasmota #9603: Zigbee2Tasmota Support for Terncy Smart Dial

BansEN77, 2020-10-22:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
I have been able to pair [Terncy Smart Dial](https://www.terncy.com/en-us/smartdial/) with a Sonoff ZbBridge running Tasmota 8.5.0 and looks like the payloads are not being translated. 

**Describe the solution you'd like**  

1) I am specially interested in serving the following payload, which contains '<' character i.e. 
`tele/ZigbeeBridge/1BFA/SENSOR = {"ZbReceived":{"0x1BFA":{"Device":"0x1BFA","FCCC<00":"3001","Endpoint":1,"LinkQuality":45}}}`

This is received after each type of press of the knob. The knob recognizes single, double, triple, quadriple up to 9-tuple presses.
The type of the press is represented within least significant 16bits.
The most significant 16 bits represent the number of executed presses per each particular press-type (separate counters per press-type) in hex value.

The above log means, that a 48th single-press happend.

2) Another type of functional messages representing the dialing actions can be serviced within tasmota rules. The knob sends the following message after the dialling action has been stopped and some time passes.
`{"ZbReceived":{"0x1BFA":{"Device":"0x1BFA","FCCC/001B":72,"FCCC/001A":483,"Endpoint":1,"LinkQuality":26}}}`
This time is quite long (1s), so I am able to rotate for seconds before getting the message:
`{"ZbReceived":{"0x1BFA":{"Device":"0x1BFA","FCCC/001B":-2568,"FCCC/001A":5265,"Endpoint":1,"LinkQuality":74}}}`

FCCC/001B - signed delta value with minimal increment of 12 (positive value for dialling right and negative for left direction). Decimal value 
FCCC/001A - time in milliseconds taken for the dialling action. Decimal value

I am able to cover this situation with:
`Rule1 ON ZbReceived#0x1BFA#FCCC/001B DO var1 %value% ENDON 
ON ZbReceived#0x1BFA#FCCC/001A DO publish delta %var1% time %value% ENDON`

**Describe alternatives you've considered**  
I considered going with a solution like:
```
var2 FCCC<00
Rule1 ON ZbReceived#0x1BFA#%VAR2% DO publish success\%value% ENDON
```

I've looked at the code in xdrv_10_rules.ino and the RulesRuleMatch parsing function.  Firsty the rule expression is parsed for finding the compare operator from the beginning of the rule string and then divided into left and right part of the rule expression. The %VARX% (and others) are then substituted with their actual values only in the right part (after the compare operator).

Most future prove solution would be to provide some means to be able to use trigger names with comparator operators within Rules. Maybe in the way I suggested above or with some special coding of ASCII characters... 

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-10-22:
Maybe it's easier to add aliases for those attributes and commands. This would solve your issue. I will propose something.

s-hadinger, 2020-10-22:
@BansEN77 Could you please provide logs of single/double press and knob turn, with `weblog 3`? It will help me do some tests without the actual device.

I propose the following mappings:
- `FCCC<00`: `TerncyPress`:<x> indicating the number of press (single, double...) corresponding to low 8 bits; `TerncyCount` with the high 8 bits
- `FCCC/001B`: `TerncyRotate` with the signed value
- `FCCC/001A`: `TerncyDuration` with the millisecond value

BansEN77, 2020-10-22:
Thanks for picking this up. I am fine with your proposal. Here go the logs:

Single press
```
21:59:45 MQT: stat/ZigbeeBridge/RESULT = {"WebLog":3}
21:59:46 CFG: Saved to flash at F9, Count 61, Bytes 4096
21:59:49 ZIG: {"ZbEZSPReceived":"4500000401CCFC016E40010000B558B2FA1BFFFF070D281263003B0102"}
21:59:49 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":64716,"srcaddr":"0x1BFA","srcendpoint":1,"dstendpoint":110,"wasbroadcast":0,"LinkQuality":24,"securityuse":0,"seqnumber":181,"fc":"0x0D","manuf":"0x1228","transact":99,"cmdid":"0x00","payload":"3B01"}}
21:59:49 ZIG: ZbZCLRawReceived: {"0x1BFA":{"FCCC<00":"3B01","Endpoint":1,"LinkQuality":24}}
21:59:49 MQT: tele/ZigbeeBridge/1BFA/SENSOR = {"ZbReceived":{"0x1BFA":{"Device":"0x1BFA","FCCC<00":"3B01","Endpoint":1,"LinkQuality":24}}}
```

Double press
```
22:01:05 ZIG: {"ZbEZSPReceived":"4500000401CCFC016E40010000B65CB3FA1BFFFF070D281264000D0202"}
22:01:05 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":64716,"srcaddr":"0x1BFA","srcendpoint":1,"dstendpoint":110,"wasbroadcast":0,"LinkQuality":26,"securityuse":0,"seqnumber":182,"fc":"0x0D","manuf":"0x1228","transact":100,"cmdid":"0x00","payload":"0D02"}}
22:01:05 ZIG: ZbZCLRawReceived: {"0x1BFA":{"FCCC<00":"0D02","Endpoint":1,"LinkQuality":26}}
22:01:05 MQT: tele/ZigbeeBridge/1BFA/SENSOR = {"ZbReceived":{"0x1BFA":{"Device":"0x1BFA","FCCC<00":"0D02","Endpoint":1,"LinkQuality":26}}}
```

Knob turn right 1 click:
```
22:01:38 ZIG: {"ZbEZSPReceived":"4500000401CCFC016E40010000B78CBFFA1BFFFF0D18650A1B00290C001A0021600102"}
22:01:38 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":64716,"srcaddr":"0x1BFA","srcendpoint":1,"dstendpoint":110,"wasbroadcast":0,"LinkQuality":58,"securityuse":0,"seqnumber":183,"fc":"0x18","manuf":"0x0000","transact":101,"cmdid":"0x0A","payload":"1B00290C001A00216001"}}
22:01:38 ZIG: ZbZCLRawReceived: {"0x1BFA":{"FCCC/001B":12,"FCCC/001A":352,"Endpoint":1,"LinkQuality":58}}
22:01:38 MQT: tele/ZigbeeBridge/1BFA/SENSOR = {"ZbReceived":{"0x1BFA":{"Device":"0x1BFA","FCCC/001B":12,"FCCC/001A":352,"Endpoint":1,"LinkQuality":58}}}
```

s-hadinger, 2020-10-23:
I made the patch in blind mode. I hope I got it right. Can you please confirm?

closed 2020-10-23 by s-hadinger

BansEN77, 2020-10-24:
I've loaded the 9.0.0.2 build and... all works as expected ;-)
Thanks!

Single knob press:
`11:06:24 ZIG: {"ZbEZSPReceived":"4500000401CCFC016E400100007F84BDFA1BFFFF070D281246003D01"}
11:06:24 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":64716,"srcaddr":"0x1BFA","srcendpoint":1,"dstendpoint":110,"wasbroadcast":0,"LinkQuality":52,"securityuse":0,"seqnumber":127,"fc":"0x0D","manuf":"0x1228","transact":70,"cmdid":"0x00","payload":"3D01"}}
11:06:24 ZIG: ZbZCLRawReceived: {"0x1BFA":{"FCCC<00":"3D01","TerncyPress":1,"TerncyCount":61,"Endpoint":1,"LinkQuality":52}}
11:06:24 MQT: tele/ZigbeeBridge/1BFA/SENSOR = {"ZbReceived":{"0x1BFA":{"Device":"0x1BFA","FCCC<00":"3D01","TerncyPress":1,"TerncyCount":61,"Endpoint":1,"LinkQuality":52}}}`

6-tuple press:
`
11:07:52 ZIG: {"ZbEZSPReceived":"4500000401CCFC016E40010000817CBBFA1BFFFF070D281248000506"}
11:07:52 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":64716,"srcaddr":"0x1BFA","srcendpoint":1,"dstendpoint":110,"wasbroadcast":0,"LinkQuality":47,"securityuse":0,"seqnumber":129,"fc":"0x0D","manuf":"0x1228","transact":72,"cmdid":"0x00","payload":"0506"}}
11:07:52 ZIG: ZbZCLRawReceived: {"0x1BFA":{"FCCC<00":"0506","TerncyPress":6,"TerncyCount":5,"Endpoint":1,"LinkQuality":47}}
11:07:52 MQT: tele/ZigbeeBridge/1BFA/SENSOR = {"ZbReceived":{"0x1BFA":{"Device":"0x1BFA","FCCC<00":"0506","TerncyPress":6,"TerncyCount":5,"Endpoint":1,"LinkQuality":47}}}
`

Knob turn to the right:
`11:08:14 ZIG: {"ZbEZSPReceived":"4500000401CCFC016E40010000825CB3FA1BFFFF0D18490A1B002948001A00212602"}
11:08:14 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":64716,"srcaddr":"0x1BFA","srcendpoint":1,"dstendpoint":110,"wasbroadcast":0,"LinkQuality":26,"securityuse":0,"seqnumber":130,"fc":"0x18","manuf":"0x0000","transact":73,"cmdid":"0x0A","payload":"1B002948001A00212602"}}
11:08:14 ZIG: ZbZCLRawReceived: {"0x1BFA":{"FCCC/001B":72,"FCCC/001A":550,"Endpoint":1,"LinkQuality":26}}
11:08:15 MQT: tele/ZigbeeBridge/1BFA/SENSOR = {"ZbReceived":{"0x1BFA":{"Device":"0x1BFA","TerncyRotate":72,"TerncyDuration":550,"Endpoint":1,"LinkQuality":26}}}`

Knob turn to the left:
`11:09:01 ZIG: {"ZbEZSPReceived":"4500000401CCFC016E400100008360B4FA1BFFFF0D184A0A1B0029A0FF1A00215902"}
11:09:01 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":64716,"srcaddr":"0x1BFA","srcendpoint":1,"dstendpoint":110,"wasbroadcast":0,"LinkQuality":29,"securityuse":0,"seqnumber":131,"fc":"0x18","manuf":"0x0000","transact":74,"cmdid":"0x0A","payload":"1B0029A0FF1A00215902"}}
11:09:01 ZIG: ZbZCLRawReceived: {"0x1BFA":{"FCCC/001B":-96,"FCCC/001A":601,"Endpoint":1,"LinkQuality":29}}
11:09:01 MQT: tele/ZigbeeBridge/1BFA/SENSOR = {"ZbReceived":{"0x1BFA":{"Device":"0x1BFA","TerncyRotate":-96,"TerncyDuration":601,"Endpoint":1,"LinkQuality":29}}}`

s-hadinger, 2020-10-24:
Thanks for the confirmation

PR #9612 (2020-10-23): Zigbee support for Terncy Smart Dial
## Description:

**Related issue (if applicable):** fixes #9603 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.5
  - [x] The code change is tested and works on core ESP32 V.1.12.2
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

_NOTE: The code change must pass CI tests. **Your PR ca [... cut]