arendst/Tasmota #9605: Bug in setting the Timer via the console

R4C3R, 2020-10-22:
### PROBLEM DESCRIPTION
If I enter "Timer 2" in the console I expect to see the "Timer 2" settings, However "Timer 1" is shown with the "Timer 2" settings. Also the "Timer 1 settings" are changed to the "Timer 2" settings.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Shelly 1
- [ ] Tasmota binary firmware version number used: 8.5.1
  - [ YES] Pre-compiled
  - [ NO] Self-compiled
  - [ ] IDE / Compiler used: 
- [ ] Flashing tools used: Tasmota OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
17:33:25 MQT: stat/tasmota_8CDF0A/RESULT = {"NAME":"Oprit","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
17:33:25 MQT: stat/tasmota_8CDF0A/RESULT = {"Module":{"46":"Shelly 1"}}
17:33:25 MQT: stat/tasmota_8CDF0A/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"21":"Relay1"},"GPIO5":{"82":"Switch1n"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

No rules used

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
17:24:53 CMD: status 0
17:24:53 MQT: stat/tasmota_8CDF0A/STATUS = {"Status":{"Module":46,"DeviceName":"Oprit","FriendlyName":["Oprit"],"Topic":"tasmota_8CDF0A","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:24:53 MQT: stat/tasmota_8CDF0A/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"16T02:09:08","StartupUTC":"2020-10-06T14:15:45","Sleep":50,"CfgHolder":4617,"BootCount":20,"BCResetTime":"2020-06-01T12:09:07","SaveCount":239,"SaveAddress":"F4000"}}
17:24:53 MQT: stat/tasmota_8CDF0A/STATUS2 = {"StatusFWR":{"Version":"8.5.1(tasmota)","BuildDateTime":"2020-10-02T10:09:47","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"389/699"}}
17:24:53 MQT: stat/tasmota_8CDF0A/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["rolfs_TP-LINK4",""],"TelePeriod":300,"Resolution":"55A180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000001","00006000","00000000"]}}
17:24:53 MQT: stat/tasmota_8CDF0A/STATUS4 = {"StatusMEM":{"ProgramSize":596,"Free":404,"Heap":22,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
17:24:53 MQT: stat/tasmota_8CDF0A/STATUS5 = {"StatusNET":{"Hostname":"tasmota_8CDF0A-7946","IPAddress":"192.168.1.72","Gateway":"192.168.1.50","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.50","Mac":"E0:98:06:8C:DF:0A","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:24:53 MQT: stat/tasmota_8CDF0A/STATUS6 = {"StatusMQT":{"MqttHost":"mosquitto.fritz.box","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8CDF0A","MqttUser":"DVES_USER","MqttCount":20,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:24:53 MQT: stat/tasmota_8CDF0A/STATUS7 = {"StatusTIM":{"UTC":"2020-10-22T16:24:53","Local":"2020-10-22T17:24:53","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:03","Sunset":"17:22"}}
17:24:53 MQT: stat/tasmota_8CDF0A/STATUS10 = {"StatusSNS":{"Time":"2020-10-22T17:24:53","Switch1":"OFF"}}
17:24:53 MQT: stat/tasmota_8CDF0A/STATUS11 = {"StatusSTS":{"Time":"2020-10-22T17:24:53","Uptime":"16T02:09:08","UptimeSec":1390148,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":20,"POWER":"OFF","Wifi":{"AP":1,"SSId":"rolfs_TP-LINK4","BSSId":"2C:3A:FD:14:87:EA","Channel":11,"RSSI":74,"Signal":-63,"LinkCount":17,"Downtime":"0T00:01:02"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

17:04:47 CMD: Timer 1
17:04:47 MQT: stat/tasmota_8CDF0A/RESULT = {"Timer1":{"Enable":1,"Mode":1,"Time":"00:30","Window":0,"Days":"1111111","Repeat":1,"Output":1,"Action":0}}
17:05:23 CMD: Timer 2
17:05:23 MQT: stat/tasmota_8CDF0A/RESULT = {"Timer1":{"Enable":1,"Mode":2,"Time":"00:00","Window":0,"Days":"1111111","Repeat":1,"Output":1,"Action":1}}
17:05:40 CMD: Timer 1
17:05:40 MQT: stat/tasmota_8CDF0A/RESULT = {"Timer1":{"Enable":1,"Mode":2,"Time":"00:00","Window":0,"Days":"1111111","Repeat":1,"Output":1,"Action":1}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Create a "Timer 1" and a different "Timer 2" via the webinterface (Configure Timer) (I used mode 1 and mode 2)
goto the console and type "Timer 1"
goto the console and type "Timer 2"
goto the console and type "Timer 1" again. Now, one can see that the settings of "Timer 2" have been applied to "Timer 1".

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
when entering "Timer 2" at the console I expect to see the settings of "Timer 2"

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-10-22:
Hi,

Your command has a syntax error. 

Instead of `Timer 2` it must be `Timer2` without the space as explained in the docs.

R4C3R, 2020-10-22:
Works indeed, sry!

closed 2020-10-22 by R4C3R