arendst/Tasmota #9611: Trigger Name of "virtual buttons" to create Trigger Event

B0F1B0, 2020-10-23:
### PROBLEM DESCRIPTION
I have a Sonoff RF-Bridge.

I have learned a button from a remote control to a "virtual button" [RFKey1].

The following works:
I created an trigger event with:

rule1 on RfReceived#RfKey=1 do something endon

--> So when i press the button on remote control the RF-Bridge get an trigger with "RfReceived#RfKey=1" and it do something.

But how can I trigger a "trigger event" when I press the "virtual button" [RfKey1] in the browser?

the command should be something like that:

`rule1 on RfTransmitted#RfKey=1 do ....`

or

`rule1 on RfKeyPressed#RfKey=1 do .....`

So what is the TRIGGER-NAME for the "virtual button"?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff RFBridge
- [x] Tasmota binary firmware version number used: 8.4.0.1
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: Arduino IDE
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-10-23:
What I see in the console when clicking a web button is `.../stat/RFKEY = {"RfKey1":"Default sent"}`
This can be matched by a rule like `on RfKey1 do ... endon`

sfromis, 2020-10-23:
If it does not work, try upgrading to current Tasmota version.

B0F1B0, 2020-10-23:
Thanks for that "simple" solution. It Works

B0F1B0

closed 2020-10-23 by B0F1B0