arendst/Tasmota #9617: Dimmer value retain

cprussin, 2020-10-24:
**Have you looked for this feature in other issues and in the docs?**  

Yes.  I'm absolutely certain this has been answered somewhere, but I've searched and searched and I can't seem to find a clear answer.  So maybe not?  And apologies if this is redundant.

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

I can't seem to find a way to get tasmota to issue any kind of state for the dimmer value such that the state is retained.  Does one exist?

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

I would like some state including dimmer value to be retained.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

Issuing automation from my subscriber to restart the tasmota device after starting up? Or just living with incorrect dimmer values upon restarting my subscriber.

**Additional context**  
_Add any other context or screenshots about the feature request here._

I found some other issues (e.g. https://github.com/arendst/Tasmota/issues/2851) which mention this issue or similar issues, but I don't see any clear resolution on those other issues.  They do seem to mention that part of the problem is that `RESULT` cannot (for good reason) be retained, but they don't seem to offer any solutions that allow me to retain anything more descriptive than the power state with `PowerRetain on`.

stefanbode, 2020-10-26:
Retaining anything else than POWER is not implemented. Maybe also not required. For these special use-cases it might be a suitable idea to add to the rules the option to retain a send MQTT message, which is already possible. In this case, anybody can retain whatever value. For example, adding a `publishretain ` on top of `publish` to retain the send message. I assume this will only add a few bytes to the code.

ascillato, 2020-10-26:
A command for Publish retained values is already implemented. Use `publish2` commander (it is explained in the docs).

stefanbode, 2020-11-02:
@cprussin : Can you check with a rule to set the Dimmer value with the publish2 command to retain the value? Feedback welcome

cprussin, 2020-11-02:
Ah sorry @stefanbode I have actually stopped using tasmota (fwiw I switched to esphome because so many of these kinds of things work as expected without me having to do additional work, so perhaps there's a feature request in here) so this is no longer an issue for me. I will close this PR and apologize for wasting your time.

closed 2020-11-02 by cprussin