arendst/Tasmota #9624: Disable all leds on Sonoff S20

HennyN1955, 2020-10-25:
Would it be possible to disable all leds on Sonoff S20?

I have several S20s that I control with Domoticz on a RaspberryPI with MQTT
When I turn on the switch I would like the led in the button to remain off. Now I can see that a led turns blue when the switch is turned on. I have LedState 0 and LedPower1 off, but this does not give the desired result.
How can I achieve that? Or is this impossible?

Thanks for anyanswer

**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-10-25:
Disabling the blue led is only possible by physical means, as it is wired to gpio12, same as the relay. Snip or cover.

HennyN1955, 2020-10-25:
> 
> 
> Disabling the blue led is only possible by physical means, as it is wired to gpio12, same as the relay. Snip or cover.

:-(
....is there no other way???

sfromis, 2020-10-25:
Sure. You can burn it. Or use strong acid. Or get another device. :-D

closed 2020-10-26 by ascillato2