arendst/Tasmota #9632: Shutter issues since 8.5.1

OppiM, 2020-10-27:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Since 8.5.1 my shutters don't act normal anymore. Operating the shutter from the webinterface and via MQTT does not give the expected positions. Operating by button works but position values on the webinterface get values bigger than 100. Tested with 8.5.1 and 9.0.0.3. Same shutter works flawless with 8.5.0.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Tepao Shutter
- [x] Tasmota binary firmware version number used: 8.5.1 and 9.0.0.3
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
09:52:18 MQT: stat/Shutter_03/RESULT = {"NAME":"Teepao Shutter","GPIO":[288,0,0,33,225,34,0,0,0,224,289,0,32,0],"FLAG":0,"BASE":18}
09:52:18 MQT: stat/Shutter_03/RESULT = {"Module":{"0":"Teepao Shutter"}}
09:52:19 MQT: stat/Shutter_03/RESULT = {"GPIO0":{"288":"Led1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"33":"Button2"},"GPIO4":{"225":"Relay2"},"GPIO5":{"34":"Button3"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"224":"Relay1"},"GPIO14":{"289":"Led2"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
09:54:16 MQT: stat/Shutter_03/RESULT = {"Rule1":"on","Once":"off","StopOnError":"off","Length":37,"Free":474,"Rules":"on button3#state do SHUTTERSTOP endon"}
09:54:16 MQT: stat/Shutter_03/RESULT = {"Rule2":"off","Once":"off","StopOnError":"off","Length":0,"Free":511,"Rules":""}
09:54:16 MQT: stat/Shutter_03/RESULT = {"Rule3":"off","Once":"off","StopOnError":"off","Length":0,"Free":511,"Rules":""}

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
09:54:45 MQT: stat/Shutter_03/STATUS = {"Status":{"Module":0,"DeviceName":"Kinderzimmer Front","FriendlyName":["Kinderzimmer Front","Shutter_03_Down"],"Topic":"Shutter_03","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:54:45 MQT: stat/Shutter_03/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:09:20","StartupUTC":"2020-10-27T08:45:25","Sleep":50,"CfgHolder":4617,"BootCount":60,"BCResetTime":"2020-03-22T09:13:43","SaveCount":1997,"SaveAddress":"F7000"}}
09:54:45 MQT: stat/Shutter_03/STATUS2 = {"StatusFWR":{"Version":"9.0.0.3(tasmota)","BuildDateTime":"2020-10-26T16:03:27","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"440/699"}}
09:54:45 MQT: stat/Shutter_03/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Oppenheimer-IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C8000100060000005AFF000000000000","C0000000","00006000","00000000"]}}
09:54:45 MQT: stat/Shutter_03/STATUS4 = {"StatusMEM":{"ProgramSize":599,"Free":404,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
09:54:45 MQT: stat/Shutter_03/STATUS5 = {"StatusNET":{"Hostname":"Shutter_03","IPAddress":"192.168.18.233","Gateway":"192.168.18.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.18.27","Mac":"3C:71:BF:31:20:DD","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
09:54:45 MQT: stat/Shutter_03/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.18.24","MqttPort":1883,"MqttClientMask":"Shutter_03","MqttClient":"Shutter_03","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
09:54:45 MQT: stat/Shutter_03/STATUS7 = {"StatusTIM":{"UTC":"2020-10-27T08:54:45","Local":"2020-10-27T09:54:45","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"07:30","Sunset":"17:37"}}
09:54:45 MQT: stat/Shutter_03/STATUS10 = {"StatusSNS":{"Time":"2020-10-27T09:54:45","Shutter1":{"Position":0,"Direction":0,"Target":0}}}
09:54:45 MQT: stat/Shutter_03/STATUS11 = {"StatusSTS":{"Time":"2020-10-27T09:54:45","Uptime":"0T00:09:20","UptimeSec":560,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER1":"off","POWER2":"off","Wifi":{"AP":1,"SSId":"Oppenheimer-IoT","BSSId":"2A:E8:29:E1:DC:D9","Channel":6,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:04"}}}
09:54:45 MQT: stat/Shutter_03/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":230,"Close":230,"50perc":50,"Delay":0,"Opt":"0000","Calib":[6928,3220,4605,900,1000],"Mode":"0"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
Steps taken:

- Restarted shutter
- shutteropenduration
- shuttercloseduration
- shutteropen
- shutterclose
- Pressed open button
- Pressed close button

```
  Console output here:
00:00:00 CFG: Loaded from flash at F6, Count 1998
00:00:00 QPC: Count 1
00:00:00 SER: Set to 8N1 115200 bit/s
00:00:00 SHT: mode undef.. calculate...
00:00:00 SHT1: Init. Pos: 0,inverted 0, locked 0, end stop time enabled 0, webButtons inverted 0
00:00:00 SRC: Restart
[... 594 lines cut ...]
09:58:43 SHT: Shutter 1: source: Shutter, powerstate_local 0, ShutterGlobal.RelayCurrentMask 2, manual change 0
09:58:43 MQT: stat/Shutter_03/RESULT = {"POWER2":"off"}
09:58:43 MQT: stat/Shutter_03/POWER2 = off
09:58:43 SHT: Shutter 1 Real 0, Start 0, Stop 0, Dir 0, Delay 0, Rtc 23.05 [s], Freq 0, PWM 0
09:58:43 MQT: stat/Shutter_03/SHUTTER1 = 0
09:58:43 MQT: stat/Shutter_03/RESULT = {"Shutter1":{"Position":0,"Direction":0,"Target":0}}
09:58:44 CFG: Saved to flash at F5, Count 2007, Bytes 4096

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Install 8.5.1 or 9.0.0.3 on shutter

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Operating via Webinterface and MQTT should work as in 8.5.0

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

kugelkopf123, 2020-10-27:
Perhaps something to do with this issue. That is by the way not fully answered:

https://github.com/arendst/Tasmota/issues/9458#issuecomment-706413938

stefanbode, 2020-10-29:
I can see that the target position is fine (0, 20000) when using the shutteropen and shutterclose as it should be. But I see that on Button1 mulipress-1 the new target is 430.000. This is the root cause of the issue. Also the log skipped a lot of messages at the beginning. Do you operate with the button directly the relay or do you use the shutterbutton command to introduce the action. Anyhow I agree it should work either both ways.

stefanbode, 2020-10-29:
There was an error with the `shuttersethalfway`. As long as it was changed once to be !=50 it has to be !=50 all the time because it changes the calibration mask. If you change it to 50 later on our're in trouble with the values. With the new release please do a `shuttersethalfway 50` to reset your calibration matrix. As an intermediate solution with the current release please use: `shuttercallibration 30 50 70 90 100` to get back to default.

commit cd91f09f (2020-10-29): Merge pull request #9660 from stefanbode/patch-4

Resolve #9632 after sethalfway back to 50
commit e40c736d (2020-10-29): Resolve #9632 after sethalfway back to 50
closed 2020-10-29 by arendst