arendst/Tasmota #9634: Enable pressure_correction and temperature_correction via mqtt (i.e. on a Xiaomi WSDCGQ11LM)

snowwiddow, 2020-10-27:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
My zigbee2tasmota coordinator is working and the Xiaomi WSDCGQ11LM (Aqara temperature, humidity and pressure sensor) reports the recommended values.

This sensor has the ability to set a correction value for temperature and for pressure (which is actually reported at sea level).
It would be great to set the correction values via mqtt.
Every time I send "ZbSend {"Device":"0XBEA7","Send":{"PressureScale":50}}" the coordinator answers with
"stat/zigbee2tasmota/thp_wz/RESULT = Unrecognized zigbee command: PressureScale"

Maybe I'm too stupid to send the correct command. If it is so, could someone tell me the right syntax?
If not it would be great to set this via mqtt. And also the temperature correction.

**Describe the solution you'd like**  
Enable pressure_correction and temperature_correction via mqtt (i.e. on a Xiaomi WSDCGQ11LM)

**Describe alternatives you've considered**  
Show me the correct syntax

**Additional context**  
-
**(Please, remember to close the issue when the problem has been addressed)**  
ok

sfromis, 2020-10-27:
According to the docs, that format is only supported for [commands](https://tasmota.github.io/docs/Zigbee/#sending-device-commands) specifically listed in the table. On a more technical level, there is also support for unlisted [low-level commands](https://tasmota.github.io/docs/Zigbee/#low-level-commands).

snowwiddow, 2020-10-27:
Yes I've read the commands table. But the sensor is definitly supported by zigbee2mqtt
[https://www.zigbee2mqtt.io/devices/WSDCGQ11LM.html](url)

and my coordinator (CC2530) is flashed with the right firmware. If I understand zigbee2tasmota the right way, it "only" transforms the zigbee communication to tasmota-mqtt.
So in simple thoughts: I just have to send the right low level command.

Jason2866, 2020-10-27:
There is no command `PressureScale` in Tasmota.. The SeaPressure is calculated in Tasmota. It is not a value from the sensor.
Temp and Humi correction (offset) is also done in Tasmota. The sensor is not related in this values.

snowwiddow, 2020-10-27:
So the topic "PressureScale:1" is created only by tasmota and not by the sensor (or the CC2530 firmware)? And also the topic "PressureScaledValue" and "SeaValue"?
Sorry that makes no sense to me. Nevertheless when the topic PressureScale is published by zigbee2tasmota it should be able to set this value and with this the software could calculate the right pressure at my actual altitude. A simple sea-level-pressure + pressure-scale = pressure at altiude. (Must not be the barometric height formula)
Sure this could be done in every home automation software but this should be done by the low-level-software. For example in zigbee2tasmota.
If the pressure scale value couldn't be set via mqtt, could someone tell me where in the tasmota source this should be inserted?

sfromis, 2020-10-27:
Seems that things are getting mixed up now. Tasmota does have features for calibrating temperature and humidity (not pressure), using an offset added to the sensor output. For pressure, Tasmota can calculate sea level pressure based on a configured altitude (allowing a form of calibration). This happens purely inside of Tasmota.

If the Zigbee sensor itself is able to to something similar, that takes sending commands to the device. Here, Tasmota *only* supports the specific commands already listed in the docs. In addition, you can yourself construct a low level command using the appropriate codes, and send that. Having Tasmota support the format you tried with would take an expansion of the available commands.

s-hadinger, 2020-10-27:
It looks that this device does not support offset nor calibration. There is currently no plan to support offsetting and calibration in Z2T. This may change but 

The attribute `PressureScale` (cluster 0x0403, attribute 0x0014) is read-only, and I don't think it does what you think. From the ZCL doc:
"The Scale attribute indicates the base 10 exponent used to obtain ScaledValue (see 4.5.2.2.2.1)."

`SeaPressure` is automatically calculated from the `Altitude` (if configured). Isn't what you want?

snowwiddow, 2020-10-27:
Ok. Thanks to all for this explanations. So I will do it the old-fashioned-way to add an offset to the pressure value in my smart home software. That's precise enough for me.
(Or if I'm too bored I'll write a function with the barometric formula :-) )
Nice evening to all from Germany

closed 2020-10-27 by snowwiddow