arendst/Tasmota #9636: integration with the OTGW

spiderrr, 2020-10-27:
**Have you looked for this feature in other issues and in the docs?**  
yes. Please note, this integration does not work with the OTGW. You might want to take a look at the HA OpenTherm GW
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
no
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

integration with the OTGW like 
Compatible Open Therm Adapters~
Ihor Melnyk OpenTherm Adapter
OpenTherm Gateway Arduino shield
or why its a problem.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
no
**Additional context**  
_Add any other context or screenshots about the feature request here._
no
**(Please, remember to close the issue when the problem has been addressed)**  
ok

Jason2866, 2020-10-27:
The Opentherm part is user contributed and not active maintained. See #8373 Well documented https://tasmota.github.io/docs/OpenTherm/#connecting-to-the-boiler 
It is as it is. You are welcome to enhance and submit a PR.

ascillato2, 2020-11-09:
Closing this request as there is no active development. Ask to reopen if you want to work on this. Thanks.

closed 2020-11-09 by ascillato2

yury-sannikov, 2021-05-20:
OTGW uses its own internal firmware and uses serial commands to interoperate. Tasmota OpenTherm integration works directly with the OpenTherm physical layer using the OpenTherm adapter.
It seems, the simplest OTGW integration might be based on rules, like 
```
on SerialReceived#Data$<"PS:" do publish my/boiler/status %value% ENDON
```