arendst/Tasmota #9640: DS18B20 sensor only reading 85.0 °C, or flapping temperature

TheOriginalMrWolf, 2020-10-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

After a number of days of running perfectly, my install ( Program Version | 8.5.0(tasmota) 2020-09-09T11:41:02 2_7_4_1/2.2.2-dev(38a443e) - precompiled & applied via OTA - on Wemos D1 mini with a single DS18B20 sensor on D3 GPIO0, and reed switches on GPIOs 4&5 (which are working perfectly)) shows temperature as a constant 85.0 °C.

A secondary issue which I notice if I graph the results of every measurement taken over a number of days is that every now and then the reported temperature starts to 'flap' on successive readings (drops 3-5 degrees C, then comes up to actual, then drops, etc).  Measured temp never exceeds actual, but always drops below.

Power cycling the Wemos & sensor immediately resolves the issue for several more days.

All connections are soldered (and good quality), and the power supply is stable and more than sufficient.  Wiring of the sensor is to +3.3v, D3 GPIO0 Button1, and GND, with a 4.7k resistor between data and +3.3v.  As mentioned, this works well for several days before exibiting strange symptoms; and a power cycle immediately resolves the issue.

Interestingly, I've seen the same issues on a Raspberry Pi logging system.

I resolved the issue on the Pi by powering the DS18b20 via a spare GPIO pin and power cycling it every few hours (GPIO supply pin is normally 'high', however taken low for a few seconds every few hours).

No issue since.

Can something similar be done with Tasmota?

Thanks!!!

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
19:31:07 CMD: Backlog Template; Module; GPIO 255
19:31:08 MQT: stat/front-gate/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
19:31:08 MQT: stat/front-gate/RESULT = {"Module":{"18":"Generic"}}
19:31:08 MQT: stat/front-gate/RESULT = {"GPIO0":{"4":"DS18x20"},"GPIO1":{"0":"None"},"GPIO2":{"56":"Led1i"},"GPIO3":{"0":"None"},"GPIO4":{"10":"Switch2"},"GPIO5":{"9":"Switch1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
19:31:56 CMD: Backlog Rule1; Rule2; Rule3
19:31:56 MQT: stat/front-gate/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":434,"Free":77,"Rules":"ON Wifi#Connected DO backlog ruletimer1 30; event sendupdate ENDON ON switch1#boot DO var1 %value% ENDON ON switch2#boot DO var2 %value% ENDON ON switch1#state DO backlog var1 %value%; event sendupdate ENDON  ON switch2#state DO backlog var2 %value%; event sendupdate ENDON  ON rules#timer=1 DO event sendupdate ENDON  ON event#sendupdate DO backlog publish carport/gate/state  {\"Switch1\":%var1%,\"Switch2\":%var2%}; ruletimer1 30 ENDON"}
19:31:56 MQT: stat/front-gate/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
19:31:57 MQT: stat/front-gate/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
19:32:57 CMD: Status 0
19:32:57 MQT: stat/front-gate/STATUS = {"Status":{"Module":18,"DeviceName":"front-gate","FriendlyName":["front-gate"],"Topic":"front-gate","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":6,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[2,2,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:32:57 MQT: stat/front-gate/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"20T00:21:00","StartupUTC":"2020-10-08T08:11:57","Sleep":50,"CfgHolder":4617,"BootCount":213,"BCResetTime":"2020-09-20T05:07:10","SaveCount":534,"SaveAddress":"F5000"}}
19:32:57 MQT: stat/front-gate/STATUS2 = {"StatusFWR":{"Version":"8.5.0(tasmota)","BuildDateTime":"2020-09-09T11:41:02","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"457/699"}}
19:32:57 MQT: stat/front-gate/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["frontBananas","frontBananas"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["04008009","2805C8000100060000005A00000000000000","000000C8","00006000","00000000"]}}
19:32:57 MQT: stat/front-gate/STATUS4 = {"StatusMEM":{"ProgramSize":595,"Free":408,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
19:32:57 MQT: stat/front-gate/STATUS5 = {"StatusNET":{"Hostname":"front-gate","IPAddress":"192.168.2.191","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.5","Mac":"5C:CF:7F:1B:3B:B3","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:32:57 MQT: stat/front-gate/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.5","MqttPort":1883,"MqttClientMask":"front-gate","MqttClient":"front-gate","MqttUser":"front-gate","MqttCount":322,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:32:57 MQT: stat/front-gate/STATUS7 = {"StatusTIM":{"UTC":"2020-10-28T08:32:57","Local":"2020-10-28T19:32:57","StartDST":"2020-10-04T02:00:00","EndDST":"2020-04-05T03:00:00","Timezone":99,"Sunrise":"05:57","Sunset":"19:18"}}
19:32:57 MQT: stat/front-gate/STATUS10 = {"StatusSNS":{"Time":"2020-10-28T19:32:57","Switch1":"OFF","Switch2":"ON","DS18B20":{"Id":"031661AD0BFF","Temperature":85.0},"TempUnit":"C"}}
19:32:57 MQT: stat/front-gate/STATUS11 = {"StatusSTS":{"Time":"2020-10-28T19:32:57","Uptime":"20T00:21:00","UptimeSec":1729260,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":322,"Wifi":{"AP":1,"SSId":"frontBananas","BSSId":"E0:63:DA:B7:71:40","Channel":11,"RSSI":34,"Signal":-83,"LinkCount":43,"Downtime":"0T00:44:33"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
N/A

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Run uninterrupted for about 3-5 days.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Actual temperature to be measured & reported at all times.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-10-28:
As gpio0 already has a pullup resistor, adding an extra one might be too much for what the sensor can pull down. Does it work differently on an unencumbered pin? There could also be some risk of a bad sensor, with many DS18B20 being cheap clones.

Jason2866, 2020-10-28:
Not a Tasmota software issue.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and the [Support](https://github.com/arendst/Tasmota#support-information).

Thanks.

### Support Information
See [Documentation](http://tasmota.com/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

TheOriginalMrWolf, 2020-10-28:
@sfromis Thanks for the great suggestion!  I had thought that setting SetOption74 to OFF was sufficient to disable pullup - but prompted by your suggestion I checked the Wemos D1 Mini schematic and, whaddyaknow, it does indeed have an _external_ 12k pullup resistor - which of course is not affected by SetOption74...

I will give it a go & let you know the outcome (honestly, not overly optimistic, but you never know!) - but this will take a couple of days at least to get to it, disassemble, desolder etc & then wait for the issue to recur (or not).

@Jason2866 Does this not highlight a possible software and/or documentation issue?  According to ESP8266 docco, GPIOs 0,2, & 15 are boot configuration pins which should have external resistors (pullup, pullup, & pulldown respectively) - this either makes them unsuitable for certain purposes (in which case the software shouldn't allow it, so a software issue if it does), or necessary to be used with caution (in which case the documentation should be enhanced - DS18b20 (https://tasmota.github.io/docs/DS18x20/ not suggesting a pin to use) being a case in point).

Also, it would be great if you could consider making DS temp readings more reliable by powering the DS via a GPIO & allowing power cycling prior to read as an enhancement request.

Thanks all!

lalo-uy, 2020-10-28:
DS18B20 always work better with Vcc to 5V and dq pulled up to 3,3V as Esp
limit.
A 100nF cap between Vcc and gnd right at the sensor improves lockup errors.
85 value indicate that the sensor had reset after start conversion, so look
to power line.
The pull up could be as low as 1k, since 7 mA is specified in the data sheet

Jason2866, 2020-10-28:
@TheOriginalMrWolf  It is documented that the ESP82XX has special GPIOs. 
https://tasmota.github.io/docs/Peripherals/#restrictions

TheOriginalMrWolf, 2020-10-28:
@lalo-uy Thanks for the suggestion!  I have seen that configuration before, but worried that this would cause Vdq to swing above acceptable limits for the ESP.  I have a CRO now so will check :)

@Jason2866 
> @TheOriginalMrWolf It is documented that the ESP82XX has special GPIOs.
> https://tasmota.github.io/docs/Peripherals/#restrictions
Great link, thanks.

> It is no general problem to connect a DS18B20 to GPIO 0.
12k plus 4.7k in parallel bring the Rpu down to about 3.37k.  DS docco says this should be close to 5k.  I'd be concerned that the DS18b20 would struggle (or be damaged) trying to pull line level down for signalling, and as a result there would be significant errors on the bus.  Do you know for a fact that there is normally no problem connecting to GPIO 0?

> Software can not change hardware designs. It is up to the user
Yes, however software should not allow configuration which contradicts hardware specs.  For instance - you should not attach switches to GPIO0 & GPIO2, yet the Tasmota firmware allows this as a configuration.  So I see that as a software defect which needs to be resolved.

Thanks all!

sfromis, 2020-10-28:
Nothing needs to be "resolved", the check you mention would be a bad idea. Lots of commercial relay devices deliberately have the switch on gpio0, and there is no problem in using gpio2 there, as long as you take a bit of time to know how to connect stuff. Gpio 1,3,15,16 also have special considerations.

TheOriginalMrWolf, 2020-10-29:
@sfromis 
> Lots of commercial relay devices deliberately have the switch on gpio0

Sorry, lots of devices have an input switch on GPIO0?  So if the user has the switch in the 'wrong' position when the device boots it doesn't boot properly?

I'm intrigued...

sfromis, 2020-10-29:
Well, when it is a momentary pushbutton switch, the issue of a plug only booting if the button is not pushed while applying power does not seem to be much of a problem, and it saves a resistor. Quite convenient for flashing; one less wire to mess with :-)

sfromis, 2020-10-29:
To put it differently. People choosing to build devices can quite reasonably be expected to look into how the hardware works. This is not a Tasmota issue, as same hardware can be used with other firmwares too.

stefanbode, 2020-10-29:
Make a friendly compromise: Lets enhance the documentation for the DS and point out what wors OOTB and what may need additional resistors. My DS18's have a hard life. Ony connected to ground and e.g. D1. Simple but still work as long as there is only one DS.

closed 2020-10-29 by ascillato2