arendst/Tasmota #9644: SMI modbus readings are mostly unsuccessfuly / console readings are not

TheChatty, 2020-10-28:
### PROBLEM DESCRIPTION
I connect my modbus device (Trovis 5573) to my ESP-01 at the default pins. I compiled Tasmota 9 including [SMI with NO_OP](https://github.com/arendst/Tasmota/issues/9592#issuecomment-716330449) to switch easily between script and console modbus communication.

When I send manual modbus commands every request is answered properly.
When I have the script request values it takes a few tries until it eventually succeeds (roughly 3 tries till web UI shows device values).

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Device used (e.g., Sonoff Basic): [ESP-01](https://ae01.alicdn.com/kf/H6092985c0d394b54938e4b963c180b4aI/0.jpg_.webp)
- [x] Tasmota binary firmware version number used: Tasmota 9.0.0.2
  - [x] Self-compiled with gitpod.io
- [x] Flashing tools used: esptool.py
- [x] Configuration: nothing but script (see addition context)
### TO REPRODUCE
- compile Tasmota 9 with SMI and NO_OP
- connect modbus device to GPIO1 and GPIO3
- connect to Wifi, insert script - don't activate yet
- at console: `serialsend5 F70300000001909C` --> `{"SerialReceived":"F7030215C5BE92"}`, request: get device type, response: 5573
- activate script
- wait for values to turn from 0 to actual value (blue LED indicates several modbus requests but no UI update)

### EXPECTED BEHAVIOUR
Every modbus request initiated by script succeed.

### SCREENSHOTS
n/a

### ADDITIONAL CONTEXT
Most simple scripts...
```
>D
>B
->sensor53 r
>M 1
+1,3,m,0,19200,Trovis,1,10,F7030009
1,F70304UUuu@i0:10,Außentemp.,°C,Temp_Outside,1
#
```
and extended scripts...
```
>D
>B
->sensor53 r
>M 1
+1,3,m,0,19200,Trovis,1,2,rF7030009000E,rF703001C0004,F703006A
1,F7031CUUuu@i0:10,Außentemp.,°C,Temp_Outside,1
1,F7031CxxxxxxxxxxxxUUuu@i0:10,Vorlauftemp.,°C,Temp_Flow,1
1,F7031CxxxxxxxxxxxxxxxxxxxxxxxxxxxxUUuu@i0:10,Rücklauftemp.,°C,Temp_Return,1
1,F7031CxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxUUuu@i0:10,Speichertemp.,°C,Temp_Vessel,1
1,F70308UUuu@i1:1,MesswertImp-h,imp/h,Metric_ImpH,0
1,F70308xxxxUUuu@i1:100,Messwertm3-h,m³/h,Metric_M3H,2
1,F70308xxxxxxxxUUuu@i1:10,AA10-10V,V,Metric_AA10,1
1,F70308xxxxxxxxxxxxUUuu@i1:10,AA20-10V,V,Metric_AA20,1
1,F70304UUuu@i2:1,StellsignalRk1,%,CtrlSig_RK1,0
#
```
need roughly three tries till values get updated.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-10-28:
Please close this and open it in Gemu2015 github.
The changes you are referring are not in Tasmota.

TheChatty, 2020-10-28:
My problem exists with current Tasmota with SMI - no matter whether the upcoming feature is in place. I adjusted the ticket to reflect this.

gemu2015, 2020-10-30:
don't get your point clearly.
the WEBUI is updated per default at about every 2300 ms by Tasmota and thus in case of your small descriptor will update about every 3 seconds.

Keep in mind that, depending on a particular MODBus device the meter response delays may vary. there are devices that need several 100 milliseconds to respond.

Thus the poll rate must be adjusted to the response time of the meter

simply type sensor53 d1 in console to debug if poll rate is appropriate 
you should see every response of the meter.

if not your poll rate is too small

TheChatty, 2020-10-30:
I just tried your command which gave me:
```
22:05:01 CMD: sensor53 d1
22:05:01 CMD: Gruppe 0, Index 53, Befehl "SENSOR", Daten "d1"
22:05:01 RSL: stat/tasmota_0BBCBE/RESULT = {"Time":"2020-10-30T22:05:01","SML":{"CMD":"dump: 1"}}
22:05:01 : f7 03 
22:05:01 : 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:01 : f7 03 04 00 64 00 01 ec 23 
22:05:01 : f7 03 1c 00 89 00 89 
22:05:01 : 00 89 01 99 7f ff 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 c5 02 69 61 d5 
22:05:02 : f7 03 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:02 : f7 03 04 00 64 00 01 ec 23 
22:05:02 : f7 03 1c 00 89 00 89 00 89 01 99 7f ff 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 c5 02 69 61 d5 
22:05:02 : f7 03 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:02 : f7 03 04 00 64 00 01 ec 23 
22:05:03 : f7 03 1c 00 89 00 89 00 89 01 99 7f ff 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 c5 02 69 61 d5 
22:05:03 : f7 03 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:03 : f7 03 04 00 64 00 01 ec 23 
22:05:03 : f7 03 1c 00 89 00 89 00 89 01 99 7f ff 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 c5 02 69 61 d5 
22:05:03 : f7 03 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:04 : f7 03 04 00 64 00 01 ec 23 
22:05:04 : f7 03 1c 00 89 00 89 00 89 01 99 7f 
22:05:04 : ff 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 c5 02 69 61 d5 
22:05:04 : f7 03 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:04 : f7 03 04 00 64 00 01 ec 23 
22:05:05 : f7 03 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:05 : f7 03 04 00 64 00 01 ec 23 
22:05:05 : f7 03 1c 00 89 00 89 00 89 01 99 7f ff 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 c5 02 69 61 d5 
22:05:05 : f7 03 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:05 : f7 03 04 00 64 00 01 ec 23 
22:05:06 : f7 03 1c 00 89 00 89 00 89 01 99 7f ff 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 c5 02 69 61 d5 
22:05:06 : f7 03 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:06 : f7 03 04 00 64 00 01 ec 23 
22:05:06 : f7 03 1c 00 89 00 89 00 89 01 99 7f ff 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 c5 02 69 61 d5 
22:05:06 : f7 03 08 00 00 00 
22:05:06 : 98 00 64 00 64 e9 b4 
22:05:07 : f7 03 04 00 64 00 01 ec 23 
22:05:07 : f7 03 1c 00 89 00 89 00 89 01 99 7f ff 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 c5 02 68 a0 15 
22:05:07 : f7 03 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:07 : f7 03 04 00 64 00 01 ec 23 
22:05:07 : f7 03 1c 00 89 00 89 00 89 01 99 
22:05:07 : 7f ff 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 c5 02 68 a0 15 
22:05:08 : f7 03 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:08 : f7 03 04 00 64 00 01 ec 23 
22:05:08 : f7 03 1c 00 89 00 89 00 89 01 99 7f ff 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 
22:05:08 : c5 02 68 a0 15 
22:05:08 : f7 03 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:08 : f7 03 04 00 64 00 01 ec 23 
22:05:09 : f7 03 1c 00 89 00 89 00 89 01 99 7f ff 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 c5 02 68 a0 15 
22:05:09 : f7 03 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:09 : f7 03 04 00 64 00 01 ec 23 
22:05:09 : f7 03 1c 00 89 00 89 00 89 01 99 7f ff 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 c5 02 68 a0 15 
22:05:09 : f7 03 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:10 : f7 03 04 
22:05:10 : 00 64 00 01 ec 23 
22:05:10 : f7 
22:05:10 : 03 1c 00 89 00 89 00 89 01 99 7f ff 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 c5 02 68 a0 15 
22:05:10 : f7 03 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:10 : f7 03 04 00 64 00 01 ec 23 
22:05:10 : f7 03 1c 00 89 00 89 00 89 01 99 7f ff 
22:05:10 : 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 c5 02 68 a0 15 
22:05:11 : f7 03 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:11 : f7 03 04 00 64 00 01 ec 23 
22:05:11 : f7 03 1c 00 89 00 89 00 89 01 99 7f ff 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 c5 02 68 a0 15 
22:05:11 : f7 03 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:11 : f7 03 04 00 64 00 01 ec 23 
22:05:12 : f7 03 1c 00 89 00 89 00 89 01 99 7f ff 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 c5 02 68 a0 15 
22:05:12 : f7 03 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:12 : f7 03 04 00 64 00 01 ec 23 
22:05:12 : f7 03 1c 00 89 00 
[... 298 lines cut ...]
22:05:32 : f7 03 04 00 64 00 01 ec 23 
22:05:32 : f7 03 1c 00 89 00 89 00 89 01 99 7f ff 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 c5 02 68 a0 15 
22:05:32 : f7 03 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:32 : f7 03 04 00 64 00 01 ec 23 
22:05:32 : f7 03 1c 00 89 00 89 00 89 01 99 7f ff 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 c5 02 68 a0 15 
22:05:32 : f7 03 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:33 : f7 03 04 00 64 00 01 ec 23 
22:05:33 : f7 03 1c 00 89 00 89 00 89 01 99 7f ff 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 c5 02 68 a0 15 
22:05:33 : f7 03 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:33 : f7 03 04 00 64 00 01 ec 23 
22:05:33 : f7 03 1c 00 89 00 89 00 89 01 99 7f ff 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 c5 02 68 a0 15 
22:05:34 : f7 03 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:34 : f7 03 04 00 64 00 01 ec 23 
22:05:34 : f7 03 1c 00 89 00 89 00 89 01 99 7f ff 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 c5 02 68 a0 15 
22:05:34 : f7 03 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:35 : f7 03 04 00 64 00 01 ec 23 
22:05:35 : f7 03 1c 00 89 00 89 00 89 01 99 7f ff 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 c5 02 68 a0 15 
22:05:35 : f7 03 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:35 : f7 03 04 00 64 00 01 ec 23 
22:05:35 : f7 03 1c 00 89 00 89 00 89 01 99 7f ff 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 c5 02 68 a0 15 
22:05:36 : f7 03 08 00 00 00 98 00 64 00 64 e9 b4 
22:05:36 : f7 03 04 00 64 00 01 ec 23 
22:05:36 : f7 03 1c 00 89 00 89 00 89 01 99 7f ff 7f ff 7f ff 01 5c 7f ff 15 c5 7f ff 7f ff 15 c5 02 68 a0 15 
```
This manual view still misses one response around "22:05:04 : f7 03 1c ..." (the long line). Anything I can try to adjust?

TheChatty, 2020-10-30:
If I change `txPeriod` from 2 to 10 values get never updated in web UI (remain at 0) and LoadAvg reaches 100. If I set `sleep 250` LoadAvg goes down again and I can even raise `txPeriod`to 20. Most probably another problem.

kugelkopf123, 2020-10-31:
As long as you have Sensor53 d1 active, the webdisplay won’t get updates. You have to switch of the raw output first with Sensor53 d0

gemu2015, 2020-10-31:
as @kugelkopf123 mentioned to view values in the GUI again you must switch off debug mode.

there is always the possibility to miss a value due to hardware glitches but since the MODBus is checksum protected you won't get false values.

to speed up web gui there would be the option to set webrefresh to a lower value. however this could lead to an overload of the cpu and i would not recommend it.

TheChatty, 2020-10-31:
You are correct, without `Sensor53 d1` and with `WebRefresh 1000` GUI gets updated in  any case. `LoadAvg` is well below 100. And missed readings are very rare. Thus I will close this issue.

closed 2020-10-31 by TheChatty