arendst/Tasmota #9645: Shelly Plug US - Powerdelta returns extra/negative values after load removed

oester, 2020-10-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

After I apply load on the plug, tele reports a powerdelta change. However, after the load is removed (switched off) and the relay is still active, powerdelta reports many more values (both positive and negative) for 3-5 seconds.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X] Device used (e.g., Sonoff Basic): Shelly Plug US
- [X] Tasmota binary firmware version number used: 8.5.1
  - [X ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ X] Flashing tools used: OTA (yaourdt/mgos-to-tasmota)
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

13:44:46 MQT: stat/ShellyPlug1/RESULT = {"NAME":"ShellyPlugUS","GPIO":[52,255,57,255,21,134,0,0,131,17,132,157,0],"FLAG":0,"BASE":45}
13:44:47 MQT: stat/ShellyPlug1/RESULT = {"Module":{"0":"ShellyPlugUS"}}
13:44:47 MQT: stat/ShellyPlug1/RESULT = {"GPIO0":{"52":"Led1"},"GPIO1":{"0":"None"},"GPIO2":{"57":"Led2i"},"GPIO3":{"0":"None"},"GPIO4":{"21":"Relay1"},"GPIO5":{"134":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"131":"HLWBL SELi"},"GPIO13":{"17":"Button1"},"GPIO14":{"132":"HLWBL CF1"},"GPIO15":{"157":"LedLink"},"GPIO16":{"0":"None"}}

```
- [ X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
13:46:05 MQT: stat/ShellyPlug1/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":138,"Free":373,"Rules":"on power1#state do backlog ledpower1 %value%; ledpower2 %value% endon on power1#boot do backlog ledpower1 %value%; ledpower2 %value% endon"}
13:46:05 MQT: stat/ShellyPlug1/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
13:46:05 MQT: stat/ShellyPlug1/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ X] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
13:48:30 MQT: stat/ShellyPlug1/STATUS = {"Status":{"Module":0,"DeviceName":"ShellyPlug1","FriendlyName":["ShellyPlug1"],"Topic":"ShellyPlug1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":8,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:48:30 MQT: stat/ShellyPlug1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T19:40:40","StartupUTC":"2020-10-27T17:07:50","Sleep":50,"CfgHolder":4617,"BootCount":4,"BCResetTime":"2020-10-27T18:05:23","SaveCount":34,"SaveAddress":"FA000"}}
13:48:30 MQT: stat/ShellyPlug1/STATUS2 = {"StatusFWR":{"Version":"8.5.1(tasmota)","BuildDateTime":"2020-10-02T10:09:47","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"390/699"}}
13:48:30 MQT: stat/ShellyPlug1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["apx66_NoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00000200","00006000","00000000"]}}
13:48:30 MQT: stat/ShellyPlug1/STATUS4 = {"StatusMEM":{"ProgramSize":596,"Free":404,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
13:48:30 MQT: stat/ShellyPlug1/STATUS5 = {"StatusNET":{"Hostname":"ShellyPlug1-1315","IPAddress":"192.168.30.51","Gateway":"192.168.30.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.25.10","Mac":"FC:F5:C4:AA:45:23","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:48:30 MQT: stat/ShellyPlug1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.25.213","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AA4523","MqttUser":"mqtt","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
13:48:30 MQT: stat/ShellyPlug1/STATUS7 = {"StatusTIM":{"UTC":"2020-10-28T12:48:30","Local":"2020-10-28T13:48:30","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:31","Sunset":"17:35"}}
13:48:30 MQT: stat/ShellyPlug1/STATUS9 = {"StatusPTH":{"PowerDelta":[10,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
13:48:30 MQT: stat/ShellyPlug1/STATUS10 = {"StatusSNS":{"Time":"2020-10-28T13:48:30","ENERGY":{"TotalStartTime":"2020-10-27T18:05:23","Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":121,"Current":0.000}}}
13:48:30 MQT: stat/ShellyPlug1/STATUS11 = {"StatusSTS":{"Time":"2020-10-28T13:48:30","Uptime":"0T19:40:40","UptimeSec":70840,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"ON","Wifi":{"AP":1,"SSId":"apx66_NoT","BSSId":"B6:FB:E4:A1:62:D5","Channel":11,"RSSI":48,"Signal":-76,"LinkCount":2,"Downtime":"0T00:00:18"}}}

```
- [X ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

13:49:23 MQT: tele/ShellyPlug1/MARGINS = {"MARGINS":{"PowerDelta":38}}
13:49:23 MQT: tele/ShellyPlug1/SENSOR = {"Time":"2020-10-28T13:49:23","ENERGY":{"TotalStartTime":"2020-10-27T18:05:23","Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":121,"Current":0.000}}
13:49:26 MQT: tele/ShellyPlug1/MARGINS = {"MARGINS":{"PowerDelta":-4}}
13:49:26 MQT: tele/ShellyPlug1/SENSOR = {"Time":"2020-10-28T13:49:26","ENERGY":{"TotalStartTime":"2020-10-27T18:05:23","Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":35,"ApparentPower":55,"ReactivePower":43,"Factor":0.63,"Voltage":121,"Current":0.454}}
13:49:28 MQT: tele/ShellyPlug1/MARGINS = {"MARGINS":{"PowerDelta":-12}}
13:49:28 MQT: tele/ShellyPlug1/SENSOR = {"Time":"2020-10-28T13:49:28","ENERGY":{"TotalStartTime":"2020-10-27T18:05:23","Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":22,"ApparentPower":54,"ReactivePower":50,"Factor":0.41,"Voltage":121,"Current":0.450}}
13:49:31 MQT: tele/ShellyPlug1/MARGINS = {"MARGINS":{"PowerDelta":-22}}
13:49:32 MQT: tele/ShellyPlug1/SENSOR = {"Time":"2020-10-28T13:49:31","ENERGY":{"TotalStartTime":"2020-10-27T18:05:23","Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":120,"Current":0.000}}
13:49:32 MQT: tele/ShellyPlug1/MARGINS = {"MARGINS":{"PowerDelta":22}}
13:49:32 MQT: tele/ShellyPlug1/SENSOR = {"Time":"2020-10-28T13:49:32","ENERGY":{"TotalStartTime":"2020-10-27T18:05:23","Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":120,"Current":0.000}}
13:49:34 MQT: tele/ShellyPlug1/MARGINS = {"MARGINS":{"PowerDelta":-22}}
13:49:34 MQT: tele/ShellyPlug1/SENSOR = {"Time":"2020-10-28T13:49:34","ENERGY":{"TotalStartTime":"2020-10-27T18:05:23","Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":121,"Current":0.000}}
13:49:35 MQT: tele/ShellyPlug1/MARGINS = {"MARGINS":{"PowerDelta":22}}
[... 2 lines cut ...]
13:49:38 MQT: tele/ShellyPlug1/SENSOR = {"Time":"2020-10-28T13:49:38","ENERGY":{"TotalStartTime":"2020-10-27T18:05:23","Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":121,"Current":0.000}}
13:49:38 MQT: tele/ShellyPlug1/MARGINS = {"MARGINS":{"PowerDelta":22}}
[... cut ...]
Set powerdelta to "10", switch on load, wait a few seconds and turn it off. tele values are sent after load is switched off.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

I'd expect powerdelta to return to zero and not report repetitive positive/negative values.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-10-28:
Verified on a Shelly1PM too.

For now it's a feature ;-)

Will see how to handle this dynamic situation where power is reported intermittendly while load has been removed.

arendst, 2020-10-28:
What is the load?

What happens if you set ``PowerDelta 50`` (= 50% load change)

oester, 2020-10-28:
I've been testing with a small fan, 35 watts. If I switch it to "powerdelta 50" I see similar results.

```
15:19:00 MQT: stat/ShellyPlug1/RESULT = {"PowerDelta1":50}
15:19:06 MQT: tele/ShellyPlug1/MARGINS = {"MARGINS":{"PowerDelta":10}}
15:19:06 MQT: tele/ShellyPlug1/SENSOR = {"Time":"2020-10-28T15:19:06","ENERGY":{"TotalStartTime":"2020-10-27T18:05:23","Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":10,"ApparentPower":10,"ReactivePower":0,"Factor":1.00,"Voltage":120,"Current":0.084}}
15:19:07 MQT: tele/ShellyPlug1/MARGINS = {"MARGINS":{"PowerDelta":24}}
15:19:07 MQT: tele/ShellyPlug1/SENSOR = {"Time":"2020-10-28T15:19:07","ENERGY":{"TotalStartTime":"2020-10-27T18:05:23","Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":10,"ApparentPower":10,"ReactivePower":0,"Factor":1.00,"Voltage":120,"Current":0.084}}

(I switched off the fan here)

15:19:26 MQT: tele/ShellyPlug1/MARGINS = {"MARGINS":{"PowerDelta":-12}}
15:19:26 MQT: tele/ShellyPlug1/SENSOR = {"Time":"2020-10-28T15:19:26","ENERGY":{"TotalStartTime":"2020-10-27T18:05:23","Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":21,"ApparentPower":53,"ReactivePower":49,"Factor":0.39,"Voltage":120,"Current":0.444}}
15:19:30 MQT: tele/ShellyPlug1/MARGINS = {"MARGINS":{"PowerDelta":-21}}
15:19:30 MQT: tele/ShellyPlug1/SENSOR = {"Time":"2020-10-28T15:19:30","ENERGY":{"TotalStartTime":"2020-10-27T18:05:23","Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":120,"Current":0.000}}
15:19:31 MQT: tele/ShellyPlug1/MARGINS = {"MARGINS":{"PowerDelta":21}}
15:19:31 MQT: tele/ShellyPlug1/SENSOR = {"Time":"2020-10-28T15:19:31","ENERGY":{"TotalStartTime":"2020-10-27T18:05:23","Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":120,"Current":0.000}}
15:19:33 MQT: tele/ShellyPlug1/MARGINS = {"MARGINS":{"PowerDelta":-21}}
15:19:33 MQT: tele/ShellyPlug1/SENSOR = {"Time":"2020-10-28T15:19:33","ENERGY":{"TotalStartTime":"2020-10-27T18:05:23","Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":120,"Current":0.000}}
15:19:34 MQT: tele/ShellyPlug1/MARGINS = {"MARGINS":{"PowerDelta":21}}
15:19:34 MQT: tele/ShellyPlug1/SENSOR = {"Time":"2020-10-28T15:19:34","ENERGY":{"TotalStartTime":"2020-10-27T18:05:23","Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":120,"Current":0.000}}
15:19:36 MQT: tele/ShellyPlug1/MARGINS = {"MARGINS":{"PowerDelta":-21}}
```
I then checked another ShellyPlug with a 60W lamp attached and I see this (powerdelta is set to 10 on this plug):

```
15:24:30 MQT: tele/ShellyPlug2/MARGINS = {"MARGINS":{"PowerDelta":58}}
15:24:30 MQT: tele/ShellyPlug2/SENSOR = {"Time":"2020-10-28T15:24:30","ENERGY":{"TotalStartTime":"2020-10-27T16:55:46","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":59,"ApparentPower":59,"ReactivePower":0,"Factor":1.00,"Voltage":119,"Current":0.491}}
15:24:43 MQT: tele/ShellyPlug2/STATE = {"Time":"2020-10-28T15:24:43","Uptime":"0T17:31:11","UptimeSec":63071,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":1,"SSId":"apx66_NoT","BSSId":"76:83:C2:AF:C8:D8","Channel":1,"RSSI":74,"Signal":-63,"LinkCount":2,"Downtime":"0T00:00:23"}}
15:24:43 MQT: stat/ShellyPlug2/RESULT = {"POWER":"OFF"}
15:24:43 MQT: stat/ShellyPlug2/POWER = OFF
15:24:43 RUL: POWER1#STATE performs "backlog ledpower1 0; ledpower2 0"
15:24:43 MQT: stat/ShellyPlug2/RESULT = {"LedPower1":"OFF"}
15:24:43 MQT: stat/ShellyPlug2/RESULT = {"LedPower2":"OFF"}
15:24:45 MQT: tele/ShellyPlug2/MARGINS = {"MARGINS":{"PowerDelta":-59}}
15:24:46 MQT: tele/ShellyPlug2/SENSOR = {"Time":"2020-10-28T15:24:45","ENERGY":{"TotalStartTime":"2020-10-27T16:55:46","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
```
This appears normal. So, it may be that the "oscillation" is occurring because the load is a motor and not something like a light bulb? This makes some sense since the load is inductive. However, it complicates the monitoring of any load that uses a motor (for instance, my intended use, a sump pump monitor.

oester, 2020-10-28:
I tested it with "powerdelta 0; powerlow 10" and I see a similar oscillation. It's definitely tied to the fact that the fan blades are still spinning after power off, so I'm guessing it's feeding back into the plug and tricking it into thinking there is still power being used.

arendst, 2020-10-28:
That makes sense.

On blitzwolf SHP2 using the same energy hardware as the Shelly Plug US I see the same as you on your 60W bulb; only one entry.

oester, 2020-10-28:
I'm not sure it's either easy or practical to fix this is software, since the oscillation will occur for an indeterminate amount of time with this sort of a load. I'm OK with closing this if you want. I'll have to ponder a way around this.

oester, 2020-10-28:
I may have to retract my previous statement on it being OK to close. I'm still seeing some anomalies with a light bulb load. When I switch the outlet on/off via the relay, I see one message generated, but when I leave the relay on and switch the load on and off I see multiple entries. Here is an example with a 60W bulb, using powerlow instead of powerdelta:

```
16:44:44 MQT: tele/ShellyPlug2/MARGINS = {"MARGINS":{"PowerLow":"OFF"}}
16:44:44 MQT: tele/ShellyPlug2/SENSOR = {"Time":"2020-10-28T16:44:44","ENERGY":{"TotalStartTime":"2020-10-27T16:55:46","Total":0.003,"Yesterday":0.000,"Today":0.003,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":119,"Current":0.000}}

The bulb was on here (Powerlow == OFF) and then I flicked the switch on the light:

16:45:18 MQT: tele/ShellyPlug2/MARGINS = {"MARGINS":{"PowerLow":"ON"}}
16:45:18 MQT: tele/ShellyPlug2/SENSOR = {"Time":"2020-10-28T16:45:18","ENERGY":{"TotalStartTime":"2020-10-27T16:55:46","Total":0.004,"Yesterday":0.000,"Today":0.004,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":119,"Current":0.000}}
16:45:19 MQT: tele/ShellyPlug2/MARGINS = {"MARGINS":{"PowerLow":"OFF"}}
16:45:19 MQT: tele/ShellyPlug2/SENSOR = {"Time":"2020-10-28T16:45:19","ENERGY":{"TotalStartTime":"2020-10-27T16:55:46","Total":0.004,"Yesterday":0.000,"Today":0.004,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":119,"Current":0.000}}
16:45:21 MQT: tele/ShellyPlug2/MARGINS = {"MARGINS":{"PowerLow":"ON"}}
16:45:21 MQT: tele/ShellyPlug2/SENSOR = {"Time":"2020-10-28T16:45:21","ENERGY":{"TotalStartTime":"2020-10-27T16:55:46","Total":0.004,"Yesterday":0.000,"Today":0.004,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":120,"Current":0.000}}
16:45:22 MQT: tele/ShellyPlug2/MARGINS = {"MARGINS":{"PowerLow":"OFF"}}
16:45:22 MQT: tele/ShellyPlug2/SENSOR = {"Time":"2020-10-28T16:45:22","ENERGY":{"TotalStartTime":"2020-10-27T16:55:46","Total":0.004,"Yesterday":0.000,"Today":0.004,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":120,"Current":0.000}}
16:45:24 MQT: tele/ShellyPlug2/MARGINS = {"MARGINS":{"PowerLow":"ON"}}
16:45:24 MQT: tele/ShellyPlug2/SENSOR = {"Time":"2020-10-28T16:45:24","ENERGY":{"TotalStartTime":"2020-10-27T16:55:46","Total":0.004,"Yesterday":0.000,"Today":0.004,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":120,"Current":0.000}}
```

arendst, 2020-10-28:
For your background information you might want to add the below line in file xdrv_03_energy.ino between lines 339 and 340:
```
    AddLog_P2(LOG_LEVEL_DEBUG, PSTR("NRG: APower %d, HPower0 %d, HPower1 %d, HPower2 %d"), active_power, Energy.power_history[phase][0], Energy.power_history[phase][1], Energy.power_history[phase][2]);
```
and see what really happens. You'll notice the energy monitoring hardware keeps reporting power usage long after the load has been turned off. This is explainable if you dive into power monitoring hardware which is time related. Depending on the type of hardware it can take several seconds before a load change is measured.

Any way the added logging will show at powerdelta time (every second) the current reported power usage and the last three seconds of usage. 

You'll find that it won't be easy to detect a load change from a temporary load toggle.

oester, 2020-10-28:
Yea, I see your point. The intended use case (sump pump) has a low duty cycle. (6-7 seconds on and them many mins/hours off) It's probably easy to catch the "on" event but telling exactly when it turns off might be harder. Sounds like I'm back to living within the limitations of the hardware.