arendst/Tasmota #9653: MQTT TLS issue

berzen, 2020-10-28:
### PROBLEM DESCRIPTION

00:00:46 MQT: Attempting connection...
00:00:56 MQT: TLS connection error: -1002
00:00:56 MQT: Connect failed to 192.168.20.27:8883, rc -2. Retry in 10 sec

if im doing a complete erase it worrks for 1-2 days, and after that it is not working again.
after that its possible to reflash, but it work not work anymore. only after erasing flash

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic
- [ ] Tasmota binary firmware version number used: 9.0.0.0.3, or 8.5.1
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: visual studio code
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

00:00:46 MQT: Attempting connection...
00:00:56 MQT: TLS connection error: -1002
00:00:56 MQT: Connect failed to 192.168.20.27:8883, rc -2. Retry in 10 sec

### TO REPRODUCE
1., compile 9.0.0.03 (or 8.5.1) with TLS support
2., tasmotizer:flash the firmware
3., set the parameters, 
4., wait 1-2 days
5. voálá...

### EXPECTED BEHAVIOUR
working TLS

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-10-28:
Error -1002 means "Cannot connect to TCP port". This is not a tls issue. 

What are you connecting to?

berzen, 2020-10-28:
Thanks for the quick feedback. I can connect to the Sonoff from the PC, and from the PC i can connect over TLS to the mosquitto server. So it is strange, if its a TCP issue. The second issue is: why its visible after 1-2 days, and why is it gone, after reflashing the device.

Jason2866, 2020-10-28:
Could be a ARP treatment issue. Try `SetOption41 <x>`  see docs https://tasmota.github.io/docs/Commands/#setoptions

berzen, 2020-10-28:
i have got Linksys APs.
one more information: if i disable TLS it works again...

Jason2866, 2020-10-28:
Not a easy issue to solve. Could be a problem with wifi equipment and the ESP82XX

berzen, 2020-10-28:
i do not think. im using tasmota with the same APs since the beginning. without problems. now, i changed to TLS, and its not working stable anymore

Jason2866, 2020-10-28:
Which device?

berzen, 2020-10-28:
sonoff basic.
after your comment i tried something: AP rebooted, and it works!
do you have a new idea?

Jason2866, 2020-10-28:
Wifi problems occur when power supply is weak. Since TLS needs a lot of MCU performance
the MCU draws more current. If it is a wemos mini it is quit sure that the onboard 3.3v vreg is to blame

Jason2866, 2020-10-28:
Ahh, sonoff basic. It is known for a very weak power supply. I am for 99% sure that is the problem!

berzen, 2020-10-28:
i have got limited space avaiable. is the shelly power supply better?

Jason2866, 2020-10-28:
Shelly has no known problems

berzen, 2020-10-28:
ok thanks for the answer. i will make a 1week test run, and i will close the issue, if the problem is not avaiable anymore

stefanbode, 2020-10-29:
I run also my devices all with TLS. Sometimes I have temporary problems that the device cannot connect to the MQTT server. I found that my FritzBox causes the issue because a reboot of the router fixes the problem. Also, it helps in many cases just wait. Not sure what went wrong. The power supply is an interesting hint, but as mentioned I only have the issue very rarely and also only on devices I boot very often because these are on my desk. I do not have issues with the 50+ always-on devices in the house. My shellys use the shelly cloud. No issues there.

berzen, 2020-10-29:
Thanks Stefan for the hint. for me to restart the router (self made pfsense router with 12 core intel atom) is not a solution. For that time i do not have VPN, and my complete family (living not only in Germany) are connected to the internet over my router (firewall) to be on the safe side. i will start to investigate the power supply issue. As i am electrical engineer, it will be easy to find out how ESP8266 reacts on load steps (this is the idea, that the power supply can support the current in general, but its to slow to react on load steps, especially if the TLS calculation begin)

berzen, 2020-10-30:
Hello guys, i have got bad news... after 2 days, the first Shelly is also failing with the same problem... So its most probably not the power supply of Sonoff. Do you have an idea?

berzen, 2020-10-30:
today all of the 3 shellys failed, and the 2 sonoffs. the forst one after 6hours, the last one after 10 hours. 2 sonoffs are working now, both are equipped with BME280 sensors. The other tasmotas without sensors are failed.

berzen, 2020-11-02:
hello everybody. the problem is not solved. What i found out, the problem is coming, if the AP is changing the channels

Jason2866, 2020-11-02:
Dont change channels ;-)

berzen, 2020-11-02:
> Dont change channels ;-)

that is what i will try today in the night. I switched off the auto channeling, and tomorrow i will come back with the result. stay tuned :-)

berzen, 2020-11-03:
hello guys,
i have bad news again. Tasmota is failing without auto channeling...im nearly sure, that is a tasmota issue.

berzen, 2020-11-04:
Hello guys again,
i have got the same feedbach, like yesterday. The own-compiled Tasmota with nearly minimum configuration is stable, but the others, in the same environment are failing. im sure, it will be a runtime issue. Whats your oppinion?

Jason2866, 2020-11-04:
Which LoadAvg do you have? Compile a version with 160Mhz.

berzen, 2020-11-07:
> 
> 
> Which LoadAvg do you have? Compile a version with 160Mhz.
19

i have got the same behavior since days:
1., same access point
2., shellys, sonoffs are connected to this AP.
3. 1 Shelly is equipped with the "nearly-minimal" compile. Working fine.
4. all of the other devices are failing. Some device after 2 hours, others after 10 hours, but within 2 days every device with the original firmware is failing.

i will use the own compiled version, and i will deny the all of the communication (expect to the MQTT server) on the firewall. In that case i do not need the updates, and i will stick for 9.0.0.3. Its not a nice solution, because the problem is not solved, but for me, its a local workaround.

Im sure, its not an issue in the network, its a tasmota issue.

closed 2020-12-11 by github-actions[bot]

berzen, 2020-12-11:
This issue is not solved. Its easy to reproduce, and there is no solution. This cant be closed

ralphrmartin, 2021-04-04:
I have the same issue with a nodeMCU

s-hadinger, 2021-04-04:
@ralphrmartin This issue is old and closed. Please open a new issue and give details and logs

berzen, 2021-04-06:
> I have the same issue with a nodeMCU

I was also very unhappy. Nobody checked the issue, but the most important thing is to close the issue. I described the problem very well. Its easy to reproduce, but they closed it. Solution is of course missing...

s-hadinger, 2021-04-06:
Are you kidding when you say nobody checked your issue? Do you realize that Tasmota is managed by benevolents in their free time? @berzen 

TLS is working fine for most people.

berzen, 2021-04-06:
> 
> 
> Are you kidding when you say nobody checked your issue? Do you realize that Tasmota is managed by benevolents in their free time? @berzen
> 
> TLS is working fine for most people.

Thank you very much for the quick answer.
I cant agree with you. 99% of the people are not using TLS, because its not compiled in the "standard" versions. On the other hand its not very easy, and user friendly to create the key in the format, which is needed by tasmota. That is why, im sure, only very few people are using TLS. I can imagine, that it works in some situation, but its not stable. I also found an issue that it is not working if you are using WLAN roaming. The hint from the community was: switch off roaming. Thank you very much, its a very good hint...
Im working in the automotive business, and i know very well about sw development, and software testing. This is, what was made here (or was communicated here)  is definetly not that, what is my expectation, sorry.

Jason2866, 2021-04-06:
The difference is you get paid for your time invest in software development in your business.
Feel free to enhance the TLS part of Tasmota. It is a Open Source project and everyone can contribute.

berzen, 2021-04-06:
> 
> 
> The difference is you get paid for your time invest in software development in your business.
> Feel free to enhance the TLS part of Tasmota. It is a Open Source project and everyone can contribute.

Thanks Jason for the invitation.
I tried to bugfix the issue on my own, after nobody really helped me. I was also not able to find the problem i have to say.
After that i made a workaround, i created a virtual LAN (and WLAN) for the tasmota and shelly devices, i set rules in pfsense to block the not needed access, and now, its ok. The communication in not secured, but nobody can have access to that network, only openhab.