arendst/Tasmota #9664: Tasmota ESP32: board hangs after applying template

utech-git, 2020-10-29:
### PROBLEM DESCRIPTION
After flashing tasmota32-webcam.bin to a [AIThinker Cam](https://pt.aliexpress.com/item/32961163394.html) whenever I apply a template or change de Module on the configuration pages the board "stops working", ie, it doesn't connect to wifi nor start the hotspot again.

For it to work again, I have first to erase the firmware and flash tasmota32-webcam.bin again. Only flashing without erasing doesn't work either.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): [ESP32CAM](https://pt.aliexpress.com/item/32961163394.html)
- [X] Tasmota binary firmware version number used: 9.0.0.3
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: Esptool.py on windows with the following command
```
esptool.py --chip esp32 --port COM7 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dout --flash_freq 40m --flash_size detect 0x1000 bootloader_dout_40m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 tasmota32-webcam.bin
```
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
(this command was issued BEFORE applying the template, because after applying it the board doesnt connect to wifi anymore)
```
20:31:09 RSL: stat/tasmota_F20B68/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
20:31:10 RSL: stat/tasmota_F20B68/RESULT = {"Module":{"1":"ESP32-DevKit"}}
20:31:10 RSL: stat/tasmota_F20B68/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
No rules were used
```
- [X] Provide the output of this command: `Status 0`:
```
20:35:43 MQT: stat/tasmota32/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota32","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:35:43 MQT: stat/tasmota32/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software reset CPU","Uptime":"0T00:03:33","StartupUTC":"2020-10-29T19:32:10","Sleep":50,"CfgHolder":4617,"BootCount":3,"BCResetTime":"2020-10-29T20:30:38","SaveCount":10}}
20:35:43 MQT: stat/tasmota32/STATUS2 = {"StatusFWR":{"Version":"9.0.0.3(webcam)","BuildDateTime":"2020-10-25T19:21:40","Core":"1_0_4","SDK":"v3.2.3-14-gd3e562907","CpuFrequency":240,"Hardware":"ESP32","CR":"363/699"}}
20:35:43 MQT: stat/tasmota32/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Nao Tem Biscoito",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
20:35:43 MQT: stat/tasmota32/STATUS4 = {"StatusMEM":{"ProgramSize":1184,"Free":1984,"Heap":103,"PsrMax":4095,"PsrFree":3345,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","00140001","000000CF","010013C0","40000981","80004000","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,12,20,21,24,26,27,29,81","Sensors":"1,2,3,5,6"}}
20:35:43 MQT: stat/tasmota32/STATUS5 = {"StatusNET":{"Hostname":"tasmota32-2920","IPAddress":"192.168.0.20","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"181.213.132.2","Mac":"24:62:AB:F2:0B:68","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
20:35:43 MQT: stat/tasmota32/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.150","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F20B68","MqttUser":"DEVuser","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:35:43 MQT: stat/tasmota32/STATUS7 = {"StatusTIM":{"UTC":"2020-10-29T19:35:43","Local":"2020-10-29T20:35:43","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:33","Sunset":"17:34"}}
20:35:43 MQT: stat/tasmota32/STATUS10 = {"StatusSNS":{"Time":"2020-10-29T20:35:43"}}
20:35:43 MQT: stat/tasmota32/STATUS11 = {"StatusSTS":{"Time":"2020-10-29T20:35:43","Uptime":"0T00:03:33","UptimeSec":213,"Heap":103,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Nao Tem Biscoito","BSSId":"D4:AB:82:4B:45:13","Channel":8,"RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
Here's the **SERIAL LOG** for after I press the "save" button on the template page:
```
20:40:16 WIF: Checking connection...
20:40:16 HTP: Configure Other
20:40:16 CFG: CR 363/699, Busy 0
20:40:16 CFG: CR 363/699, Busy 0
20:40:16 CFG: CR 363/699, Busy 0
20:40:16 CFG: CR 363/699, Busy 0
20:40:16 CFG: CR 363/699, Busy 0
20:40:16 CFG: CR 363/699, Busy 0
20:40:16 CFG: CR 371/699, Busy 0
20:40:16 CFG: CR 379/699, Busy 0
20:40:16 CFG: CR 387/699, Busy 0
20:40:16 CFG: CR 395/699, Busy 0
20:40:16 OTH: MQTT enable ON, Emulation 0, DeviceName Tasmota, FriendlyName Tasmota, Tasmota2, Tasmota3, Tasmota4, Tasmota5, Tasmota6, Tasmota7, Tasmota8
20:40:16 SRC: WebGui from 192.168.0.3
20:40:16 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template {"NAME":"AITHINKER CAM","GPIO":[4992,1,1,1,1,5088,1,1,1,1,1,1,1,1,5089,5090,0,5091,5184,5152,0,5120,5024,5056,0,0,0,0,4928,1,5094,5095,5092,0,0,5093],"FLAG":0,"BASE":1}; Module 0"
20:40:16 HTP: Restart
20:40:16 SRC: WebGui from 192.168.0.3
20:40:16 SRC: Backlog
20:40:16 CMD: Group 0, Index 1, Command "TEMPLATE", Data "{"NAME":"AITHINKER CAM","GPIO":[4992,1,1,1,1,5088,1,1,1,1,1,1,1,1,5089,5090,0,5091,5184,5152,0,5120,5024,5056,0,0,0,0,4928,1,5094,5095,5092,0,0,5093],"FLAG":0,"BASE":1}"
[... 13 lines cut ...]
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
00:00:00 CFG: Loaded, Count 12
00:00:00 QPC: Count 1
00:00:00 CFG: CR 396/699, Busy 0
00:00:00 SER: Set to 8N1 115200 bit/s
```

### TO REPRODUCE
Flash tasmota32-webcam.bin on the [AIThinker Cam](https://pt.aliexpress.com/item/32961163394.html). Apply the AI Thinker template from the [documentation](https://tasmota.github.io/docs/ESP32/#aithinker-cam): 
{"NAME":"AITHINKER CAM","GPIO":[4992,1,1,1,1,5088,1,1,1,1,1,1,1,1,5089,5090,0,5091,5184,5152,0,5120,5024,5056,0,0,0,0,4928,1,5094,5095,5092,0,0,5093],"FLAG":0,"BASE":1}

### EXPECTED BEHAVIOUR
The board should restart and then boots with the new template. But it "hangs" and doesn't connect to wifi and doesn't start the hotspot.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
If I can provide any other information, please let me know

**(Please, remember to close the issue when the problem has been addressed)**

utech-git, 2020-10-30:
I don't know if the commit above was supposed to fix this or you are still working on it, but I downloaded the new version of the [tasmota32-webcam.bin](https://github.com/arendst/Tasmota/tree/firmware/firmware/tasmota32), all the "[Needed Files](https://github.com/arendst/Tasmota/tree/firmware/firmware/tasmota32/ESP32_needed_files)", erased the previous firmware, flashed the new one and the situation still persists, here's the Serial log for after I hit the Save button

```
20:56:40 HTP: Configure Other
20:56:40 CFG: CR 362/699, Busy 0
20:56:40 CFG: CR 362/699, Busy 0
20:56:40 CFG: CR 362/699, Busy 0
20:56:40 CFG: CR 362/699, Busy 0
20:56:40 CFG: CR 362/699, Busy 0
20:56:40 CFG: CR 362/699, Busy 0
20:56:40 CFG: CR 370/699, Busy 0
20:56:40 CFG: CR 378/699, Busy 0
20:56:40 CFG: CR 386/699, Busy 0
20:56:40 CFG: CR 394/699, Busy 0
20:56:40 OTH: MQTT enable ON, Emulation 0, DeviceName Tasmota, FriendlyName Tasmota, Tasmota2, Tasmota3, Tasmota4, Tasmota5, Tasmota6, Tasmota7, Tasmota8
20:56:40 SRC: WebGui from 192.168.0.3
20:56:40 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template {"NAME":"AITHINKER CAM","GPIO":[4992,1,1,1,1,5088,1,1,1,1,1,1,1,1,5089,5090,0,5091,5184,5152,0,5120,5024,5056,0,0,0,0,4928,1,5094,5095,5092,0,0,5093],"FLAG":0,"BASE":1}; Module 0"
20:56:40 HTP: Restart
20:56:40 SRC: WebGui from 192.168.0.3
20:56:40 SRC: Backlog
20:56:40 CMD: Group 0, Index 1, Command "TEMPLATE", Data "{"NAME":"AITHINKER CAM","GPIO":[4992,1,1,1,1,5088,1,1,1,1,1,1,1,1,5089,5090,0,5091,5184,5152,0,5120,5024,5056,0,0,0,0,4928,1,5094,5095,5092,0,0,5093],"FLAG":0,"BASE":1}"
20:56:40 CFG: CR 395/699, Busy 0
20:56:40 RSL: stat/tasmota_F20B68/RESULT = {"NAME":"AITHINKER CAM","GPIO":[4992,1,1,1,1,5088,1,1,1,1,1,1,1,1,5089,5090,0,5091,5184,5152,0,5120,5024,5056,0,0,0,0,4928,1,5094,5095,5092,0,0,5093],"FLAG":0,"BASE":1}
20:56:41 SRC: Backlog
20:56:41 CMD: Group 0, Index 1, Command "MODULE", Data "0"
20:56:41 RSL: stat/tasmota_F20B68/RESULT = {"Module":{"0":"AITHINKER CAM"}}
20:56:41 CFG: Saved, Count 10, Bytes 4096
20:56:42 APP: Restarting
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
⸮00:00:00 CFG: Loaded, Count 10
00:00:00 QPC: Count 1
00:00:00 CFG: CR 395/699, Busy 0
00:00:00 SER: Set to 8N1 115200 bit/s

```

Here's the Serial log for when I power off the board after the "crash":
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0⸮
```

And the Serial Log for the subsequent power ons

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
⸮00:00:00 CFG: Loaded, Count 10
00:00:00 QPC: Count 4
00:00:00 CFG: CR 395/699, Busy 0
00:00:00 SER: Set to 8N1 115200 bit/s

```

Staars, 2020-10-31:
I can confirm this. Will take a look too.

Staars, 2020-10-31:
Okay, found something.
@arendst May I suggest to try to fix this in the TasmotaSerial library? I would need a bit some time for testing these changes.

jonas5064, 2020-10-31:
I have the same problem..

Jason2866, 2020-11-01:
Have you tried after the changes from yesterday?

Staars, 2020-11-01:
For me it works now.

utech-git, 2020-11-02:
Yep, that's working for me too now. Thanks everybody!

closed 2020-11-02 by utech-git