arendst/Tasmota #9666: Wi-Fi password with special characters

tobiknue, 2020-10-29:
### PROBLEM DESCRIPTION
There are many closed issues with the context of special characters in the Wi-Fi password, e.g. #9153, #5231 or #5010. The documentation only says, not to use "special chars or white spaces in your Password". That doesn't help very much, if your WiFi has special characters and you don't want to do without the security of special chars.

To help users in future cases, i just tried it out: The base were the special characters, AVMs FritzBox allows as documentated in their help: https://service.avm.de/help/en/FRITZ-Box-Fon-WLAN-7490/014/hilfe_zeichen_fuer_kennwoerter
For changing the password i used not the web-ui of tasmota but decode-config with only the sta_pwd and sta_ssid-section in a single .json-file. Code is below.

The result is: ALL the special characters allowed by FritzBox work in Tasmota!!! You only have to use escape syntax \ for " and \ itself. Thats all!
Summarized, the password 

> Test _-!"#$%&'()*+,./:;<=>?@[\\]^`{|}~Test

works with tasmota when you use

> Test _-!**\\**"#$%&'()*+,./:;<=>?@[**\\**\\]^`{|}~Test

in decode-config.

For me its an amazing result, because after all the issues i expected something completely different. So please use this result for the offical documentation, that the users of your great software know what to do when they have special characters in their existing wifi-password.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [X] Tasmota binary firmware version number used: Program Version 8.5.0(tasmota) Build Date & Time 2020-09-09T11:41:02
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [X] Table of special Characters tried out
```

ASCII	ASCII	ASCII			
DEC	HEX	Glyph/Name		usedPW		remark
32	20	  Space			Test Test	FritzBox: "The WiFI-key may not end mith space"
33	21	! exclamation mark	TestTest!
34	22	" quotation marks	TestTest\"	escape character syntax \ needed! without, error decode-config: invalid JSON: Expecting ',' delimiter: line 3 column 15 (char 31).
35	23	# hash			TestTest#
36	24	$ dollar sign		TestTest$
37	25	% percent sign		TestTest%
38	26	& ampersand		TestTest&
39	27	' apostrophe		TestTest'
40	28	( left bracket		TestTest(
41	29	) right bracket		TestTest)
42	2A	* astersik		TestTest*
43	2B	+ plus sign		TestTest+
44	2C	, comma			TestTest,
45	2D	- hyphen minus		TestTest-
46	2E	. full stop		TestTest.
47	2F	/ slash			TestTest/
58	3A	: colon			TestTest:
59	3B	; semicolon		TestTest;
60	3C	< less-than sign	TestTest<
61	3D	= equals sign		TestTest=
62	3E	> greater-than sign	TestTest>
63	3F	? question mark		TestTest?
64	40	@ at sign		TestTest@
91	5B	[ left square bracket	TestTest[
92	5C	\ backslash		TestTest\\	escape character syntax \ needed! without, error decode-config: invalid JSON: Invalid control character at: line 3 column 17 (char 33).
93	5D	] right square bracket 	TestTest]
94	5E	^ caret			TestTest^
95	5F	_ underscore		TestTest_
96	60	` grave accent		TestTest`
123	7B	{ left curly bracket	TestTest{
124	7C	| vertical bar		TestTest|
125	7D	} right curly bracket	TestTest}
126	7E	~ tilde			TestTest~

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Wi-Fi password in router: Test _-!"#$%&'()*+,./:;<=>?@[\]^`{|}~Test

decode-config_win64 --device <your-ip> -i Wifi.json
Wifi.json:
{
  "sta_pwd": [
    "Test _-!\"#$%&'()*+,./:;<=>?@[\\]^`{|}~Test",
    ""
  ], 
  "sta_ssid": [
    "SSID of your WiFi",
    ""
  ]
}

### EXPECTED BEHAVIOUR
works!

### SCREENSHOTS
not needed

### ADDITIONAL CONTEXT
not needed

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-10-29:
So you made no test including all the many ways passwords can be set and used. One single method is far from enough to claim "works!". Given that passwords can be included in many different formats, the varying usage of delimiters and escapes lead to significant risks of it not working all the time, or "just" the user not figuring out the details of how to push a password through the layers. While it is possible to stray outside of the high-reliability rules documented, it carries a risk of users getting into too much trouble.

ascillato, 2020-10-29:
Please, update to latest Tasmota Version (9.0.0.3) and try again.

tobiknue, 2020-10-30:
> 
> 
> Please, update to latest Tasmota Version (9.0.0.3) and try again.

ok, but this will take a few days

Jason2866, 2020-10-30:
To change the docs (in general) it has to work with ALL ways to set up Tasmota supports.
I agree a note in the docs can be added to use decode-config for setting up wifi credentials.
This way works because the settings are written in flash directly by decode-config.

ascillato2, 2020-11-16:
Please, help us and add this useful information to the docs. Thank you very much for your contributions.

closed 2020-11-16 by ascillato2

calipee, 2024-02-22:
I was also struggling with my AP with special chars in SSID and password but I managed to connect it to my network using [WifiTest](https://github.com/arendst/Tasmota/pull/16141) serial commands via the WebUI like `WifiTest mySSID+myPASSWORD`.