arendst/Tasmota #9667: Use 'Friendly Name' on dashboard instead of 'Toggle 1 and Toggle 2' names

bluemodena, 2020-10-29:
Hi there... Since there is already a field to label the two tasmota outlets ... Configuration -> Configure Other 

[image]

it would be helpful if I could rename them on the Dashboard

[image]

something like this...
[image]

**Have you looked for this feature in other issues and in the docs?**  
YES

**Is your feature request related to a problem? Please describe.**  
No, this is a pure enhancement request to make it easier to tell which outlet does what.

**Describe the solution you'd like**  
Listed above

**Describe alternatives you've considered**  
N/A

**Additional context**  
screenshots added inline above

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-10-29:
Did you consider just using the feature Tasmota already has for customizing the toggle button labels in the web UI? [Webbutton](https://tasmota.github.io/docs/Commands/#webbutton)

bluemodena, 2020-10-29:
Wonderful!  Thank you!  

Closing

closed 2020-10-29 by bluemodena