arendst/Tasmota #9669: Make Dimmer +/- step configurable

stefxx, 2020-10-30:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
I like to have my dimmers controlled by a single push button. I am using the following rule for this:

ON switch1#state=2 DO publish cmnd/kitchenLights/POWER TOGGLE ENDON
ON switch1#state=4 DO publish cmnd/kitchenLights/DIMMER %mem1% ENDON
ON switch1#state=5 DO mem1 - ENDON
ON switch1#state=6 DO mem1 + ENDON

However, the dim steps by using + and - are way to large. I am requesting to make the dim step configurable from the fixed 10.

**Describe the solution you'd like**  
I am thinking of 2 solutions:
1. Make a new SetOption (or similar) that configures the + and - step
2. Allow a number behind the + and - that indicates the required step

**Describe alternatives you've considered**  
I have been playing with alternative rules like the one below, but it gets messy and when the SetOption32 is lower than 20, it lags so much that it doesn't really work anymore.

rule1 
on system#boot do backlog var1 up; var2 80 endon
on switch1#state=2 do POWER TOGGLE endon
on switch1#state=4 do backlog event updown=%var1%; DIMMER %var2% endon
on event#updown=down do sub2 4 endon
on event#updown=up do add2 4 endon
on switch1#state=5 do var1 down endon
on switch1#state=6 do var1 up endon

**Additional context**  
Having some control on the upper and lower dimming range would be great to. I know we have the DimmerRange setting, but that doesn't impact the DIMMER command. Currently, I have to make the rule even more complex (and laggy) by adding these to additional rules:

on var2#state>90 do var2 90 endon
on var2#state<30 do var2 30 endon

**(Please, remember to close the issue when the problem has been addressed)**  
Thanks!

stefxx, 2020-10-30:
It has just been implemented here: https://github.com/arendst/Tasmota/issues/9412

Thanks again!

stefxx, 2020-10-31:
Re-opened so it can be tracked separately and @scoobler can be given the proper credits.

ascillato2, 2020-10-31:
Closing this issue in favour of #9412

closed 2020-10-31 by ascillato2