arendst/Tasmota #9672: Alexa does not recognize Tasmota Gosund SP1 v23 devices

rolarndt, 2020-10-30:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Alexa Echo V3 does  no longer recognize the Gosund SP1 v23 Module. The Device was found before and works well, but since some days it is not found any longer. Deleting it and search again for device does not work. This happened to 4 devices that works well before. It makes no difference which emulation ist used (Wemo or Hue Bridge).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Tasmota Gosund SP1 v23
- [x] Tasmota binary firmware version number used: 8.5.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tasmotizer_____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:

  Configuration output here: 11:51:20 MQT: TASMOTAS/stat/T1/RESULT = {"NAME":"Gosund SP1 v23","GPIO":[0,56,0,17,134,132,0,0,131,57,21,0,0],"FLAG":0,"BASE":55}
11:51:20 MQT: TASMOTAS/stat/T1/RESULT = {"Module":{"55":"Gosund SP1 v23"}}
11:51:20 MQT: TASMOTAS/stat/T1/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"158":"LedLinki"},"GPIO2":{"0":"None"},"GPIO3":{"17":"Button1"},"GPIO4":{"134":"BL0937 CF"},"GPIO5":{"132":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"131":"HLWBL SELi"},"GPIO13":{"56":"Led1i"},"GPIO14":{"21":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
11:51:21 MQT: TASMOTAS/stat/T1/STATUS8 = {"StatusSNS":{"Time":"2020-10-30T11:51:21","ENERGY":{"TotalStartTime":"2020-08-12T16:43:33","Total":1.037,"Yesterday":0.010,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":233,"Current":0.000}}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here: 
11:59:28 MQT: TASMOTAS/stat/T1/STATUS = {"Status":{"Module":55,"DeviceName":"T1","FriendlyName":["T1"],"Topic":"T1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:59:28 MQT: TASMOTAS/stat/T1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-sensors.bin","RestartReason":"Software/System restart","Uptime":"0T01:58:51","StartupUTC":"2020-10-30T09:00:37","Sleep":50,"CfgHolder":4617,"BootCount":65,"BCResetTime":"2020-10-09T14:41:13","SaveCount":644,"SaveAddress":"F6000"}}
11:59:28 MQT: TASMOTAS/stat/T1/STATUS2 = {"StatusFWR":{"Version":"8.5.1(tasmota)","BuildDateTime":"2020.10.02 10:15:28","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"385/699"}}
11:59:28 MQT: TASMOTAS/stat/T1/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NixGibts",""],"TelePeriod":60,"Resolution":"55A180C0","SetOption":["00008009","2805C8000100068000005A00000000000000","00000000","00006000","00000000"]}}
11:59:28 MQT: TASMOTAS/stat/T1/STATUS4 = {"StatusMEM":{"ProgramSize":597,"Free":404,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
11:59:28 MQT: TASMOTAS/stat/T1/STATUS5 = {"StatusNET":{"Hostname":"T1-7921","IPAddress":"10.42.25.157","Gateway":"10.42.25.1","Subnetmask":"255.255.255.0","DNSServer":"10.42.25.1","Mac":"E0:98:06:CB:DE:F1","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
11:59:28 MQT: TASMOTAS/stat/T1/STATUS6 = {"StatusMQT":{"MqttHost":"10.42.25.106","MqttPort":1883,"MqttClientMask":"T1","MqttClient":"T1","MqttUser":"rolarndt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:59:28 MQT: TASMOTAS/stat/T1/STATUS7 = {"StatusTIM":{"UTC":"2020-10-30T10:59:28","Local":"2020-10-30T11:59:28","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"07:34","Sunset":"17:32"}}
11:59:28 MQT: TASMOTAS/stat/T1/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
11:59:28 MQT: TASMOTAS/stat/T1/STATUS10 = {"StatusSNS":{"Time":"2020-10-30T11:59:28","ENERGY":{"TotalStartTime":"2020-08-12T16:43:33","Total":1.037,"Yesterday":0.010,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":233,"Current":0.000}}}
11:59:28 MQT: TASMOTAS/stat/T1/STATUS11 = {"StatusSTS":{"Time":"2020-10-30T11:59:28","Uptime":"0T01:58:51","UptimeSec":7131,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"NixGibts","BSSId":"18:E8:29:C7:56:3F","Channel":1,"RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Start "search devices" at Alexa Echo Version 3 or Alexa App

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Device should recognized by Alexa

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-10-30:
Close all Alexa Apps and Programs using Alexa. After doing this repower the Alexa Echo v.3. Let the Alexa Echo do the device discovery.

rolarndt, 2020-10-30:
Thanks for your quick answer.
This I already did before and tried now again, but with no success.

ascillato2, 2020-11-09:
Hi, any news on this? were you able to solve your issue or perform other tests? Thanks.

rolarndt, 2020-11-10:
> Hi, any news on this? were you able to solve your issue or perform other tests? Thanks.

I repower the Echo and all devices, but the device discovery wasn't successful. I also flashed a NodeMCU with Tasmota, configured it as switch and relay but it is also not discovered by the Echo. So for the moment I give up.

ascillato2, 2020-11-16:
> I also flashed a NodeMCU with Tasmota, configured it as switch and relay but it is also not discovered by the Echo.

So, in that case, seems that there is a problem in your network. 

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you.

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-11-16 by ascillato2