arendst/Tasmota #9673: SeeedStudio Grove 125khz RFID reader Wiegand protocol

dbvcode, 2020-10-30:
**Have you looked for this feature in other issues and in the docs?**  
Nothing like this there.

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

The problem is RDM6300 like readers seem to output fake data on the serial TX. See issue #9522 

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

[SeeedStudio Grove 125khz](https://www.seeedstudio.com/Grove-125KHz-RFID-Reader.html) readers also offer Wiegand communication.
[They have a wiki page](https://wiki.seeedstudio.com/Grove-125KHz_RFID_Reader/) with details about the readers.
And also in the page you can find [Arduino sample code for the wiegand interface](https://files.seeedstudio.com/wiki/Grove-125KHz_RFID_Reader/res/RFID_Wiegand_INT.zip).
It's a matter of tasmotizing the code.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

SeeedStudio Grove 125khz is superior to the RDM6300 and even if they output the same data on the serial, these readers pick up the cards from greater distance and don't miss tags.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-10-30:
To rule out useless work being done I suggest you start to use the default Wiegand arduino code provided by seeedstudio on their devices and see if no fake data pops up.

I don't expect this will solve the fake issue as the serial fake data was valid data with a valid header and a valid tail. I would expect to see the same using the wiegand interface.

I guess there is a reason why there is different kinds of RFID hardware ;-)

ascillato2, 2020-11-06:
Reopening as requested by other users.

Just for reference:
* The issue explained by @dbvcode was also previously reported at https://github.com/arendst/Tasmota/pull/4791#issuecomment-451016080
* Duplicated old issues with some code example from ESPEasy: https://github.com/arendst/Tasmota/issues/3647 https://github.com/arendst/Tasmota/issues/2915

dbvcode, 2020-11-06:
> I'll close the issue as I need an Arduino for that. When I have one, I'll post my findings..

My Arduino is still on it's way...

closed 2020-12-11 by github-actions[bot]