arendst/Tasmota #9675: Make a Tasmota Device trigger Other Tasmota device based on sensors or other triggers like "Motion Sensor to turn on switch in other device

Ashustar, 2020-10-30:
**Have you looked for this feature in other issues and in the docs?**  
Yes 
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-10-30:
You did not read the docs, you can already use rules to publish commands to other devices via MQTT, or other means.

Ashustar, 2020-10-30:
I did, we have to avoid the middle server to interact among these devices

> 
> 
> You did not read the docs, you can already use rules to publish commands to other devices via MQTT, or other means.

lalo-uy, 2020-10-30:
You can make direct http requests.

Ashustar, 2020-10-30:
yes, we can through commands but not web UI. it must be easy to set up a trigger without using the command, just give IP, select type switch/relay/pin, its number,  state to attain, on what Sensor state/value.
not through command "http://<ip>/cm?user=<username>&password=<password>&cmnd=Power%20On"