arendst/Tasmota #9681: Need assistance with MagicHome Configuration:

neuman1812, 2020-10-30:
### PROBLEM DESCRIPTION
Colors on LED strip do not mach colors when sent.   OR match Slider on WebUI
Using the Color command,  I tried the list   (xx) is the expected result according to Documents.  What follows is the actual result.
1 = (red) Turns off Led
2 = (green)Bright green
3 = (blue) red
4 = (orange) Dim Green
5 = (light green) Bright Green but lower than #2
6 = (light blue) red,  dimmer than 3
7 = (amber)Dim green,  lower than  6
8 = (cyan)  Green..middle brightness
9 = (purple) Bright Red
10 = (yellow) Brightest green
11 = (pink) Dim red
12 = (white) Bright Yellow

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Magic Home LED controller (url at bottom)
- [ ] Tasmota binary firmware version number used: 8.5.1
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:

21:15:05 CMD: Backlog Template; Module; GPIO 255
21:15:05 MQT: stat/BarAccent/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
21:15:05 MQT: stat/BarAccent/RESULT = {"Module":{"34":"MagicHome"}}
21:15:05 MQT: stat/BarAccent/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"56":"Led1i"},"GPIO3":{"0":"None"},"GPIO4":{"147":"ALux IrRcv"},"GPIO5":{"38":"PWM2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"39":"PWM3"},"GPIO13":{"0":"None"},"GPIO14":{"37":"PWM1"},"GPIO15":{"159":"ALux IrSel"},"GPIO16":{"0":"None"}}
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
21:16:14 MQT: stat/BarAccent/STATUS = {"Status":{"Module":34,"DeviceName":"BarAccent","FriendlyName":["BarAccent"],"Topic":"BarAccent","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:16:14 MQT: stat/BarAccent/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"1T02:56:08","StartupUTC":"2020-10-29T17:20:06","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2020-10-29T17:38:02","SaveCount":130,"SaveAddress":"FA000"}}
21:16:14 MQT: stat/BarAccent/STATUS2 = {"StatusFWR":{"Version":"8.5.1(tasmota)","BuildDateTime":"2020-10-02T10:09:47","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"376/699"}}
21:16:14 MQT: stat/BarAccent/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["LeFort",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["40008009","2805C8000100060000005A00000000000000","00000200","00006000","00000000"]}}
21:16:14 MQT: stat/BarAccent/STATUS4 = {"StatusMEM":{"ProgramSize":596,"Free":404,"Heap":21,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
21:16:14 MQT: stat/BarAccent/STATUS5 = {"StatusNET":{"Hostname":"BarAccent-7453","IPAddress":"192.168.1.164","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"10:52:1C:CC:3D:1D","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:16:14 MQT: stat/BarAccent/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.103","MqttPort":1883,"MqttClientMask":"BarAccent","MqttClient":"BarAccent","MqttUser":"sonoff","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:16:14 MQT: stat/BarAccent/STATUS7 = {"StatusTIM":{"UTC":"2020-10-30T20:16:14","Local":"2020-10-30T21:16:14","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:34","Sunset":"17:32"}}
21:16:14 MQT: stat/BarAccent/STATUS10 = {"StatusSNS":{"Time":"2020-10-30T21:16:14"}}
21:16:14 MQT: stat/BarAccent/STATUS11 = {"StatusSTS":{"Time":"2020-10-30T21:16:14","Uptime":"1T02:56:08","UptimeSec":96968,"Heap":23,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":100,"Color":"FFFFFF","HSBColor":"0,0,100","Channel":[100,100,100],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"LeFort","BSSId":"88:3D:24:8B:58:19","Channel":6,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Install tasmoda on MagicHome device. Change module to MagicHome.  Attempt to adjust colors.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
Product Used: https://www.amazon.com/gp/product/B07QBKRCW1/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Top of Controller: https://i.imgur.com/HOD5ZDa.jpg
Bottom Of controller: https://i.imgur.com/ww8hmS4.jpg

Example output using Slider, Hopefully you can see the LED is in the background, The slider is set to red:
https://i.imgur.com/1DV2QGL.jpg

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-10-30:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and the [Support](https://github.com/arendst/Tasmota#support-information).

Thanks.

### Support Information
See [Documentation](http://tasmota.com/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

neuman1812, 2020-10-30:
Thank you.  Wasn't sure if this was a bug or not.  Magic home is funky.

closed 2020-10-31 by ascillato2