arendst/Tasmota #9685: Rule ON AM2301#Humidity=NULL not working on 8.5.1,  did work on 8.4.0.

gwietze, 2020-10-30:
# PROBLEM DESCRIPTION
Rule 3 is doing nothng anymore after update 8.5.1
If I go back to 8.4.0 it works again, it looks like "NULL" is not a value anymore in 8.5.1

### REQUESTED INFORMATION
Rule 3 is needed because sometimes DHT22 doesn't has output anymore and givings "NULL"
Then this will power down the DHT22 and after 30 seconds RuleTimer 2 set back power on DHT22 and set after 30 seconds Rule3 On again.

( tussentijd in rust )
mem1 60
( tussentijd met fan aan )
mem2 600
( Waarde Humidity )
mem3 90
Timer in sec

Rule1
ON system#boot DO Backlog RuleTimer1 %mem1%; Rule2 On; Rule3 On ENDON
ON Rules#Timer=1 DO Rule2 On ENDON
ON Rules#Timer=2 DO backlog Power2 1; RuleTimer3 30 ENDON
ON Rules#Timer=3 DO Rule3 On ENDON
ON Power1#State=1 DO RuleTimer1 %mem2% ENDON
ON Power1#State=0 DO RuleTimer1 %mem1% ENDON

read Humidity
Rule2
ON AM2301#Humidity>%mem3% DO backlog Power1 1; Rule2 off; RuleTimer1 %mem2% ENDON
ON AM2301#Humidity<%mem3% DO backlog Power1 0; Rule2 off; RuleTimer1 %mem1% ENDON

Rule3
ON AM2301#Humidity=NULL DO backlog Power2 0;RuleTimer2 30; Rule3 Off ENDON

### TO REPRODUCE
With firmware  8.5.1 Rule3 is doing nothing anymore

### EXPECTED BEHAVIOUR
Rule 3 is needed because sometimes DHT22 doesn't has output anymore and is giving "NULL"
Then this will power down the DHT22 and after 30 seconds RuleTimer 2 set back power on DHT22 and set after 30 seconds Rule3 On again to check the next NULL.

### SCREENSHOTS
No screenshots

### ADDITIONAL CONTEXT
back to 8.4.0 and Rule3 is running again.

Jason2866, 2020-10-31:
Please provide logs where the "NULL" output of the DHT22 can be seen.

JorgeMaTeixeira, 2020-11-01:
Happens to me too.. Null result. And I'm trying rule 3.

Jason2866, 2020-11-01:
Still no log provided where the `NULL` value appears

gwietze, 2020-11-01:
The NULL appears when DHT22 is not given no serial data anymore when this happens I don't know. The result is that the output goes to NULL. Then I can start up the DHT22 again to power down for 30 seconds and then power it up again. I use a power2 to do this. Rule3 is starting the procedure when the output is NULL. So how can I get logging of that.

gwietze, 2020-11-01:
The big question is why does it work on 8.4.0?

Jason2866, 2020-11-01:
You cant trigger on a event that never appears.
So the first step is checking for the trigger.
There are changes in the DHT driver done.
There is the possibility that NULL is not appearing anymore.

sfromis, 2020-11-01:
Did a small test on 9.0.0.2. Configured a DHT11 without connecting one.
```
{"Time":"2020-11-01T23:39:52.516+01:00","DHT11":{"Temperature":null,"Humidity":null,"DewPoint":null},"TempUnit":"C"}
{"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":40,"Free":471,"Rules":"on DHT11#Humidity=null do var1 123 endon"}
No reaction.

arendst, 2020-11-02:
I suspect the change of ArduinoJson with JSMN.

Will have a look how null is treated by JSMN.

sfromis, 2020-11-02:
My test case now also got rapid fire with null values.

closed 2020-11-04 by ascillato2