arendst/Tasmota #9687: PZEM-004T V3 issues

Anon0ne, 2020-10-31:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Installed a PZEM-004T on 8.5.1 and had issues with no readings. Eventually I swapped the config for the module for RX to RX and TX to TX and it began working. But the information being read is extremely high.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Install PZEM-004T on 8.5.1 and look at data

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Have normal readings, and have the Hardware Serial setup to RX to TX and TX to RX. 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-10-31:
What does the webfrontend show? What values are in the log messages?

Anon0ne, 2020-10-31:
MQTT log matches web frontend
```
Voltage | 122 V
87.758 A
3578 W
10706 VA
2945 VAr
0.33
60 Hz
4.696 kWh
0.000 kWh
125.459 kWh

```

Jason2866, 2020-10-31:
Sorry, i dont get what the issue is. You get values. What is wrong?

Anon0ne, 2020-10-31:
> Sorry, i dont get what the issue is. You get values. What is wrong?

The issue is the current is really high. It's 10x higher than I think it should be. This causes all readings to be high.

Anon0ne, 2020-10-31:
The issue is the current reading is 10x too high. The kWh I read via the meter is also much lower than these readings

Anon0ne, 2020-10-31:
[image]
The energy today is still resetting often also.. and the device isn't powering off.

Jason2866, 2020-10-31:
Please add all the data which is asked in the issue template.

Anon0ne, 2020-10-31:
Okay, updated the information. You can see that the Today's energy is all over the place, up and down. I just enabled setOption72 ON to see if that helps or worsens it. Before when I was on software serial it seemed to help.

Anon0ne, 2020-10-31:
```
","ENERGY":{"TotalStartTime":"2020-10-31T00:00:00","Total":139.935,"Yesterday":0.000,"Today":0.004,"Period":42949660,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":0,"Voltage":0,"Current":0.000}}
"ENERGY":{"TotalStartTime":"2020-10-31T00:00:00","Total":139.934,"Yesterday":0.000,"Today":0.003,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":0,"Voltage":0,"Current":0.000}}
```

Now i am getting a bunch of zero, not sure what the heck is going on. Also I see this "Period" is going from a huge number to 0 and back.

ascillato2, 2020-11-15:
Hi, please, update to latest Tasmota version from development branch and try again. Thanks.

ascillato2, 2020-11-16:
Please, when you have time, let us know if this solves your issue. Thanks

closed 2020-11-16 by ascillato2