arendst/Tasmota #9688: Web Ui does not update dimming settings

broncosis, 2020-10-31:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): _Ds01____
- [x ] Tasmota binary firmware version number used: _9.0.0.3 after 8.5.1 ____
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: __OTA___
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
12:57:09 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
12:57:09 SRC: Backlog
12:57:09 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
12:57:09 MQT: stat/moesdim/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
12:57:09 SRC: Backlog
12:57:09 CMD: Group 0, Index 1, Command "MODULE", Data ""
12:57:09 MQT: stat/moesdim/RESULT = {"Module":{"54":"Tuya MCU"}}
12:57:09 SRC: Backlog
12:57:09 CMD: Group 0, Index 1, Command "GPIO", Data "255"
12:57:09 MQT: stat/moesdim/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
12:57:15 TYA: Send "55aa00000000ff"

```
- [x ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

12:59:44 CMD: Backlog Rule1; Rule2; Rule3
12:59:44 SRC: WebConsole from 10.0.1.50
12:59:44 CMD: Group 0, Index 1, Command "BACKLOG", Data "Rule1; Rule2; Rule3"
12:59:44 SRC: Backlog
12:59:44 CMD: Group 0, Index 1, Command "RULE", Data ""
12:59:44 MQT: stat/moesdim/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
12:59:44 SRC: Backlog
12:59:44 CMD: Group 0, Index 2, Command "RULE", Data ""
12:59:44 MQT: stat/moesdim/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
12:59:45 SRC: Backlog
12:59:45 CMD: Group 0, Index 3, Command "RULE", Data ""
12:59:45 MQT: stat/moesdim/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
```
- [x ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
12:59:13 CMD: Status 0
12:59:13 SRC: WebConsole from 10.0.1.50
12:59:13 CMD: Group 0, Index 1, Command "STATUS", Data "0"
12:59:13 MQT: stat/moesdim/STATUS = {"Status":{"Module":54,"DeviceName":"Moesdim","FriendlyName":["Moesdim"],"Topic":"moesdim","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:59:13 MQT: stat/moesdim/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:02:20","StartupUTC":"2020-10-31T11:56:53","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2020-10-31T11:50:35","SaveCount":97,"SaveAddress":"FB000"}}
12:59:13 MQT: stat/moesdim/STATUS2 = {"StatusFWR":{"Version":"9.0.0.3(tasmota)","BuildDateTime":"2020-10-31T11:05:49","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"380/699"}}
12:59:13 MQT: stat/moesdim/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["thingsthatgobump",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
12:59:13 MQT: stat/moesdim/STATUS4 = {"StatusMEM":{"ProgramSize":599,"Free":404,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
12:59:13 MQT: stat/moesdim/STATUS5 = {"StatusNET":{"Hostname":"moesdim-7501","IPAddress":"10.0.1.73","Gateway":"10.0.1.1","Subnetmask":"255.255.0.0","DNSServer":"64.59.144.16","Mac":"5C:CF:7F:AB:1D:4D","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
12:59:13 MQT: stat/moesdim/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.11","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AB1D4D","MqttUser":"bobcatmqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
12:59:13 MQT: stat/moesdim/STATUS7 = {"StatusTIM":{"UTC":"2020-10-31T11:59:13","Local":"2020-10-31T12:59:13","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:36","Sunset":"17:30"}}
12:59:13 MQT: stat/moesdim/STATUS10 = {"StatusSNS":{"Time":"2020-10-31T12:59:13"}}
12:59:13 MQT: stat/moesdim/STATUS11 = {"StatusSTS":{"Time":"2020-10-31T12:59:13","Uptime":"0T00:02:20","UptimeSec":140,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":15,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"thingsthatgobump","BSSId":"1C:AB:C0:B1:BF:F8","Channel":3,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:03"}}}
12:59:16 TYA: Send "55aa00000000ff"

```
- [x ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
13:00:59 {"TuyaReceived":{"Data":"55AA030700080202000400000096AF","Cmnd":7,"CmndData":"0202000400000096","DpType2Id2":150,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000096"}}}
13:00:59 TYA: fnId=21 is set for dpId=2
13:00:59 TYA: RX value 150 from dpId 2 
13:00:59 {"TuyaReceived":{"Data":"55AA030700080202000400000096AF","Cmnd":7,"CmndData":"0202000400000096","DpType2Id2":150,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000096"}}}
13:00:59 TYA: fnId=21 is set for dpId=2
13:00:59 TYA: RX value 150 from dpId 2 
13:00:59 {"TuyaReceived":{"Data":"55AA0307000802020004000001E5FF","Cmnd":7,"CmndData":"02020004000001E5","DpType2Id2":485,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000001E5"}}}
13:00:59 TYA: fnId=21 is set for dpId=2
13:00:59 TYA: RX value 485 from dpId 2 
13:00:59 {"TuyaReceived":{"Data":"55AA0307000802020004000003536F","Cmnd":7,"CmndData":"0202000400000353","DpType2Id2":851,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000353"}}}
13:00:59 TYA: fnId=21 is set for dpId=2
13:00:59 TYA: RX value 851 from dpId 2 
13:00:59 {"TuyaReceived":{"Data":"55AA0307000802020004000003E804","Cmnd":7,"CmndData":"02020004000003E8","DpType2Id2":1000,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000003E8"}}}
13:00:59 TYA: fnId=21 is set for dpId=2
13:00:59 TYA: RX value 1000 from dpId 2 
13:00:59 {"TuyaReceived":{"Data":"55AA0307000802020004000003E804","Cmnd":7,"CmndData":"02020004000003E8","DpType2Id2":1000,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000003E8"}}}
13:00:59 TYA: fnId=21 is set for dpId=2
13:00:59 TYA: RX value 1000 from dpId 2 
13:01:00 TYA: Send "55aa000600080202000400000096ab"
13:01:00 {"TuyaReceived":{"Data":"55AA030700080202000400000096AF","Cmnd":7,"CmndData":"0202000400000096","DpType2Id2":150,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000096"}}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
change dimmer manually and notice not change

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

I expected the web ui to update as the manual dimmer was adjusted 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

I noticed the same Mqtt problem  as issue #9583
so I followed and updated to the latestest release as this device in not in use anywhere
and I now have the toggle updates on the web ui but the Dimming still does not 
before under 8.5.1 niether the toggle or the Dimming updated 
I have not checked Mqtt output since the update to 9.0.0.3

hopefully the dual monitor screenshots with both console and web ui open are useful

[image]
[image]

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-10-31:
Well, your expectations were too strong. The web UI does not try to cover everything, and if you want updated sliders you'll have to refresh the page. The web UI is not meant for full operation of the device, more for admin/debug.

ascillato2, 2020-10-31:
Sorry, as designed. The WEBUI don't reflects the change being done by other means.

closed 2020-10-31 by ascillato2