arendst/Tasmota #9692: No Power Measurements on Shelly 2.5 with only 1 connected load

stefxx, 2020-10-31:
### PROBLEM DESCRIPTION
I have 15 Shelly 2.5 devices with Tasmota running. 13 are connected to 2 loads, they are reporting power measurements as expected. 2 of them have only a single load (connected to O1) and those two are **not** reporting any power measurements. 

_A clear and concise description of what the problem is._
No power measurements when only a single load is connected. I would expect the have power measurements even when a single load is connected.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [v] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [v] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [v] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [v] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [v] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [v] Device used (e.g., Sonoff Basic): Shelly 2.5
- [v] Tasmota binary firmware version number used: 9.0.0.3
  - [v] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [v] Flashing tools used: OTA
- [v] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
21:49:08 CMD: Backlog Template; Module; GPIO 255
21:49:08 MQT: stat/KeukenBuitenLicht/RESULT = {"NAME":"Shelly 2.5","GPIO":[320,0,34,0,224,129,0,0,640,128,608,225,3456,4736],"FLAG":0,"BASE":18}
21:49:08 MQT: stat/KeukenBuitenLicht/RESULT = {"Module":{"0":"Shelly 2.5"}}
21:49:09 MQT: stat/KeukenBuitenLicht/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"34":"Button3"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"129":"Button_in2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"128":"Button_in1"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"225":"Relay2"},"GPIO16":{"3456":"ADE7953 IRQ"},"GPIO17":{"4736":"ADC Temp"}}

```
- [v] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
21:50:04 CMD: Backlog Rule1; Rule2; Rule3
21:50:04 MQT: stat/KeukenBuitenLicht/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":190,"Free":321,"Rules":"on power2#state do publish cmnd/TuinLicht/Power1 %value% endon on power2#state do publish cmnd/TuinLicht/Power2 %value% endon on power2#state do publish cmnd/SchuurLicht/Power2 %value% endon"}
21:50:04 MQT: stat/KeukenBuitenLicht/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
21:50:04 MQT: stat/KeukenBuitenLicht/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [v] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
21:50:37 CMD: Status 0
21:50:37 MQT: stat/KeukenBuitenLicht/STATUS = {"Status":{"Module":0,"DeviceName":"KeukenBuitenLicht","FriendlyName":["KeukenBuitenLicht1","KeukenBuitenLicht2"],"Topic":"KeukenBuitenLicht","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[6,6,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:50:37 MQT: stat/KeukenBuitenLicht/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:53:51","StartupUTC":"2020-10-31T19:56:46","Sleep":50,"CfgHolder":4617,"BootCount":4,"BCResetTime":"2020-10-31T20:53:56","SaveCount":19,"SaveAddress":"F9000"}}
21:50:37 MQT: stat/KeukenBuitenLicht/STATUS2 = {"StatusFWR":{"Version":"9.0.0.3(tasmota)","BuildDateTime":"2020-10-31T17:14:54","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"455/699"}}
21:50:37 MQT: stat/KeukenBuitenLicht/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["DeDroog_IoT",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["0000A009","2805C8000100060000005A00000000000000","00000008","00006000","00000000"]}}
21:50:37 MQT: stat/KeukenBuitenLicht/STATUS4 = {"StatusMEM":{"ProgramSize":599,"Free":404,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15701C","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
21:50:37 MQT: stat/KeukenBuitenLicht/STATUS5 = {"StatusNET":{"Hostname":"KeukenBuitenLicht-8185","IPAddress":"10.0.3.63","Gateway":"10.0.3.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.3.1","Mac":"40:F5:20:17:5F:F9","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:50:37 MQT: stat/KeukenBuitenLicht/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.15","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_175FF9","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:50:37 MQT: stat/KeukenBuitenLicht/STATUS7 = {"StatusTIM":{"UTC":"2020-10-31T20:50:37","Local":"2020-10-31T21:50:37","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"07:33","Sunset":"17:09"}}
21:50:37 MQT: stat/KeukenBuitenLicht/STATUS10 = {"StatusSNS":{"Time":"2020-10-31T21:50:37","ANALOG":{"Temperature":60.5},"TempUnit":"C"}}
21:50:37 MQT: stat/KeukenBuitenLicht/STATUS11 = {"StatusSTS":{"Time":"2020-10-31T21:50:37","Uptime":"0T00:53:51","UptimeSec":3231,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"DeDroog_IoT","BSSId":"C6:FB:E4:C1:64:92","Channel":1,"RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Connect a load to O1 only, leaving O2 unused. Watch the web interface, only temperature will be displayed. No power consumption values:
[image]

### EXPECTED BEHAVIOUR
See screenshot from a device with both loads connected:
[image]

### SCREENSHOTS
See above.

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-10-31:
Does not seem to be a general problem, with one load connected I have:
[image]

stefxx, 2020-10-31:
Thanks @sfromis . I assume there is actually nothing connected to O2? Or just switched off?

What Tasmota version are you using?

sfromis, 2020-10-31:
Right, only one load is connected, no wires on O2. Running version 9.0.0.2, but I also remember it working on older versions.

sfromis, 2020-10-31:
Still works after upgrading to latest dev 9.0.0.3

stefxx, 2020-11-01:
I was able to fix this on both Shelly 2.5 devices by flashing Tasmota 8.1.0 and resetting the template. Yes, I had to set (the same) template on 8.1.0 and that instantly started the power measurements on the device. Then I returned to 8.5.0 and 9.0.0.3 and the power measurements continued to work. Even after a "reset 5" and even a "reset 1". 

Note that I did reset the Shelly 2.5 multiple times, reset the template multiple times, all without success on 8.5.0 and 9.0.0.3. Only setting the template at 8.1.0 made it work. Don't quite understand, and I am unsure if this is a bug or just a hickup with my devices. I still think it is quite strange that it only happened to 2 out of the 15 devices, and the only 2 with only a single load connected.

Should I close issue for now?

arendst, 2020-11-01:
Leave open as I want to investigate.

arendst, 2020-11-01:
What happens if you remove power from the device and then reapply power. Does it still work with 9.0.0.3?

stefxx, 2020-11-01:
After removing and reapplying power it still works. I waited around 60 seconds before reapplying the power.

arendst, 2020-11-01:
OK. Let's close it then.

Perhaps you used a different template while tackling the higher temperature issue. Removing or relocating the ADE7953 IRQ gpio did lower the temp but would also make detecting the energy sensor impossible in later versions.

stefxx, 2020-11-01:
One of the affected devices was brand new, never used before. OTA flashed to Tasmota straight out of the box (by using this procedure here: https://github.com/yaourdt/mgos-to-tasmota).

Either way, as I can't reproduce the issue (even with another brand new device) I will close this issue. Thanks!

closed 2020-11-01 by stefxx

foobar26, 2021-07-17:
I can reproduce the issue on my side. Used OTA flash with mgos-to-tasmota with the firmware version 8.5.1.
The power measurement was missing.
Then I reverted back to 7.2.0 and after setting the template again the power measurement was back.
I then got back to 8.5.1 and power management was again there.
Maybe the issue is related to the OTA update?
BTW, I even used reset 5 to erase the flash, but it didn't help.

clau-bucur, 2021-11-12:
I managed to get some pictures of the spare one, which does not work either. Here they are, looks like a rather new revision:
[image]
[image]
[image]

sfromis, 2021-11-12:
Apart from the date, 2020-07-28, looks the same as my from 2019-11-05
[image]

The only issue I know of is energy monitoring not working if not power cycling after flashing Tasmota.

clau-bucur, 2021-11-13:
Then what's the problem here? I have rebooted the device, of course.

barbudor, 2021-11-13:
reboot/restart and power-cycling is not the same thing

clau-bucur, 2021-11-13:
That is correct and embarrassing! My bad. After a power cycle they all provide power measurements.
Thank you.

sanchosk, 2023-04-15:
> reboot/restart and power-cycling is not the same thing

Thanks, helped me, too :)