arendst/Tasmota #9696: Tuya Dimmer it's not possible to turn off with wall switch

ihavenonick, 2020-11-01:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Tuya MCU Modul
- [ x] Tasmota binary firmware version number used: 9.0.0.3
  - [ ] Pre-compiled
  - [ x] Self-compiled
    - [ x] IDE / Compiler used: VSCODE 1.50.1
- [ x] Flashing tools used: OTA
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
07:36:11 MQT: stat/tasmota_8C81B2/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
07:36:11 MQT: stat/tasmota_8C81B2/RESULT = {"Module":{"54":"Tuya MCU"}}
07:36:11 MQT: stat/tasmota_8C81B2/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"288":"Led1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ x] Provide the output of this command: `Status 0`:
```
07:37:12 MQT: stat/tasmota_8C81B2/STATUS = {"Status":{"Module":54,"DeviceName":"Theke","FriendlyName":["Theke"],"Topic":"tasmota_8C81B2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
07:37:12 MQT: stat/tasmota_8C81B2/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:10:22","StartupUTC":"2020-11-01T06:26:50","Sleep":100,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2020-10-31T08:32:13","SaveCount":206,"SaveAddress":"F5000"}}
07:37:12 MQT: stat/tasmota_8C81B2/STATUS2 = {"StatusFWR":{"Version":"9.0.0.3(tasmota)","BuildDateTime":"2020.10.31 08:26:02","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"414/699"}}
07:37:12 MQT: stat/tasmota_8C81B2/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ!","FRITZ!"],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
07:37:12 MQT: stat/tasmota_8C81B2/STATUS4 = {"StatusMEM":{"ProgramSize":493,"Free":508,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashFrequency":40,"FlashMode":3,"Features":["00000407","0F002796","04008001","00000006","000000C0","00000800","00004000","00000000"],"Drivers":"1,2,3,4,9,10,16,20,29","Sensors":"3"}}
07:37:12 MQT: stat/tasmota_8C81B2/STATUS5 = {"StatusNET":{"Hostname":"Theke","IPAddress":"192.168.100.53","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.1","Mac":"BC:DD:C2:8C:81:B2","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
07:37:12 MQT: stat/tasmota_8C81B2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.100.10","MqttPort":1883,"MqttClientMask":"Tuya-Theke","MqttClient":"Tuya-Theke","MqttUser":"Jan","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
07:37:12 MQT: stat/tasmota_8C81B2/STATUS7 = {"StatusTIM":{"UTC":"2020-11-01T06:37:12","Local":"2020-11-01T07:37:12","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"07:13","Sunset":"17:02"}}
07:37:12 MQT: stat/tasmota_8C81B2/STATUS10 = {"StatusSNS":{"Time":"2020-11-01T07:37:12"}}
07:37:12 MQT: stat/tasmota_8C81B2/STATUS11 = {"StatusSTS":{"Time":"2020-11-01T07:37:12","Uptime":"0T00:10:22","UptimeSec":622,"Vcc":3.469,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":101,"MqttCount":1,"POWER":"ON","Dimmer":50,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"FRITZ!","BSSId":"44:4E:6D:58:11:01","Channel":11,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```
- [ x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
07:39:33 MQT: stat/tasmota_8C81B2/RESULT = {"WebLog":4}
07:39:34 CFG: in Flash gespeichert am F4, zählen 207, Bytes 4096
07:39:35 WIF: Prüfe Verbindung...
07:39:40 TYA: Send "55aa00000000ff"
07:39:40 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
07:39:40 TYA: Heartbeat
07:39:51 TYA: Send "55aa00000000ff"
07:39:51 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
07:39:51 TYA: Heartbeat
07:39:55 WIF: Prüfe Verbindung...
07:40:00 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
07:40:00 TYA: fnId=11 is set for dpId=1
07:40:00 TYA: RX Relay-1 --> MCU State: Off Current State:Off
07:40:00 {"TuyaReceived":{"Data":"55AA0007000802020004000000273D","Cmnd":7,"CmndData":"0202000400000027","DpType2Id2":39,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000027"}}}
07:40:00 TYA: fnId=21 is set for dpId=2
07:40:00 TYA: RX value 39 from dpId 2 
07:40:00 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
07:40:00 TYA: fnId=11 is set for dpId=1
07:40:00 TYA: RX Relay-1 --> MCU State: Off Current State:Off
07:40:00 {"TuyaReceived":{"Data":"55AA0007000802020004000000243A","Cmnd":7,"CmndData":"0202000400000024","DpType2Id2":36,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000024"}}}
07:40:00 TYA: fnId=21 is set for dpId=2
07:40:00 TYA: RX value 36 from dpId 2 
07:40:00 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
07:40:00 TYA: fnId=11 is set for dpId=1
07:40:00 TYA: RX Relay-1 --> MCU State: Off Current State:Off
07:40:00 CFG: in Flash gespeichert am FB, zählen 208, Bytes 4096
[... 36 lines cut ...]
07:40:13 TYA: Send "55aa00000000ff"
07:40:13 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
07:40:13 TYA: Heartbeat
07:40:15 WIF: Prüfe Verbindung...
07:40:24 TYA: Send "55aa00000000ff"
07:40:24 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
07:40:24 TYA: Heartbeat

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
if i press power off at wall switch, light will goes off but on again by herself

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
if i press power off at wall switch, light will goes off and still off

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ihavenonick, 2020-11-01:
It's very strange. If the slider or the dimmer on the wall is set to 0 before switching off, the light remains off even if the dimmer is changed afterwards on the switch. Only when I change something again via MQTT does the light come on again by itself when it was switched off.
I think the previous bug 9583 was not fixed properly because it was similar and the web interface and real status were not correct.

effelle, 2020-11-01:
Play a bit with `setoption54` and `setoption20`.

ihavenonick, 2020-11-02:
OK, I'll test this, although it actually does the opposite of what the problem is here. For me the lamp goes on, no matter what I do, and it switches on again by itself.
After restarting the Tuya Dimmer, everything works fine. The problem only starts after a while.
I have now set SetOption54 and 20 to 1, let's see if the light goes out today.

ihavenonick, 2020-11-03:
It works, but i think its not right.

**SetOption20 | Update of Dimmer/Color/CT without turning power on**
I want do power on by set the dimmer and with SetOption20 1 the Ligt goes on wenn i set the dimmer. That is that wat i want, but not wat this option shut do.

**Apply SetOption20 settings to commands from Tuya device**
I dont anderstand this option

closed 2020-11-03 by ihavenonick