arendst/Tasmota #9698: Stability issues when external relay switches

TheChatty, 2020-11-01:
### PROBLEM DESCRIPTION
I have an [ESP-01](https://gloimg.gbtcdn.com/images/pdm-provider-img/straight-product-img/20181016/T012005/T0120050126/source-img/175919-4273.jpg_500x500.jpg) connected to my heating regulator ([Trovis 5573](https://www.samsongroup.com/media/_processed_/e/f/csm_5573_243f28ddb6.png)) via TTL ModBus interface. A [100µF capacitor](https://www.lamaisonduhautparleur.com/3027/condensateur-100f-63v-10-50.jpg) between GND and 3V3 supports the [AMS1117](https://www.pcboard.ca/image/cache/catalog/products/voltage_regulator/ams1117/ams1117-module-800x800.jpg) powered by [1A mains supply](https://www.restore-hannover.de/WebRoot/Store11/Shops/78209156/5750/39EC/69BE/8576/F45A/C0A8/2BB8/FCFB/Original_Apple_Netzteil_5V_1000mA.jpg) or power bank.

I went as far as only connecting ESP GND to 5573 GND and experience ESP hangs (seldom reboots) as soon as I hear the 5573 relay clicking (opening or closing the valve in small amounts).

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP-01
- [x] Tasmota binary firmware version number used: 9.0.0.2
  - [x] Self-compiled with gitpod.io
- [x] Flashing tools used: esptool.py

### TO REPRODUCE
Connect ESP to 5573 and wait for the heating regulator doing its job.

### EXPECTED BEHAVIOUR
No hangs (worse than reboots).

### SCREENSHOTS
n/a

### ADDITIONAL CONTEXT
When I configure the 5573 to fixed water flow (e.g. do not touch the valve) everything works as expected.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-11-01:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and the [Support](https://github.com/arendst/Tasmota#support-information).

Thanks.

### Support Information
See [Documentation](http://tasmota.com/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-11-01 by ascillato2

TheChatty, 2020-11-06:
I wanted to point out that I eventually solved my stability problems by getting rid of my [breadboard](https://ae01.alicdn.com/kf/Ha366d280d4bf44a88dab6a1fc588441fS.jpg). Now Tasmota runs absolutely stable on a ESP-01 communicating with a 5V modbus device without any voltage regulators in between Rx/Tx.