arendst/Tasmota #9704: MQTT-TLS with Let's Encrypt Cert not working

tinkerdudeno1, 2020-11-01:
First of all: Thanks for this awesome projects. I'm sure the market of cheap wifi sockets would not nearly look the way it does if it weren't for Tasmota.
I have an MQTT broker (RabbitMQ) running as a Docker container behind a Traefik reverse proxy. All the TLS stuff is handled by Traefik with a wildcard certificate by Let's Encrypt. With ```openssl s_client --showcerts mqtt.myawesomedomain.tld:443``` I can see that Traefik delivers all intermediate certs without the root CA cert.
I tested this setup with my non-Tasmota ESP8266 projects using the standard Arduino built-in core libs (WifiClientSecure) together with PubSubClient. With MQTT.FX, I'm able to connect and publish/subscribe to topics and see that the ESPs can handle this setup just fine (both publishing and subscribing). So this should work just as fine in Tasmota, I though. But I'm failing miserably.
I am building with the docker image on Windows 10 which is working flawlessly.  In my ```user_config_override.h``` I have:
```
#ifndef USE_MQTT_TLS 
#define USE_MQTT_TLS   
#define USE_MQTT_TLS_CA_CERT
#define USE_MQTT_TLS_FORCE_EC_CIPHER
#endif
#define USE_4K_RSA      // Traefik and Let's encrypt use a 4096 bit public key
#undef  MQTT_PORT
#define MQTT_PORT              443  
#undef  MQTT_TLS_ENABLED
#define MQTT_TLS_ENABLED       true  //there is a checkbox in the mqtt settings of the webgui but it's not persistent when I click it, so I hard coded it into this config
#undef  MQTT_HOST
#define MQTT_HOST         "mqtt.myawesomedomain.tld"
#undef  MQTT_USER
#define MQTT_USER         "bestuserever"
#undef  MQTT_PASS
#define MQTT_PASS         "supersecretpass"
#undef  STA_SSID1
#define STA_SSID1         "funnywifiname"
#undef  STA_PASS1
#define STA_PASS1         "superextrasecretpass"
```
I am flashing the built binary onto a NodeMCU for testing purposes which works just as fine, except the damn thing won't connect to my MQTT server.
With the config above, I get a ```TLS connection error: 27``` which translates to ```BR_ERR_BAD_SIGNATURE```. Unfortunately, Google is not very helpful here so I'm kind of stuck.
I've played around with the above config and it turns out, when I comment out the EC_CIPHER, part, I get a 23 connection error ("```BR_ERR_LIMIT_EXCEEDED```: the server's public key is too large. Tasmota TLS is limited to 2048 RSA keys"). The ```USE_4K_RSA``` is ignored in this case (is it because the weaker cipher can't handle this large key?).
Dropping the CA_CERT option does not change the errors: The EC cipher still fails with a bad signature and the weak cipher complains about the key being too long. I made sure that Traefik supports the two Ciphers that are mentioned in the docs (removing these from the Traefik conf results in a 296 (SSL handshake error)).
Has anybody got a clue where else I should be looking?
Thanks in advance!

closed 2020-11-01 by close-issue-app[bot]

s-hadinger, 2020-11-02:
Could you do a quick test with `tasmota-zbbridge` from the development branch here: http://ota.tasmota.com/tasmota/

This is the only tasmota variant with pre-compiled TLS. This would confirm that the issue is not due to a compilation issue.

tinkerdudeno1, 2020-11-02:
Didn't expect to get a reply on this issue since it's beyond the guidelines/template. Thanks :-)

As suggested, I flashed the dev version of the zbbridge which went fine. I updated my wifi info and was able to connect to the webinterface. However, after like 30 secs or so, the microcontroller would reboot and fall back into AP mode for me to enter my wifi credentials again (and again and again). This happens on both a NodeMCU and a Wemos D1 Mini. Flashed the 8.5 release version only to find out that indeed the dev version was the only one with pre-compiled TLS support. 
Tl;dr: Wasn't able to test the MQTT TLS funcationality with this firmware.

But thanks for the hint, I'll try building on one of my Linux machines to make sure there is no stupid Windows/Docker-compilation issue here.

Should I open another issue so that we're not discussing on a closed issue?

s-hadinger, 2020-11-02:
That's fine. We can continue on this issue

tinkerdudeno1, 2020-11-03:
So the build off my Ubuntu machine show the same result: Error 23 when not using the EC cipher and 27 when I do.

s-hadinger, 2020-11-04:
Can you give more details about your setup? Especially your internal address and the server certificate you are using?

tinkerdudeno1, 2020-11-05:
Sure can: Below, you'll find a docker-compose.yml file with a traefik reverse proxy and the mqtt broker that I use. This is a minimal working environment. You will need to have a domain registered with a provider that grants you access to their API, otherwise Traefik (or rather: the Let's Encrypt certbot) won't be able to get a wildcard certificate for your domain.
To make this work in my private network, my DNS server holds A-records that redirect *.myawesomedomain.tld to my server's private ip address. Traefik pulls a valid wildcard Let's encrypt certificate so whenever I browse to a Traefik service at foo.myawesomedomain.tld, the cert is valid and trusted in any modern browser. No need to install self-signed certs and no need even expose/forward ports to my private network since the DNS challenge happens out of my network. This also works if you just modify your hosts-file and add the domains used below and point them to the machine that's running the services.

So here's the docker-compose.yml:
```
version: "2"

services:
  broker:
    restart: unless-stopped
    image: rabbitmq:management
    # without a hostname, rabbitmq config is not persistent:
    hostname: 'rabbitmq'
    environment:
      - RABBITMQ_DEFAULT_USER=user42
      - RABBITMQ_DEFAULT_PASS=PassW0Rd
    labels:
      # explicitly enables traefik:
      - "traefik.enable=true"
      # creates a tcp router which 'listens' to the host:
      - "traefik.tcp.routers.mqtt-tcp.rule=HostSNI(`mqtt.myawesomedomain.tld`)"
      # enables tls:
      - "traefik.tcp.routers.mqtt-tcp.tls=true"
      # links the router to the service 'mqtt'
      - "traefik.tcp.routers.mqtt-tcp.service=mqtt"
      # defines defines the service loadbalancer port
      - "traefik.tcp.services.mqtt.loadbalancer.server.port=1883"
      # sets the entrypoint of the http router 'rabbitmq' to the entrypoint defined in the traefik service  [optional]
      - "traefik.http.routers.rabbitmq.entrypoints=websecure"
      # sets the host of the http router  [optional]
      - "traefik.http.routers.rabbitmq.rule=Host(`broker.myawesomedomain.tld`)"
      # enables tls:
      - "traefik.http.routers.rabbitmq.tls=true"
      # defines defines the service loadbalancer port
      - "traefik.http.services.rabbitmq.loadbalancer.server.port=15672"
      # sets the network (important if more than one network is attached to a service)
      - "traefik.docker.network=traefik_net"
    networks:
      - traefik_net
    volumes:
      # make sure these two files exist before starting the container:
      - /path/to/config/rabbitmq/enabled_plugins:/etc/rabbitmq/enabled_plugins:ro
      - /path/to/config/rabbitmq/rabbitmq.conf:/etc/rabbitmq/rabbitmq.conf:ro
      # this can be empty, in fact you don't need it for testing but if you want
      # your rabbitmq to be persistent, add this volume:
      - /path/to/config/rabbitmq/data:/var/lib/rabbitmq:rw

  
  traefik:
    # The official v2 Traefik docker image
    image: traefik:v2.2
    # Enables the web UI and tells Traefik to listen to docker
    command:
      # enables the dashboard
      - "--api.insecure=true"
      # sets the directory for the static config files
      - "--providers.file.directory=/configuration/"
      # tells traefik to watch the above folder and auto-update
      - "--providers.file.watch=true"
      # tells traefik to listen to docker events
      - "--providers.docker=true"
      # don't expose services automatically but only if they are enabled explicitly
      - "--providers.docker.exposedbydefault=false"
      # so that traefik can listen to docker events
      - "--providers.docker.endpoint=unix:///var/run/docker.sock"
      # defines an entrypoint named 'web'
      - "--entrypoints.web.address=:80"
      # auto-redirect to https by default
      - "--entrypoints.web.http.redirections.entrypoint.to=websecure"
      - "--entrypoints.web.http.redirections.entrypoint.scheme=https"
[... 39 lines cut ...]
      - "traefik.http.routers.traefik.tls.domains[0].sans=*.myawesomedomain.tld"
      - "traefik.docker.network=traefik_net"
    networks:
      - traefik_net
    restart: unless-stopped

networks:
  traefik_net:
    driver: bridge
```

There are two files that hold a config of rabbitmq. Make sure they exist before firing up the stack with ```docker-compose up```:

/path/to/config/rabbitmq/enabled_plugins:
```
[rabbitmq_management,rabbitmq_prometheus,rabbitmq_mqtt,rabbitmq_web_mqtt].
```
/path/to/config/rabbitmq/rabbitmq.conf:
This file can be empty, there is nothing for this test, that we need to configure. But it must exist, otherwise rabbitmq does not start.

For testing if the broker itself is open for mqtt-connections, I used the tool MQTT.FX, set the url to mqtt.myawesomedomain.tld; Port: 443; User and pass as defined in the broker-service environment section of the docker-compose file; SSL/TLS enabled with TLS v1.2 and "CA signed server certificate". Connects like a charm and allows publishing, subscribing as expected.

I was thinking that it might be worth a try to replace the bearssl libs with the one that the standard Arduino IDE brings along. I understood that the lib used in Tasmota is heavily stipped and modified but I was not able to find a similar lib in my Arduino IDE installation to replace it with. So I gave up and opened this issue instead :-)

Thanks in advance for looking into this.

mkohns, 2020-11-10:
Hi, I struggle with the same issue.
In short
- Program Version:  9.1.0(tasmota)
- Core/SDK Version: 2_7_4_5/2.2.2-dev(38a443e)
- self compiled with user_config_override
#ifndef USE_MQTT_TLS
#define USE_MQTT_TLS    
#define USE_MQTT_TLS_CA_CERT
#define USE_4K_RSA
#endif
- mosquitto with let´s encrypt RSA public key length 4096 bits

I also get the message in the tasmota console
21:25:50 MQT: TLS connection error: 23
meaning: BR_ERR_LIMIT_EXCEEDED: the server's public key is too large. Tasmota TLS is limited to 2048 RSA keys

I thought that the define USE_4K_RSA should handle this as stated in /lib/lib_ssl/bearssl-esp8266/src/t_bearssl_tasmota_config.h line 30
#define BR_MAX_RSA_SIZE 4096    // max 4096 bits RSA keys

any help would be welcome!
If debugging is easier, I can provide a running mosquitto instance for reproduction.

tinkerdudeno1, 2020-11-11:
@mkohns Try including ```#define USE_MQTT_TLS_FORCE_EC_CIPHER``` and see if you get error 27 then. Then it's exactly the same behavior that I experience.

mkohns, 2020-11-11:
I can confirm. After adding #define USE_MQTT_TLS_FORCE_EC_CIPHER, I get error 27.

s-hadinger, 2020-11-11:
Is there any internet facing endpoint I can run some tests on?

mkohns, 2020-11-12:
@s-hadinger I sent the connect infos per email to you

Perhaps worth mentioning ... activating USE_4K_RSA does not reflect/change the output of consumed RAM/FLASH
RAM:   [======    ]  58.6% (used 47992 bytes from 81920 bytes)
Flash: [======    ]  63.3% (used 648057 bytes from 1023984 bytes)

I would expect that values increase - but perhaps I am wrong.

s-hadinger, 2020-11-12:
Ram consumption is essentially dynamic. It is allocated only when the TLS connection is triggered. 

I'll have a look, I don't usually use the 4K option, it may be broken.

s-hadinger, 2020-11-14:
The 4K RSA option configuration was wrong. It is fixed in #9856

tinkerdudeno1, 2020-11-14:
@s-hadinger You are THE man! Works like a charm** not even two weeks after posting this issue. That's amazing, thanks for looking into this.

** Only works when also using ```#define USE_MQTT_TLS_FORCE_EC_CIPHER``` (personally, I would always recommend this). Just out of curiosity, I tried without the EC_CIPHER and added the weak ```RSA_WITH_AES_128_GCM_SHA256``` cipher to my MQTT server conf, but then Tasmota would crash and reboot when trying to connect to the server. This just for documentation if anyone stumbles upon this issue.

If the issue weren't already closed to begin with, I'd suggest to close it now :-)

stewiem2000, 2020-12-04:
@tinkerdudeno1  I found that the reason it only works when <code>#define USE_MQTT_TLS_FORCE_EC_CIPHER</code> is also specifie is because more stack space is allocated in <code>tasmota/StackThunk_light.cpp</code> (~line 45).  Perhaps <code>|| defined(USE_4K_RSA)</code> should be added there @s-hadinger?

s-hadinger, 2020-12-06:
Yes you are probably right. Nobody tried 4K RSA without EC before. Let me do the change.

PR #10075 (2020-12-06): Change Thunk Stack size for 4K RSA #9704
## Description:

Force bigger Thunk Stack if 4K RSA even without EC ciphers.

**Related issue (if applicable):** fixes #9704

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [ ] The code change is tested and works on Tasmota core ESP32 V.1.0.4.2
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/de [... cut]