arendst/Tasmota #9705: Sunset/sunrise schedule functionality goes crazy around timezones

nuschpl, 2020-11-02:
### PROBLEM DESCRIPTION
I had tasmota set to turn light off at sunrise. Today sunset was arround 6:39 localtime. Tasmota turned of the light of at 8:39 localtime, while the UI shows it want's to turn it of at 7:39 localtime. My timezone is GMT+1 or GMT+2 depending on DST do it seems it all was mixed and handled differently at differnet level. Not sure if this was documented where timezone correction should be done - if at UI level or backend and if Tasmota or operating system should handle those.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-11-02:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-11-02 by ascillato2

nuschpl, 2020-11-02:
This is a bug because existing feature doesn't work as designed, am I missing something ?  I can spend my time into debugging and brainstorming on open-source project but I don't want waste my time on unnecessary bureaucracy. This "dynamic" channel means you are waiting for someones availability while it's not necessary to solve any problem.

sfromis, 2020-11-02:
It means that questions about how to use Tasmota are better handled there. You have posted no information to indicate some sort of bug.

nuschpl, 2020-11-04:
I don't have questions about how to use Tasmota, and I think no one should if UI is self explaining. The bug is that scheduled toggle of Tasmota input/output scheduled to available option 'sunrise' doesn't do that at proper time. I don't know why because I haven't read full Tasmote source.  I've added also suggestion that it may be timezone related as the toggle time is shifted by entire hour. I've spend some of my free time to provide as much info as I had with plan to get back, do some debug and potentially pull request, while giving others chance to put some suggestions - maybe problem is known. That's how I understand open source and Github Issues as central place to marshall all open problems which are not the code.
Unfortunatelly recently there are more and more "project managers" on GitHub who value order and closing as much tasks as possible more than contribution. It's like requesting from construction workers to put all tools into garage after placing every subsequent brick.

arendst, 2020-11-04:
Perhaps you could start to read the documentation instead of the ccode.

Considering the fact that thousand. of Polish users have no issue with sunrise and sunset it might be your parametrization is just wrong.

From your information I see at least the timezone setting is wrong. Find hints in the docs, discord or eat your heart out by scanning the code.

I feel more hints are needed in due time........

ascillato, 2020-11-04:
> I don't have questions about how to use Tasmota, and I think no one should if UI is self explaining. The bug is that scheduled toggle of Tasmota input/output scheduled to available option 'sunrise' [...]
Unfortunatelly recently there are more and more "project managers" on GitHub who value order and closing as much tasks as possible more than contribution. It's like requesting from construction workers to put all tools into garage after placing every subsequent brick.

Sorry, there is no bug, it is just a misconfiguration of DST and STD. Your issue is duplicated and has been addressed at least 6 times before. So, as you don't want to ask in Discord, please post here the output of the following commands and we can post to you the correct syntax for your zone.

```lua
timezone
timestd
timedst
```

and your city, country

arendst, 2020-11-04:
And in addition for using sunrise and sunset correctly you will need to set your latitude and longitude correctly.

But I'm quitte sure you had found that already in the docs too.

nuschpl, 2020-11-04:
Sorry I've missed this
`99 == Settings.timezone`
but is there really a reasoning behind enforcing user to dig the docs instead of self explaining interface ?

Currently setting timezone to some value may be misleading comparing to what we know from desktops where we usually start with setting timezone then apply DST or not. 

If this double use of Settings.timezone is about saving memory then we can save even more and with less confussion . If you'll look at the struct: 

```
typedef union {
  uint16_t data;
  struct {
    uint16_t hemis : 1;                    // bit 0        = 0=Northern, 1=Southern Hemisphere (=Opposite DST/STD)
    uint16_t week : 3;                     // bits 1 - 3   = 0=Last week of the month, 1=First, 2=Second, 3=Third, 4=Fourth
    uint16_t month : 4;                    // bits 4 - 7   = 1=Jan, 2=Feb, ... 12=Dec
    uint16_t dow : 3;                      // bits 8 - 10  = day of week, 1=Sun, 2=Mon, ... 7=Sat
    uint16_t hour : 5;                     // bits 11 - 15 = 0-23
  };
} TimeRule;
```

contradicting values of `TimeRule.hemis` could be used to signal that DST is inactive(and XOR on both could be used instead of `Settings.timezone=='99' `part). Then we can get rid of `Settings.timezone` at all by putting it's value in `Settings.tflag[0].offset` 

Then already existing function:
`void SettingsResetDst(void)`
can be used to revert if user do e.g:
`timedst disable`

Also probably using precompiled db based on one of those https://en.wikipedia.org/wiki/Tz_database#Usage_and_extensions would make people life easier.