arendst/Tasmota #9711: Tasmota-IR - incorrect AC command sequences

bogd, 2020-11-02:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I am trying to use Tasmota to send IR commands to a Samsung AC. However, there are two issues:
1) Many commands are not recognized (for example, no "power:off" commands seem to work)
2) The received IR code seems to differ from what I am sending out.

I did consider the fact that this might be an issue with the IRRemoteESP8266 library, but when using that library directly from my code, on the same hardware, all commands are recognized properly.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 Mini, with IR shield (transmitter on GPIO0, receiver on GPIO2)
- [x] Tasmota binary firmware version number used: 8.5.1-ir
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
17:47:00 MQT: stat/tas-ir-test/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
17:47:00 MQT: stat/tas-ir-test/RESULT = {"Module":{"0":"Generic"}}
17:47:00 MQT: stat/tas-ir-test/RESULT = {"GPIO0":{"8":"IRsend"},"GPIO1":{"0":"None"},"GPIO2":{"51":"IRrecv"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
none

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

17:47:57 CMD: status 0
17:47:57 MQT: stat/tas-ir-test/STATUS = {"Status":{"Module":0,"DeviceName":"tas-ir-test","FriendlyName":["tas-ir-test"],"Topic":"tas-ir-test","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:47:57 MQT: stat/tas-ir-test/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-ir.bin","RestartReason":"Software/System restart","Uptime":"0T00:22:32","StartupUTC":"2020-11-02T16:25:25","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2020-11-02T17:20:26","SaveCount":16,"SaveAddress":"FA000"}}
17:47:57 MQT: stat/tas-ir-test/STATUS2 = {"StatusFWR":{"Version":"8.5.1(ir)","BuildDateTime":"2020-10-02T10:12:16","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"392/699"}}
17:47:57 MQT: stat/tas-ir-test/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxx",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
17:47:57 MQT: stat/tas-ir-test/STATUS4 = {"StatusMEM":{"ProgramSize":568,"Free":432,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1625C2","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F5A0582","04000001","00000002","00000000","00000000","00000000","00000000"],"Drivers":"1,2,4,5,7,9,10,12","Sensors":""}}
17:47:57 MQT: stat/tas-ir-test/STATUS5 = {"StatusNET":{"Hostname":"tas-ir-test-6160","IPAddress":"192.168.xx.157","Gateway":"192.168.xx.1","Subnetmask":"255.255.255.0","DNSServer":"208.67.220.220","Mac":"B4:E6:2D:xx:xx:xx","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:47:57 MQT: stat/tas-ir-test/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.xx.xx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1B5810","MqttUser":"xxxxxxx","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:47:57 MQT: stat/tas-ir-test/STATUS7 = {"StatusTIM":{"UTC":"2020-11-02T16:47:57","Local":"2020-11-02T17:47:57","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:39","Sunset":"17:27"}}
17:47:57 MQT: stat/tas-ir-test/STATUS10 = {"StatusSNS":{"Time":"2020-11-02T17:47:57"}}
17:47:57 MQT: stat/tas-ir-test/STATUS11 = {"StatusSTS":{"Time":"2020-11-02T17:47:57","Uptime":"0T00:22:32","UptimeSec":1352,"Vcc":2.984,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"xxxxx","BSSId":"xx:xx:xx:xx:xx:xx","Channel":x,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
17:50:35 CMD: IRhvac {"Vendor":"SAMSUNG_AC", "Power":"Off","Mode":"Cool","FanSpeed":0,"Temp":24,"Light": "On", "Clean": "On"}
17:50:36 MQT: stat/tas-ir-test/RESULT = {"IRHVAC":{"Vendor":"SAMSUNG_AC","Model":1,"Mode":"Off","Power":"Off","Celsius":"On","Temp":24,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"On","Beep":"Off","Sleep":-1}}
17:50:36 MQT: tele/tas-ir-test/RESULT = {"IrReceived":{"Protocol":"SAMSUNG_AC","Bits":168,"Data":"0x0x02B20F000000C001D20F000000000102FF718011C0","Repeat":0,"IRHVAC":{"Vendor":"SAMSUNG_AC","Model":-1,"Mode":"Off","Power":"Off","Celsius":"On","Temp":16,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}

See below for more details

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Using Tasmota-IR, try to control Samsung AC.

For example:
1) 
[... 19 lines cut ...]
### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
N/A

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
For troubleshooting potentially non-Tasmota-related issues (like hardware issues, or compatibility issues between IRRemoteESP8266 and my hardware, I have tested with code similar to this:

```
IRSamsungAc ac(IR_LED);
ac.begin();
ac.off();
ac.setTemp(26);
ac.setMode(kSamsungAcHeat);
ac.setClean(true);
ac.send();
```

The code works consistently, and the generated IR codes are correctly recognized by my AC.

So this looks like some issue when the tasmota commands are converted into Samsung AC commands. 

Am I missing something from the docs? Is my tasmota command syntax wrong?

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-11-02:
This is strange. Tasmota doesn't do anything much different from your simple example. I will try to have 2 devices decoding each other.

bogd, 2020-11-02:
Please note that in my case I had only one device (which was basically seeing its own reflected signal, as it was sending it). I don't know if it matters, but I thought I should mention it :) .

Just to double-check, I flashed the exact same device with my simple code, and tried turning on and off the exact same AC. With the simple code, the commands were recognized every single time - with tasmota, it was impossible to get an "AC:off" command to be recognized.

If it helps, I can try sending the tasmota codes to a different device running IRRemoteESP8266's IRDump code, and post the results here.

s-hadinger, 2020-11-02:
Interestingly when I tried `IRhvac {"Vendor":"SAMSUNG_AC", "Power":"Off","Mode":"Cool","FanSpeed":0,"Temp":24,"Light": "On", "Clean": "On"}` it crashed. 

I definitely need to look closer.

utech-git, 2020-11-02:
Yeah, I second that. I tried tasmota-ir with samsung AC and not all commands were recognized. The "power off" command wasn't working, as OP said. But the command for "cool", "16 degrees", "no swing" was recgnized as "fan only" and not "cool". I'll keep an eye on this thread.

bogd, 2020-11-03:
I tried looking through the code (even though most of it is beyond my programming abilities). Even though I still cannot figure out the cause of the issue, something seems strange in [xdrv_05_irremote_full](https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_05_irremote_full.ino), on line 432. This sets the fan speed, but only if the requested value is between 1 and 5. 

With the state initialized to kMedium (line 409), and a default of "get fan speed from string" (line 435), does this mean that we can never set the fan speed to auto using "Fanspeed": 0 ?

s-hadinger, 2020-11-03:
@bogd This code can now be improved with JSMN to make it cleaner.

Actually it should have worked, because if you put `0` as input, it would be considered invalid integer, then try to parse as string, fail and revert to the default of `IRac::strToFanspeed()` which is `auto`. I agree it's not straightforward so I will improve it.

bogd, 2020-11-04:
I ran some additional tests, and the results are... somewhat mixed.

I flashed another D1 mini with the code below, trying (to the best of my abilities) to simulate the way in which Tasmota uses the IRRemoteESP8266 library.
```
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Samsung.h>
#include <IRac.h>

#define IR_LED_SEND 0
IRsend *irsend = nullptr;

void setup() {
  //Prepare IR sending port
  //irsend.begin();
  irsend = new IRsend(IR_LED_SEND);
  irsend->begin();

  Serial.begin(115200);
  Serial.println("Starting...");
}

void loop() {
  stdAc::state_t state;
  IRac ac(IR_LED_SEND);

  state.protocol = decode_type_t::UNKNOWN;
  state.model = 1;  // Some A/C's have different models. Let's try using just 1.
  state.mode = stdAc::opmode_t::kAuto;  // Run in cool mode initially.
  state.power = false;  // Initially start with the unit off.
  state.celsius = true;  // Use Celsius for units of temp. False = Fahrenheit
  state.degrees = 21.0f;  // 21 degrees.
  state.fanspeed = stdAc::fanspeed_t::kMedium;  // Start with the fan at medium.
  state.swingv = stdAc::swingv_t::kOff;  // Don't swing the fan up or down.
  state.swingh = stdAc::swingh_t::kOff;  // Don't swing the fan left or right.
  state.light = false;  // Turn off any LED/Lights/Display that we can.
  state.beep = false;  // Turn off any beep from the A/C if we can.
  state.econo = false;  // Turn off any economy modes if we can.
  state.filter = false;  // Turn off any Ion/Mold/Health filters if we can.
  state.turbo = false;  // Don't use any turbo/powerful/etc modes.
  state.quiet = false;  // Don't use any quiet/silent/etc modes.
  state.sleep = -1;  // Don't set any sleep time or modes.
  state.clean = false;  // Turn off any Cleaning options if we can.
  state.clock = -1;  // Don't set any current time if we can avoid it.
  
  // On, 26 degrees, cool, autofan
  state.protocol = decode_type_t::SAMSUNG_AC;
  state.mode = stdAc::opmode_t::kHeat;
  state.degrees = 26;
  state.power = true;
  state.fanspeed = stdAc::fanspeed_t::kAuto;
  state.light = true;
  Serial.println("Data1 ready, sending");
  ac.sendAc(state,&state);
  delay(15000); 

  // Off, 26 degrees, cool, autofan, autoclean
  state.protocol = decode_type_t::SAMSUNG_AC;
  state.mode = stdAc::opmode_t::kHeat;
  state.degrees = 26;
  state.power = false;
  state.fanspeed = stdAc::fanspeed_t::kAuto;
  state.light = true;
  state.clean = true;
  Serial.println("Data2 ready, sending");
  ac.sendAc(state,&state);
  delay(15000);
}

```
I tested the code with my AC, and the on/off commands were correctly recognized. 

I also captured the output using both Tasmota and another D1 mini running the IRDump code from the library's examples. The results look OK, even though the "clean" flag is not correctly set in the second one (but this problem seems to be on the library side).

Tasmota capture:
```
08:39:43 MQT: tele/tas-ir-test/RESULT = {"IrReceived":{"Protocol":"SAMSUNG_AC","Bits":168,"Data":"0x0x02920F000000F001D20F0000000001E2FE718011F0","Repeat":0,"IRHVAC":{"Vendor":"SAMSUNG_AC","Model":-1,"Mode":"Auto","Power":"On","Celsius":"On","Temp":16,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
08:39:58 MQT: tele/tas-ir-test/RESULT = {"IrReceived":{"Protocol":"SAMSUNG_AC","Bits":168,"Data":"0x0x02B20F000000C001D20F000000000102FF718011C0","Repeat":0,"IRHVAC":{"Vendor":"SAMSUNG_AC","Model":-1,"Mode":"Off","Power":"Off","Celsius":"On","Temp":16,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
08:40:03 MQT: tele/tas-ir-test/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":6,"Hash":"0xF257E7D6","Repeat":0}}
```

Next step - flashing the first D1 with tasmota, sending from there, and seeing what the other two actually get :)

bogd, 2020-11-04:
Interestingly enough, after flashing tasmota on another device, I cannot reproduce the issue. The exact same command as before is recognized as an "off" command by the AC.

There are still some minor issues, like:
1) Tasmota prepending "0x" twice in front of the received IR codes (see RESULT.IrReceived.Data)
2) Tasmota not decoding the temperature from the received IR signal correctly (even though the library does), and defaulting to 16 degrees
3) the "clean" flag not being set in the sent IR signal (although this might be a library issue, because I was seeing it with my simple code above as well.

The results of my tests are below. I will run some additional tests to try and pinpoint the issue - but if I cannot (and I cannot reproduce it consistently), I will probably have to close this until I can collect more details.

Results of additional testing:

"On" command sent from tasmota (ir-test2), decoded by another tasmota (ir-test) and IRDump:
```
09:59:24 CMD: IRhvac {"Vendor":"SAMSUNG_AC", "Power":"On","Mode":"Heat","FanSpeed":0,"Temp":26,"Light": "On", "Clean": "Off"}
09:59:24 MQT: stat/tas-ir-test2/RESULT = {"IRHVAC":{"Vendor":"SAMSUNG_AC","Model":1,"Mode":"Heat","Power":"On","Celsius":"On","Temp":26,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}
09:59:24 MQT: tele/tas-ir-test2/RESULT = {"IrReceived":{"Protocol":"SAMSUNG_AC","Bits":168,"Data":"0x0x02920F000000F001D20F0000000001E2FE718011F0","Repeat":0,"IRHVAC":{"Vendor":"SAMSUNG_AC","Model":-1,"Mode":"Auto","Power":"On","Celsius":"On","Temp":16,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
```
Tasmota decode:
```
09:59:24 MQT: tele/tas-ir-test/RESULT = {"IrReceived":{"Protocol":"SAMSUNG_AC","Bits":168,"Data":"0x0x02920F000000F001D20F0000000001E2FE718011F0","Repeat":0,"IRHVAC":{"Vendor":"SAMSUNG_AC","Model":-1,"Mode":"Auto","Power":"On","Celsius":"On","Temp":16,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
09:59:35 MQT: tele/tas-ir-test/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":3,"Hash":"0x4CB0FADF","Repeat":0}}
0
```
IRDump decode:
```
Protocol  : SAMSUNG_AC
                      Code      : 0x02920F000000F001D2FE71A041F0 (112 Bits)
                                                                           Mesg Desc.: Power: On, Mode: 4 (Heat), Temp: 26C, Fan: 0 (Auto), Swing: Off, Beep: Off, Clean: Off, Quiet: Off, Powerful: Off, Breeze: Off, Light: On, Ion: Off
uint16_t rawData[233] = {670, 17910,  3084, 8906,  576, 454,  574, 1448,  574, 454,  574, 452,  576, 454,  576, 450,  578, 452,  574, 456,  576, 450,  576, 1448,  576, 452,  576, 450,  574, 1448,  576, 454,  576, 452,  574, 1450,  578, 1446,  574, 1448,  574, 1450,  574, 1448,  576, 452,  574, 454,  574, 454,  578, 450,  574, 454,  576, 452,  574, 454,  574, 454,  576, 450,  576, 452,  576, 452,  576, 452,  576, 452,  576, 452,  574, 452,  576, 452,  574, 452,  574, 454,  574, 452,  576, 452,  576, 452,  574, 456,  574, 452,  574, 452,  578, 450,  576, 452,  574, 452,  576, 452,  574, 454,  574, 452,  576, 452,  574, 452,  576, 1448,  576, 1448,  576, 1446,  576, 1448,  576, 2912,  3082, 8876,  578, 1446,  576, 452,  574, 452,  576, 452,  574, 454,  574, 454,  574, 454,  572, 456,  576, 452,  574, 1448,  576, 450,  576, 452,  576, 1446,  576, 452,  574, 1448,  576, 1448,  576, 454,  572, 1450,  576, 1448,  578, 1446,  576, 1446,  576, 1448,  576, 1448,  578, 1448,  576, 1448,  576, 454,  574, 454,  574, 452,  574, 1448,  574, 1450,  576, 1448,  578, 454,  574, 454,  574, 452,  574, 454,  574, 454,  576, 452,  574, 1448,  576, 454,  574, 1448,  574, 1448,  574, 454,  574, 454,  574, 454,  572, 454,  574, 452,  572, 1450,  578, 452,  574, 454,  572, 454,  572, 454,  576, 452,  574, 1448,  574, 1450,  574, 1448,  574, 1448,  572};  // SAMSUNG_AC
                                                                                    uint8_t state[14] = {0x02, 0x92, 0x0F, 0x00, 0x00, 0x00, 0xF0, 0x01, 0xD2, 0xFE, 0x71, 0xA0, 0x41, 0xF0};
```

"Off" command:
```
10:02:55 MQT: stat/tas-ir-test2/RESULT = {"IRHVAC":{"Vendor":"SAMSUNG_AC","Model":1,"Mode":"Off","Power":"Off","Celsius":"On","Temp":26,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"On","Beep":"Off","Sleep":-1}}
10:02:55 MQT: tele/tas-ir-test2/RESULT = {"IrReceived":{"Protocol":"SAMSUNG_AC","Bits":168,"Data":"0x0x02B20F000000C001D20F000000000102FF718011C0","Repeat":0,"IRHVAC":{"Vendor":"SAMSUNG_AC","Model":-1,"Mode":"Off","Power":"Off","Celsius":"On","Temp":16,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
```
Tasmota decode:
```
10:05:02 MQT: tele/tas-ir-test/RESULT = {"IrReceived":{"Protocol":"SAMSUNG_AC","Bits":168,"Data":"0x0x02B20F000000C001D20F000000000102FF718011C0","Repeat":0,"IRHVAC":{"Vendor":"SAMSUNG_AC","Model":-1,"Mode":"Off","Power":"Off","Celsius":"On","Temp":16,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
[... cut ...] 586, 444,  584, 446,  584, 446,  584, 446,  584, 446,  584, 446,  584, 446,  586, 446,  584, 1440,  588, 1440,  586};  // SAMSUNG_AC
                                                                                                                                                                                                                        uint8_t state[21] = {0x02, 0xB2, 0x0F, 0x00, 0x00, 0x00, 0xC0, 0x01, 0xD2, 0x0F, 0x00, 0x00, 0x00, 0x00, 0x01, 0x02, 0xFF, 0x71, 0x80, 0x11, 0xC0};
```

Jason2866, 2020-11-04:
A wemos ir shield is not a powerfull transmitter. There are many nice ready made IR Bridges.
The issues you have can maybe a result of the not optimal ir setup
You find some Tasmota flashable devices here https://templates.blakadder.com/misc.html

utech-git, 2020-11-04:
I am using tasmota on a [YTF IR Bridge](https://templates.blakadder.com/ytf_ir_bridge.html) bought [here](https://www.aliexpress.com/item/32951202513.html)

s-hadinger, 2020-11-04:
Thanks for testing. I don't have time in the short term to try between 2 Tasmota devices, one for sending one for receiving. 

I don't see why the temperature is wrong. Tasmota just reports the value from the lib without any calculation.

bogd, 2020-11-06:
@Jason2866 , @utech-git - thank you for the links! Unfortunately, I cannot source those IR bridges locally, and the pandemic has made ordering from China more difficult (I've had multiple instances of sellers that could not ship the items, or items that never arrived). I do have a couple of YTF bridges on the way now - hopefully they will make it through to me!

However, it is worth mentioning that I have been using a Wemos IR shield for more than 1 year now to reliably control my AC (using some custom code written by me, that simply replays some predefined IR sequences). So while it might not be the most powerful device possible, I know for a fact that it can get the job done.

@s-hadinger - there definitely is a problem with the Tasmota temperature decoding. Note that for the exact same received code (the one ending in "8011C0", the IRDumpV3 code from the library decodes to 24 degrees (correct), while Tasmota decodes to 16 degrees (incorrect). 

So there are two issues left here (both unrelated to the original one, which I am unable to reproduce now :| ):
1) Double "0x" on IR received data. This one is easy :) 
[The code](https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_05_irremote_full.ino), on line 241, prepends the string with "0x", then calls [resultToHexidecimal()](https://github.com/arendst/Tasmota/blob/development/lib/lib_basic/IRremoteESP8266-2.7.11/src/IRutils.cpp), which returns an already-prepended string (line 307)

2) Problems with the decoded temperature. This one I cannot figure out - I can see [the conversion code](https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_05_irremote_full.ino) on lines 210 and above, but as far as I can tell this does use the returned value directly (in my case, the AC uses integer values for the temperature, so line 211 should apply).
Yet, as you can see from the captures above, the exact same hex code is decoded differently by IRDump and Tasmota.

Please let me know:
a) whether I should keep this issue open to work on these two minor problems, or (since they are unrelated to the original issue, and to the title) I should open another issue for them
b) what I could do in order to help debug/troubleshoot the issues further

Thank you!

bogd, 2020-11-23:
Update - I have managed to source some dedicated IR bridges (YTF and Eachen). They seem to be able to control the AC correctly, but the above two issues persist.

Also, I have noticed one more thing - after the initial flashing with tasmota-ir, any attempt to send an IRhvac command (via the web console, but I would expect the result to be the same over MQTT) results in a software reset. Repeatedly, and reproducible 100% of the time, on all my new hardware. 

Workaround: if I point the existing AC remote to the device and send a command, once that command is correctly decoded, everything works fine. (when I say "correctly decoded", I mean everything else except the temperature - that one is always decoded as 16 degrees - see above).

I would suspect a problem with the way tasmota handles the internal state of the device, and how it uses the default original state.

s-hadinger, 2020-11-24:
Hi, I have located the issue and opened an issue in IRRemoteESP8266. The fix is easy and will come shortly.
https://github.com/crankyoldgit/IRremoteESP8266/issues/1339

s-hadinger, 2020-11-24:
Now I understand what is happening. When you set `"Power":1` for `SAMSUNG_AC` and IRRemoteESP8266 does not know the previous state, it actually sends 2 IR commands. The first one is a fixed message to turn the device ON, and the second message contains the correct temperature.
The reason for not seeing the second message in Tasmota is because there is currently a 500ms pause before receiving a new message. I will reduce the delay.

Ex:
Sendind `IRhvac {"Vendor":"SAMSUNG_AC", "Power":"On","Mode":"Cool","FanSpeed":0,"Temp":24,"Light": "On", "Clean": "On"}`

It receives both messages (although decoding is not very reliable). The first contains 168 bits, the second has 112 bits and a correct temperature.

```
19:32:57 RSL: tele/tasmota_5C3233/RESULT = {"IrReceived":{"Protocol":"SAMSUNG_AC","Bits":168,"Data":"0x02920F000000F001D20F0000000001E2FE718011F0","Repeat":0,"IRHVAC":{"Vendor":"SAMSUNG_AC","Model":-1,"Mode":"Auto","Power":"On","Celsius":"On","Temp":16,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
19:32:57 RSL: tele/tasmota_5C3233/RESULT = {"IrReceived":{"Protocol":"SAMSUNG_AC","Bits":112,"Data":"0x02920F000000F001C2FEF18211F0","Repeat":0,"IRHVAC":{"Vendor":"SAMSUNG_AC","Model":-1,"Mode":"Cool","Power":"On","Celsius":"On","Temp":24,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"On","Filter":"Off","Clean":"On","Beep":"Off","Sleep":-1}}}
```

Also when sending `IRhvac {"Vendor":"SAMSUNG_AC", "Power":"Off","Mode":"Cool","FanSpeed":0,"Temp":24,"Light": "On", "Clean": "On"}`, since Power is off, IRRemoteESP8266 only sends a fixed OFF message and ignores all parameters.

closed 2020-11-24 by arendst

bogd, 2020-11-25:
@s-hadinger  - that is great news, thank you! Looking forward to the fix!

One more thing I need to mention - this approach of simply sending a fixed OFF message might be causing problems with some devices. In my case, there is an "autoclean" function (that simply keeps the fan running for an additional 15 minutes or so, to dry the unit) , and I believe that the bit for that function is actually part of the OFF message. 

So without the full OFF message, it might be impossible to activate the autoclean function.

All of the above is based on my (limited) understanding of the IR protocol, based on my own decoding - it might be wrong, and that is why I am hesitant to open another issue about it. I will run some more tests, and see if the unit actually preserves a "clean: on" command set at power on.

Until then, thank you once again for your help! I am looking forward to the next release, including the fix.

s-hadinger, 2020-11-25:
Thanks for the feedback. The behavior in case of Power:Off is hardcoded in the IRRemoteESP8266 lib and we have no control on it from Tasmota.
I suggest you open an issue in their Github project.

Fyi, the code is there: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Samsung.cpp#L348-L357

```
    if (this->getPower()) {
      this->sendOn(repeat);
    } else {
      this->sendOff(repeat);
      return;  // No point sending anything else if we are turning the unit off.
    }
```

PR #9969 (2020-11-24): IR fix SAMSUNG_AC
## Description:

Fix crash when sending commands to SAMSUNG_AC, until fixed in the library.
https://github.com/crankyoldgit/IRremoteESP8266/issues/1339

Reduce inter-messages pause to receive both SAMSUNG_AC messages.
Also disable receiving of IR messages while sending them, avoiding mirror effect and receiving the messages just sent.

**Related issue (if applicable):** fixes #9711

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles wit [... cut]