arendst/Tasmota #9712: Zigbee2Tasmota Support for Sinope TH1123ZB

brianneate, 2020-11-02:
**Have you looked for this feature in other issues and in the docs?**  
Yes but could not find any specific information for the Sinope TH1123ZB

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I cannot find documentation if the Sinope TH1123ZB line voltage thermostat is directly supported by the Zigbee2Tasmota implementation.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I would like to have support for the Sinope TH1123ZB line voltage thermostat in the Zigbee2Tasmota implementation so I can control the thermostat directly from a flashed ZBBridge with a schedule.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
I think that an alternative would be using the ZBBridge as a gateway only with a home assistant ZHA system interacting with the thermostat.

**Additional context**  
_Add any other context or screenshots about the feature request here._

The following wiki describes the support for this thermostat in the zigbee2mqtt project https://www.zigbee2mqtt.io/devices/TH1123ZB.html

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-11-02:
It looks like this device uses standard thermostat zigbee attributes. It should work out of the box. 

Do you have the device? Maybe some important attributes are still missing.

brianneate, 2020-11-02:
Thanks for the quick reply!

I do not have the device right now but a few of them are in the mail and I should have them in a week. I was trying to decide between ordering a ZBBridge or a zigbee stick so wanted to make sure the device could have support added to the ZBBridge flashed with Tasmota.

I'll order the ZBBridge and update here when I try it out with the Sinope thermostat.

s-hadinger, 2020-11-02:
Good. Thermostat support on zbbridge is recent so features may still be missing. But it shouldn't be a problem to add what's missing

ascillato2, 2020-11-04:
> I'll order the ZBBridge and update here when I try it out with the Sinope thermostat.

Great, Thanks.

closed 2020-11-04 by ascillato2

2o00o8e9, 2020-12-22:
Did you manage to control your thermostat?  I have done the same thing, I have flashed a ZBBridge wit tasmota, and using Z2T to control my TH1123ZB. If anybody is interested, I can help them with the setting up.

It works fine, I'm able to control the SetPoint, and able to read the values that the Thermostat is publishing, but I'm looking for the "variables" names to use to control the more advanced features (keypar lock, outdoor temp, ferquency of publishing, etc).

Any Idea how to figure put the names?  I have found the attached files and this site https://www.zigbee2mqtt.io/devices/TH1123ZB.html

 that I thought could help me figure out the names, but with no success.  Any help will be greatly appreaciated!

[Sinope-Technologies-Zigbee-Thermostat-V.1.0.0-SVN-547.txt](https://github.com/arendst/Tasmota/files/5727288/Sinope-Technologies-Zigbee-Thermostat-V.1.0.0-SVN-547.txt)
[Sinope-Technologies-TH1300ZB-V.1.0.5-SVN-503.txt](https://github.com/arendst/Tasmota/files/5727294/Sinope-Technologies-TH1300ZB-V.1.0.5-SVN-503.txt)
[TH1123ZB-TH1124ZB-Sinope.txt](https://github.com/arendst/Tasmota/files/5727295/TH1123ZB-TH1124ZB-Sinope.txt)

brianneate, 2020-12-22:
I ended up using ZHA with the ZBBridge which works fine for basic climate control and temperature/energy usage sensors. Like you I have not been able to get any of the advanced functionality (setting 12/24 hour time, outside temperature, etc...) to work though.

I was hoping to setup my schedules on the ZBBridge directly and only have the current states published to home assistant over MQTT but have not been able to figure out how that works yet.

s-hadinger, 2020-12-23:
@2o00o8e9 The variables names are device specific. As a rule of thumb, when you receive a variable from the device, you can often write the attribute too.

Are are all the predefined values in Tasmota:

LocalTemperature
OutdoorTemperature
PICoolingDemand
PIHeatingDemand
LocalTemperatureCalibration
OccupiedCoolingSetpoint
OccupiedHeatingSetpoint
UnoccupiedCoolingSetpoint
UnoccupiedHeatingSetpoint
RemoteSensing
ControlSequenceOfOperation
SystemMode
TRVMode
ValvePosition
EurotronicErrors
CurrentTemperatureSetPoint
EurotronicHostFlags

Zigbee Tuya devices uses a different set of attributes.

2o00o8e9, 2020-12-24:
Thanks @s-hadinger.

When you say that they are device specific, is there a way that I can find out what a particular device values are?  I'm interested on finding out out to write the following to my thermostat (which I know can be done since they work when using the manufacturer gateway...):
Lockin the keypad
Displaying the time
displaying the outdoor temperature
Setting the display to be illuminated only on demand

Also, I have been able to read the OutdoorTemperature.  It reports -327.68 (which as per Zigbee is equivalent to 0x8000, and means invalid/ unknown.

I get a success message also when I try to write to that attribute, but If I read from it right afterward, it still displays -327.68... any idea?

s-hadinger, 2020-12-24:
TRV support in Tasmota is recent, so there are probably things to improve.

Best case is to refer to vendor specs (Eurotronic for ex documents all attributes available).
Next you can use tools like Deconz to do an attribute discovery - this is not implemented in Tasmota and probably won't be (limited resources)
Next is to just listen to reported attributes, and try writing values to the same attributes.

If `OutdoorTemperature` returns 0x8000 then it's definitely invalid. I believe it means that the device does not support it.

The Zigbee ZCL documentation also lists attributes and whether they are Read/Write capable of.